MoshiMoshi0/TTController #1: Device support

devcompl, 2019-01-21:
I only have Riing Plus fans so I need help with confirming support for other TT Rgb Plus devices.

Notes:
- The PSUs, keyboards and mice are confirmed non working since they use different software.
- Riing Trio fans might not be supported because of the 3 zone settings.
- The pumps might use SetPwm instead of SetSpeed but the rgb should work fine.

Please include "Controllers" section from "Show Info" option if you report a device working/non-working. Make sure the service or the official software is not running while you run "Show Info".

| Device                      | Support            | Confirmed          |
|-----------------------------|--------------------|--------------------|
| Riing Plus 12/14/20         | :heavy_check_mark: | :heavy_check_mark: |
| Pure Plus 12/14             | :heavy_check_mark: | :x:                |
| Floe Riing RGB 240/280/360  | :heavy_check_mark: | :x:                |
| WaterRam RGB                | :heavy_check_mark: | :x:                |
| Pacific PR22-D5 Plus        | :heavy_check_mark: | :x:                |
| Pacific CL360/RL360         | :heavy_check_mark: | :x:                |
| Pacific W4/W5/W6            | :heavy_check_mark: | :x:                |
| Pacific V-GTX/V-RTX         | :heavy_check_mark: | :x:                |
| Pacific Lumi/Rad/R1/Fitting | :heavy_check_mark: | :x:                |
| Riing Trio 12/14            | :x:                | :x:                |
| Toughpower iRGB PLUS        | :x:                | :heavy_check_mark: |
| TT Premium X1               | :x:                | :heavy_check_mark: |
| Nemesis/Iris                | :x:                | :heavy_check_mark: |

xShojirox, 2019-01-22:
I haven't tried figuring out how to run this since I found this repository several weeks ago, and I wasn't successful then, and I know you have had several updates since. I don't have time right this second, but if you can walk me through installing it properly, I can help you test the Pacific W4, Riing Trio 12/14, Pacific V-RTX, and Pacific VR22-D5 Plus.

devcompl, 2019-01-22:
@xShojirox 
Ok I'll help you with the setup but first please run the "Dump info" option and post the output on pastebin or similar. You can find it by just running the exe. Make sure you are not running the official software while you do this.

xShojirox, 2019-01-22:
I will need a few days, sorry. My computer last night decided it wanted to try to kill itself. One of the fittings apparently came loose and let go, spraying coolant inside my computer. So now I have to take a few days to make sure all the components are dried out before I can try putting it back together again and assessing the damage.

I will keep you updated when I can. 

Also you said you needed the dump info from the exe, but I don't think there is one on github. Before the event last night when I was trying to mess around with your program, I did run the project file and do the dump, unfortunately i didn't save the info cuz I thought I could just run it later.

devcompl, 2019-01-22:
Oof, no worries take your time, I hope your pc is ok.

> Also you said you needed the dump info from the exe, but I don't think there is one on github. Before the event last night when I was trying to mess around with your program, I did run the project file and do the dump, unfortunately i didn't save the info cuz I thought I could just run it later.

You can find latest exe in the [releases ](https://github.com/devcompl/TTController/releases) section on github.

xShojirox, 2019-01-25:
Put the computer together today, was able to get the dump you requested.

The first three should be 140mm Riing Trio fans, the next three should be 120mm Riing Trio fans, then the next ones are the Pacific VR22-D5 Plus, Pacific V-RTX, Pacific W4, and the last one is a fourth 140mm Riing Trio fan.

<details><summary>Dump</summary>

```
Controllers
-------------------------------
Name: Riing Trio Controller
VendorId: 9802
ProductId: 8501
Ports:
        Id: 1
        Stats:
                Speed: 30%
                RPM: 384 RPM
                Unknown: 255
        Identifier: [9802, 8501, 1]

        Id: 2
        Stats:
                Speed: 30%
                RPM: 345 RPM
                Unknown: 255
        Identifier: [9802, 8501, 2]

        Id: 3
        Stats:
                Speed: 30%
                RPM: 370 RPM
                Unknown: 255
        Identifier: [9802, 8501, 3]

        Id: 4
        Stats:
                Speed: 30%
                RPM: 329 RPM
                Unknown: 255
        Identifier: [9802, 8501, 4]

        Id: 5
        Stats:
                Speed: 30%
                RPM: 389 RPM
                Unknown: 255
        Identifier: [9802, 8501, 5]

Name: Riing Trio Controller
VendorId: 9802
ProductId: 8502
Ports:
        Id: 1
        Stats:
                Speed: 30%
                RPM: 480 RPM
                Unknown: 255
        Identifier: [9802, 8502, 1]

        Id: 2
        Stats:
                Speed: 30%
                RPM: 0 RPM
                Unknown: 255
        Identifier: [9802, 8502, 2]

        Id: 3
        Stats:
                Speed: 30%
                RPM: 0 RPM
                Unknown: 255
        Identifier: [9802, 8502, 3]

        Id: 4
        Stats:
                Speed: 30%
                RPM: 0 RPM
                Unknown: 255
        Identifier: [9802, 8502, 4]

        Id: 5
        Stats:
                Speed: 30%
                RPM: 500 RPM
                Unknown: 255
        Identifier: [9802, 8502, 5]

Sensors
-------------------------------
Intel Core i7-9700K:
        /intelcpu/0/temperature/0:
                Name: CPU Core #1
                Value: 49
        /intelcpu/0/temperature/1:
                Name: CPU Core #2
                Value: 48
        /intelcpu/0/temperature/2:
                Name: CPU Core #3
                Value: 44
        /intelcpu/0/temperature/3:
                Name: CPU Core #4
                Value: 46
        /intelcpu/0/temperature/4:
                Name: CPU Core #5
                Value: 48
        /intelcpu/0/temperature/5:
                Name: CPU Core #6
                Value: 45
        /intelcpu/0/temperature/6:
                Name: CPU Core #7
                Value: 44
        /intelcpu/0/temperature/7:
                Name: CPU Core #8
                Value: 46
        /intelcpu/0/temperature/8:
                Name: CPU Package
                Value: 48
NVIDIA GeForce RTX 2080:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 33
WDC WD6003FZBX-00K5WB0:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 35
WDC WD1002FAEX-00Z3A0:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 36
```

</p>
</details>

devcompl, 2019-01-25:
It seems like you cant control the speed of the pump, so it must be controlled by the motherboard.

So here is a config to test RGB only:
<details><summary>Pacific VR22-D5 Plus, Pacific V-RTX and Pacific W4</summary><p>

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "2e2f71f8-465a-4665-80c8-ce9b737ed63d",
      "Ports": [
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "SnakeEffect",
          "Config": {
            "Length": 1,
            "SnakeColor": [255, 16, 16],
            "BackgroundColor": [16, 16, 16],
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```
</p></details>

---
You could try testing the trio fans too by adding `[9802, 8502, 1],` to the "Ports" array, but I think they might be using different commands so I dont know what will happen.

Best what you could do is install USBlyzer, find device with this properties:
```
idVendor = 264Ah
idProduct = 2135h
```
and while the official software is running capture the data sent to the controller. This way I will be able to find the command they use to control Trio fans.

xShojirox, 2019-01-25:
The pump speed is controlled by a turntable switch on the bottom, from speed 1 to 5.

I will see if I can test the RGB and get the USB info you requested later today when I get home from work.

devcompl, 2019-01-25:
> The pump speed, at least for this model, is controlled by a physical turnable switch on the bottom, from speed 1 to 5.

Thats terrible, I have one from NZXT and you can fully control it from the software, it also by default varies the pump speed depending on cpu/fluid temperature.

If you get around capturing the data make sure you switch to a few rgb modes. Im mostly interested in the "Fully Lighted" one where you can change each led color separately.

xShojirox, 2019-01-25:
What capture settings do you want for capturing the USBlyzer data?

devcompl, 2019-01-25:
You can use the defaults. I think those are everything selected and Hot-plugged/entire stack/irp & sl unchecked.

starstreak007, 2019-01-28:
Hi!
I'm really hoping for a simple fix for my Flow Riing 240.  All I want it to do is work with my settings when I wake it from sleep or hibernate.  Currently I have the controller software setup to change the color via temp and of course, it runs the fans faster, the hotter the CPU gets.  What do I do when I run your software?  
Listed bellow is my dump if it helps.
[dump.txt](https://github.com/devcompl/TTController/files/2801556/dump.txt)

devcompl, 2019-01-28:
@starstreak007 
Try this.

<details><summary>config.json</summary><p>

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "0bead049-edef-445b-baf5-f8078db02fcc",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "TemperatureEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "StartColor": [16, 16, 16],
            "EndColor": [255, 16, 16],
            "StartTemperature": 60,
            "EndTemperature": 85,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "PortConfig": [
    {
      "Key": [9802, 8101, 1],
      "Value": {
        "LedCount": 6
      }
    }
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/8": 90
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```
</p></details>

starstreak007, 2019-02-01:
Ok, I tried it.  I did the following.  I ran TTController.Service.exe. It said: "Service already installed. Uninstall? [y, n]:" So I said yes.
I replaced the config.json with the one you posted for me.  I ran the service again, it asked if I wanted to install. I said yes.  It asked if I wanted to install as custom user, I said No.  
Now the LED lights flicker in all modes.  So I uninstalled it, and put back the default config.json.  Now it's back to normal.

BUT.  The TT RGB PLUS did an update.  it's now v1.2.61 F2.0.3.  And I noticed it won't save my profile.  I had it set to "temp".  But now when it loads the software, it sets it to "wave" but the LEDs are still the proper setting for TEMP.

devcompl, 2019-02-01:
What do you mean flicker in all modes? Did you run the service while the TT RGB PLUS was running? You cant run both.

starstreak007, 2019-02-01:
I really must've missed an instructions doc or something.  I did NOT know I was supposed to either use yours or the TT RGB PLUS software.  When I get home from work, I'll uninstall the TT RGB PLUS, redo your config file and see if it works.  maybe that's why I've had issues. lol.  

Btw, is there a documentation on how to get full use of this software you made?  I'm not used to how GitHub works, so maybe that's why I keep making these errors.

> What do you mean flicker in all modes? Did you run the service while the TT RGB PLUS was running? You cant run both.

devcompl, 2019-02-01:
> I really must've missed an instructions doc or something. I did NOT know I was supposed to either use yours or the TT RGB PLUS software. When I get home from work, I'll uninstall the TT RGB PLUS, redo your config file and see if it works. maybe that's why I've had issues. lol.

You dont have to uninstall, just make sure they are not running at the same time.

> Just making sure I do this correct, I'm going to uninstall the TT RGB PLUS, and your software. Then I'm going to replace the config.json with the updated version you posted for me in a previous reply. Then I run your software, choose 1 and install the software, but say NO to custom user (or do I say yes? Not sure what custom user does). And that should be it? The Thermaltake AIO will be in temperature mode, and sleep and hibernation should work correctly where the Thermaltake AIO water cooler will still have the settings your software set, right?

"Custom user" makes the service run as a user specified by you, instead of the default LocalService account. 
Also, you dont have to uninstall the service every time you want to change the config, open the services.msc dialog and stop the service there, after you made changes to the config just start it. I guess I could add an option to start/stop the service from the "Manage service" menu.

> Btw, is there a documentation on how to get full use of this software you made? I'm not used to how GitHub works, so maybe that's why I keep making these errors.

Yeah, the only documentation is on the main repo page. But it kinda assumes you know what you are doing. Its very WIP but it will get better.

BTW next time if you have any issues, please make a new issue [here](https://github.com/devcompl/TTController/issues/new) as this one is only supposed to be about confirming device support, not help with the service itself.

starstreak007, 2019-02-03:
Thanks for all the info.  I came across once when I woke up the computer from sleep, it looked like it went back to default mode, but I don't know what caused it.  I'll create a new issue report when I can duplicate the issue and not post here.  Thanks again

devcompl, 2019-02-15:
@starstreak007 
Can I mark Floe Riing as confirmed working? Any problems with rgb effects?

@xShojirox 
Did you manage to get the data for the riing trio?

starstreak007, 2019-02-15:
Yes/no. It’s working but when I see the rings go red, sometimes I hear the fans spin up. Other times they don’t come on, but I don’t have a temp meter running to tell me if the cpu is really hot, or not hot enough to make the fans run at full speed. And it happened 2x where the computer was in sleep, when it woke up, it was back at default settings with the rainbow lights. Reboot of the computer brought back the settings. So I dunno what caused it so I didn’t mention it in a new forum.

devcompl, 2019-02-15:
> Yes/no. It’s working but when I see the rings go red, sometimes I hear the fans spin up. Other times they don’t come on, but I don’t have a temp meter running to tell me if the cpu is really hot, or [...]

Ok, so if the rgb is working on the pump im marking Floe as supported.

You might want to tweak the values of the config. Since the range of temperatures I've sent you starts from 60 degrees you might not see it change at all if you have good cooling.
Tweak "StartTemperature", "EndTemperature" and "CurvePoints" to match your desired temperature range.

I dont know why would wake from sleep not work, but it seems to work 100% of times on my machine. Again, if this happens more often, please make a separate issue. 
Btw you dont have to reboot, you can just restart the service.

alialooooosh055455, 2019-03-03:
@devcompl

jlaineinat0r, 2019-04-15:
I can confirm the PR22-D5 plus & the Pacific W4 are working as intended - they're obeying reverse & rotation offset settings and all the effects (temperature, ripple & flow) that I've tried to date.

And I wanted to add a thank you for your work, this provided me a controller with a light footprint that doesn't constantly crash on me like the official TT software does.

rickboy, 2019-04-16:
I dumped the Riing Trio, this is the first time I have tried to do this (dumping or reading output from Wireshark), so let me know if you need anything else.

From what I can see TT RGB Plus software is pushing two chunks of data to the USB controller for each fan, so I changed the colours of each ring to be different so they were easier to find:

ring 1 (leds 0 to 11) - white
ring 2 (leds 11 to 23) - blue (bluish)
ring 3 (leds 24 - 29) - red (reddish)

Again, let me know if you need anything else run.

<details><summary>Riing Trio config</summary>
<p>

[riing trio dump.zip](https://github.com/MoshiMoshi0/TTController/files/3084586/riing.trio.dump.zip)
[riing trio dump.zip](https://github.com/MoshiMoshi0/TTController/files/3084590/riing.trio.dump.zip)

```Name: Riing Trio Controller
VendorId: 9802
ProductId: 8502
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 30, Rpm: 532, Unknown: 255]
        Identifier: [9802, 8502, 1]

        Id: 2
        Data: [PortId: 2, Speed: 30, Rpm: 522, Unknown: 255]
        Identifier: [9802, 8502, 2]

        Id: 3
        Data: [PortId: 3, Speed: 30, Rpm: 544, Unknown: 255]
        Identifier: [9802, 8502, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8502, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8502, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full
```
</p></details>
[riing trio dump.zip](https://github.com/MoshiMoshi0/TTController/files/3084596/riing.trio.dump.zip)

MoshiMoshi0, 2019-04-16:
Nice!

You are correct, it seems the format is:
```[0x32, 0x52, port, 0x24, 0x03, chunk, 0x00, colors]```
and its sent in 2 chunks (19+11 colors).

I wonder if the 0x03 and 0x00 have any meaning, maybe its saying that the chunk id will be between 0 and 3?

I'll make a test release and post it here.

MoshiMoshi0, 2019-04-16:
Ok, please try the attached build with this config:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "50e98767-5be5-4b45-a301-bd5082e90d79",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "FlowEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfig": [
    {
      "Key": [9802, 8502, 1],
      "Value": {
        "LedCount": 30
      }
    },
    {
      "Key": [9802, 8502, 2],
      "Value": {
        "LedCount": 30
      }
    },
    {
      "Key": [9802, 8502, 3],
      "Value": {
        "LedCount": 30
      }
    }
  ]
}
```

[TTController.Service.exe.zip](https://github.com/MoshiMoshi0/TTController/files/3085098/TTController.Service.exe.zip)

rickboy, 2019-04-17:
That worked very nicely.

Nitserof, 2019-06-29:
I have the iRGB PLUS and the Riing Duo.
In my PC, speed control is worked for the Riing Duo, but not worked for the iRGB PLUS.
also, LED control is worked for the iRGB PLUS, but not worked for the Riing Duo.

Here is the dump files.
[iRGB_PULS_and_Riing_Duo.zip](https://github.com/MoshiMoshi0/TTController/files/3341777/iRGB_PULS_and_Riing_Duo.zip)

MoshiMoshi0, 2019-06-29:
@Nitserof thanks for the data. 

Try this build:
[TTController_0.7.2.0-alpha.3.b411a626f32c762c8186e745608bde49ffe3db32.zip](https://github.com/MoshiMoshi0/TTController/files/3341964/TTController_0.7.2.0-alpha.3.b411a626f32c762c8186e745608bde49ffe3db32.zip)

This should fix RGB for Riing Duo. The speed for iRGB is implemented but maybe its not supported? The app always uses fan presets "Silent" and "Performance". It does not set the speed manually like TTController.

BTW can you post your config?

Nitserof, 2019-06-30:
Sorry, the fan speed for iRGB **is working** as intended. I was not checking well.

I tried that build with the simple config below, but the color of Duo is still not changing...
(Official TT RGB Plus is working.)

<details>
<summary>My config</summary>

```json
{
  "Profiles": [
    {
      "Name": "Duo",
      "Guid": "ee6ec00d-f643-4a37-a19e-8db7c1437c12",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 100,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 0, 0],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "iRGB",
      "Guid": "49a658cf-b53f-4a75-ad66-9de71a2ddde8",
      "Ports": [
        [9802, 9001]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 100,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 0, 0],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfigs": [],
  "SensorConfigs": []
}
```

</details>
<details>
<summary>Debug report</summary>

```
Controllers
-------------------------------
Name: Dpsg
VendorId: 9802
ProductId: 9001
Ports:
        Id: 0
        Data: [, VIN: E6F8, VVOut12: FFD2, VVout5: 44D1, VVOut33: D5D0, VIOut12: 0DF0, VIOut5: 40E0, VIOut33: 06E0, Temp: 9CF0, FanSpeed: 8F10, Report: ]
        Identifier: [9802, 9001]

Available effect types:
ByLed, Full

Name: Riing Trio
VendorId: 9802
ProductId: 8501
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 0, Rpm: 522, Unknown: 255]
        Identifier: [9802, 8501, 1]

        Id: 2
        Data: [PortId: 2, Speed: 0, Rpm: 521, Unknown: 255]
        Identifier: [9802, 8501, 2]

        Id: 3
        Data: [PortId: 3, Speed: 0, Rpm: 517, Unknown: 255]
        Identifier: [9802, 8501, 3]

        Id: 4
        Data: [PortId: 4, Speed: 40, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8501, 4]

        Id: 5
        Data: [PortId: 5, Speed: 40, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8501, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

-------------------------------
Sensors
-------------------------------
Intel Core i7-8700:
        Temperature
                CPU Core #1 (/intelcpu/0/temperature/0): 45.0 °C
                CPU Core #2 (/intelcpu/0/temperature/1): 45.0 °C
                CPU Core #3 (/intelcpu/0/temperature/2): 45.0 °C
                CPU Core #4 (/intelcpu/0/temperature/3): 43.0 °C
                CPU Core #5 (/intelcpu/0/temperature/4): 49.0 °C
                CPU Core #6 (/intelcpu/0/temperature/5): 43.0 °C
                CPU Package (/intelcpu/0/temperature/6): 49.0 °C
NVIDIA GeForce GTX 1070:
        Temperature
                GPU Core (/nvidiagpu/0/temperature/0): 40.0 °C
SanDisk SDSSDH3 500G:
        Temperature
                Temperature (/hdd/0/temperature/0): 32.0 °C
ST2000DM001-1CH164:
        Temperature
                Temperature (/hdd/1/temperature/0): 37.0 °C
-------------------------------
Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll
-------------------------------
Press any key to continue...
 
```

</details>

MoshiMoshi0, 2019-06-30:
You have to add a PortConfig because by default TTController assumes 12 leds per fan.

```json

  "PortConfigs": [
    {
      "Ports": [[9802, 8501, 1], [9802, 8501, 2], [9802, 8501, 3]],
      "Config": {
        "LedCount": 18
      }
    }
  ]
```

This will change how the colors are sent to the controller and they should work. 
However I think you have to use a "ByLed" effect type but FullColorEffect uses "Full", meaning there is only one color received from that effect instead of one color per led. I think you will only see one led light up.

Try another effect like:
```json
        {
          "Type": "FlowEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```
This should work correctly.
I will add an option to force effects that use "Full" effect type to return one color per led.
I guess you could also try adding `"LedCountHandling": "Copy"` to the above PortConfig, this will copy the color returned from `FullColorEffect` to match the number of led on the fan.

Thanks for checking iRGB again, I should add a special speed controller for iRGB where you can also choose the Silent and Performance presets.

Nitserof, 2019-06-30:
I had a misunderstanding about "FullColorEffect".
Thanks for your advice, it looks worked!

MoshiMoshi0, 2019-06-30:
Ok nice, so iRGB data values are only thing left.

"FullColorEffect" does set the whole fan to the specified color, its just that in Riing Trio controller you have to send colors for each led. Other controllers have a specific command where you only pass one color and its applied to all leds.

MoshiMoshi0, 2019-06-30:
@Nitserof Try this build:
[TTController_0.7.2.0-alpha.9.a6fb65a3183a6ee21ec9ac959f768a0a51656f3c.zip](https://github.com/MoshiMoshi0/TTController/files/3343363/TTController_0.7.2.0-alpha.9.a6fb65a3183a6ee21ec9ac959f768a0a51656f3c.zip)

This should now show correct iRGB values. Please run it in console mode for a while and post the output.

Nitserof, 2019-07-01:
I tried it, then it raised an exception.

<details>
<summary>Output</summary>

```
19:30:42.4782: ================================================================
19:30:42.5001: Initializing...
19:30:42.7719: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
19:30:42.7839: Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
19:30:42.7839: Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
19:30:42.7908: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
19:30:42.7908: Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
19:30:42.7908: Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
19:30:42.7908: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.7.0.0]
19:30:42.8048: Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
19:30:42.8048: Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
19:30:42.8208: Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
19:30:42.8208: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
19:30:42.8208: Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
19:30:42.8397: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
19:30:42.8397: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.7.0.0]
19:30:42.8397: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
19:30:42.8397: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
19:30:42.8567: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
19:30:42.8567: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
19:30:42.8567: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
19:30:42.8666: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
19:30:42.8666: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
19:30:42.8666: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
19:30:42.8666: Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
19:30:42.8826: Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
19:30:42.8826: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
19:30:42.8826: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
19:30:42.8986: Creating Config Manager...
19:30:42.9471: Loading config...
19:30:43.0748: Creating DataCache...
19:30:43.0748: Creating Sensor Manager...
19:30:43.0858: Initializing Open Hardware Monitor...
19:30:44.0072: Detected 42 sensors
19:30:44.0072: Creating Effect Manager...
19:30:44.0072: Creating Speed Controller Manager...
19:30:44.0072: Creating Device Manager...
19:30:44.0432: Searching for "Dpsg" controllers
19:30:44.0651: Initialized "Dpsg" controller [0x264A, 0x2329]
19:30:44.0651: Searching for "Riing" controllers
19:30:44.0801: Searching for "Riing Plus" controllers
19:30:44.0871: Searching for "Riing Trio" controllers
19:30:44.0871: Initialized "Riing Trio" controller [0x264A, 0x2135]
19:30:44.1030: Adding speed controller: DpsgSpeedController [49a658cf-b53f-4a75-ad66-9de71a2ddde8]
19:30:44.1030: Creating Timer Manager...
19:30:44.1030: Starting 4 timers...
19:30:44.1170: Initializing done!
19:30:44.1170: ================================================================

Unhandled Exception: System.FormatException: Could not find any recognizable digits.
   at System.ParseNumbers.StringToInt(String s, Int32 radix, Int32 flags, Int32* currPos)
   at System.Convert.ToInt32(String value, Int32 fromBase)
   at TTController.Plugin.DpsgController.DpsgControllerProxy.<GetPortData>g__GetData|8_0(Byte b) at H:\.projects\csharp\TTController\Plugins\Devices\TTController.Plugin.DpsgController\DpsgControllerProxy.cs:line 70
   at TTController.Plugin.DpsgController.DpsgControllerProxy.GetPortData(Byte port) at H:\.projects\csharp\TTController\Plugins\Devices\TTController.Plugin.DpsgController\DpsgControllerProxy.cs:line 86
   at TTController.Service.TTService.DeviceSpeedTimerCallback() at H:\.projects\csharp\TTController\Source\TTController.Service\TTService.cs:line 285
   at TTController.Service.Manager.TimerManager.<>c__DisplayClass3_0.<RegisterTimer>b__0(Object sender, TimerElapsedEventArgs args) at H:\.projects\csharp\TTController\Source\TTController.Service\Manager\TimerManager.cs:line 37
   at TTController.Service.Manager.TimerManager.Timer.ExecuteTimer() at H:\.projects\csharp\TTController\Source\TTController.Service\Manager\TimerManager.cs:line 243
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()
 
```

</details>

MoshiMoshi0, 2019-07-01:
Whoops sorry about that.
This should fix it:
[TTController_0.7.2.0-alpha.9.a6fb65a3183a6ee21ec9ac959f768a0a51656f3c.zip](https://github.com/MoshiMoshi0/TTController/files/3345449/TTController_0.7.2.0-alpha.9.a6fb65a3183a6ee21ec9ac959f768a0a51656f3c.zip)

Nitserof, 2019-07-01:
<details>
<summary>Ok, output is here.</summary>

```
21:22:29.1738: ================================================================
21:22:29.1957: Initializing...
21:22:29.5099: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
21:22:29.5308: Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
21:22:29.5438: Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
21:22:29.5568: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
21:22:29.5697: Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
21:22:29.5697: Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
21:22:29.5966: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.7.0.0]
21:22:29.6096: Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
21:22:29.6236: Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
21:22:29.6665: Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
21:22:29.6804: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
21:22:29.6804: Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
21:22:29.8191: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
21:22:29.8230: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.7.0.0]
21:22:29.8230: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
21:22:29.8230: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
21:22:29.8230: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
21:22:29.8380: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
21:22:29.8380: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
21:22:29.8380: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
21:22:29.8380: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
21:22:29.8380: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
21:22:29.8560: Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
21:22:29.8560: Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
21:22:29.8560: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
21:22:29.8560: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
21:22:29.8769: Creating Config Manager...
21:22:29.9347: Loading config...
21:22:30.0594: Creating DataCache...
21:22:30.0594: Creating Sensor Manager...
21:22:30.0704: Initializing Open Hardware Monitor...
21:22:30.9859: Detected 42 sensors
21:22:30.9859: Creating Effect Manager...
21:22:30.9859: Creating Speed Controller Manager...
21:22:31.0079: Creating Device Manager...
21:22:31.0368: Searching for "Dpsg" controllers
21:22:31.0727: Initialized "Dpsg" controller [0x264A, 0x2329]
21:22:31.0727: Searching for "Riing" controllers
21:22:31.0876: Searching for "Riing Plus" controllers
21:22:31.0876: Searching for "Riing Trio" controllers
21:22:31.1047: Initialized "Riing Trio" controller [0x264A, 0x2135]
21:22:31.1047: Adding speed controller: DpsgSpeedController [49a658cf-b53f-4a75-ad66-9de71a2ddde8]
21:22:31.1275: Applying computer state profile: Boot
21:22:31.1275: Creating Timer Manager...
21:22:31.1345: Starting 4 timers...
21:22:31.1345: Initializing done!
21:22:31.1345: ================================================================
21:22:36.1401: Port [9802, 9001] data: [Rpm: 572, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 3, VIOut5: 3.9375, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:22:41.1390: Port [9802, 9001] data: [Rpm: 552, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 2.5, VIOut5: 3.875, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:22:46.1396: Port [9802, 9001] data: [Rpm: 552, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 2.5, VIOut5: 3.9375, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:22:51.1392: Port [9802, 9001] data: [Rpm: 560, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 2.5, VIOut5: 3.9375, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:22:56.1399: Port [9802, 9001] data: [Rpm: 560, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 3, VIOut5: 4, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:23:01.1396: Port [9802, 9001] data: [Rpm: 560, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 2.5, VIOut5: 3.9375, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:23:06.1394: Port [9802, 9001] data: [Rpm: 552, Temperature: 38.5, VIN: 115, VVOut12: 11.98438, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 3.5, VIOut5: 4, VIOut33: 0.375, WATTS: 1.248047, EFF: 6]
21:23:11.1390: Port [9802, 9001] data: [Rpm: 556, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 3, VIOut5: 4, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:23:16.1396: Port [9802, 9001] data: [Rpm: 576, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 2.75, VIOut5: 3.9375, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
21:23:21.1402: Port [9802, 9001] data: [Rpm: 560, Temperature: 38.5, VIN: 115, VVOut12: 12, VVOut5: 5.0625, VVOut33: 3.328125, VIOut12: 3, VIOut5: 4, VIOut33: 0.375, WATTS: 1.248047, EFF: 5]
[... cut ...]
21:24:33.9447: Finalizing Open Hardware Monitor...
21:24:33.9447: Disposing Device Manager...
21:24:33.9447: Disposed devices: 2
21:24:33.9447: Disposing Effect Manager...
21:24:33.9447: Disposed effects: 0
21:24:33.9447: Disposing Speed Controller Manager...
21:24:33.9617: Disposed speed controllers: 1
21:24:33.9617: Disposing Config Manager...
21:24:33.9617: Finalizing done!
21:24:33.9617: ================================================================
Press any key to return to the menu...
 
```
</details>

MoshiMoshi0, 2019-07-01:
Nice, the data seems correct imo. But I have to tweak watts and efficiency computation.

I'll also see if its possible to get the fan speed in % so we can use PwmSpeedController, if not I guess I could map rpm to speed.

Thank you for your help!

MoshiMoshi0, 2020-10-07:
I'm closing this and splitting remaining/future not implemented devices to separate issues for easier tracking.
Thank you again to everyone that helped implementing new devices.

closed 2020-10-07 by MoshiMoshi0