MoshiMoshi0/TTController #2: how to edit config for 3.0

starstreak007, 2019-02-18:
Ok, so when you asked me about the software working/not working, I was using an older revision.  I uninstalled the old.  And I downloaded version 3.0.0.
When I ran it, it looked like it started up with my configuration.  The one that would startup all the LEDs as blue, and as temperature went up, it would go to red and spin the fan up as temp got hotter.
So I haven't seen the software NOT work after waking computer up from sleep since I updated.

My question, where is the stored config file for the TTController_0.3.0.0-dev?  In version 2.1, the file "config.json" was in the main directory.  But it's not in version 3.0.

alialooooosh055455, 2019-02-19:
Ali

devcompl, 2019-02-19:
How did you update the service? The correct way would be to stop the service, remove all files from the directory where it is located (but leave config.json of course), unpack the newest version in the same directory and start the service.
You dont have to uninstall it to upgrade since the "installation" is just an entry in windows registry that just specifies where a service exe is located.

The config should still be located in the same folder as the service exe. Are you sure you started the v0.3.0.0 and not the old version?

closed 2019-03-22 by devcompl