MoshiMoshi0/TTController #3: Need help configurating 3 Riing Trio 12 fans

Melvasul, 2019-03-02:
**Describe your desired configuration**
what I want as a configuration is this step:
CPU temp 15° - Fan 0%
CPU temp 30° - Fan 20%
CPU temp 40° - Fan 40%
CPU temp 50° - Fan 50%
CPU temp 60° - Fan 75%
CPU temp 65° - Fan 90%
CPU temp 70° - Fan 100%

**Hardware info**
AMD Ryzen 7 1700:
        /amdcpu/0/temperature/0:
                Name: Core (Tctl)
                Value: 33,5
        /amdcpu/0/temperature/1:
                Name: Core (Tdie)
                Value: 33,5
Radeon (TM) RX 480 Graphics:
        /atigpu/0/temperature/0:
                Name: GPU Core
                Value: 31
Samsung SSD 850 EVO 500GB:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 26
Samsung SSD 840 PRO Series:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 31

**Current configuration**
```json
{
  "Profiles": [],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {},
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 33
}
```
It's the default one, I tried looking at the example you provided but I was not really good at setting it

[spoiler]```
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [ 9802, 8101, 1 ],
        [ 9802, 8101, 2 ],
        [ 9802, 8101, 3 ]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 50,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Ripple",
            "Speed": "Normal",
            "Colors": [
              [255, 0, 0]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "PortConfig": [],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/8": 90
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```[/spoiler]

devcompl, 2019-03-02:
Your hardware info does not contain any controllers, make sure you do not have the official software running when you generate the info or when you are running the service. Also you did not say how much fans/controllers do you have.

I can create a guessed config that you can try, but since I still don't have have full knowledge how leds on Riing Trio fans are controlled (they have 3 seperate zones), the profile will not change rgb effects:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": []
    }
  ],
  "ComputerStateProfiles": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 70
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

As a workaround you could configure the rgb in the official software first (if the effect does not require the software to be running) and then switch to the service for speed control.

Also there are no in between speeds between 0% and 20%, so in your config the fan will only stop if the temperature is 15° or below, otherwise it will run at 20% below 30°.

Melvasul, 2019-03-02:
Sorry I thought I wrote everything
This is the result without the software running
3 fans with 1 controller

I do not really care about the rgb shit, I bought them just because I had a nice discount on them.

Sadly no, the RGB control requires the software to be running.

> Name: Riing Plus Controller
> VendorId: 9802
> ProductId: 8101
> Ports:
>         Id: 1
>         Stats:
>                 Speed: 30%
>                 RPM: 491 RPM
>                 Unknown: 255
>         Identifier: [9802, 8101, 1]
> 
>         Id: 2
>         Stats:
>                 Speed: 30%
>                 RPM: 510 RPM
>                 Unknown: 255
>         Identifier: [9802, 8101, 2]
> 
>         Id: 3
>         Stats:
>                 Speed: 30%
>                 RPM: 539 RPM
>                 Unknown: 255
>         Identifier: [9802, 8101, 3]
> 
>         Id: 4
>         Stats:
>                 Speed: 30%
>                 RPM: 0 RPM
>                 Unknown: 255
>         Identifier: [9802, 8101, 4]
> 
>         Id: 5
>         Stats:
>                 Speed: 30%
>                 RPM: 0 RPM
>                 Unknown: 255
>         Identifier: [9802, 8101, 5]
> 
> Sensors
> -------------------------------
> AMD Ryzen 7 1700:
>         /amdcpu/0/temperature/0:
>                 Name: Core (Tctl)
>                 Value: 37,25
>         /amdcpu/0/temperature/1:
>                 Name: Core (Tdie)
>                 Value: 37,25
> Radeon (TM) RX 480 Graphics:
>         /atigpu/0/temperature/0:
>                 Name: GPU Core
>                 Value: 35
> Samsung SSD 850 EVO 500GB:
>         /hdd/0/temperature/0:
>                 Name: Temperature
>                 Value: 30
> Samsung SSD 840 PRO Series:
>         /hdd/1/temperature/0:
>                 Name: Temperature
>                 Value: 31

devcompl, 2019-03-02:
Ok, so the only thing to change is the ports array, but I also changed the rgb update interval to 5s since you will not be using any effects anyway. It will slightly (if at all) lower the cpu usage since it wont have to update 30 times per second. I'll add a feature so interval of 0 means off.

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": []
    }
  ],
  "ComputerStateProfiles": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 70
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 5000
}
```

BTW, the detected controller is a Riing Plus controller and not Riing Trio controller. Riing Trio fans have a different controller because of the 3 rgb zones. Thermaltake says you have to use Riing Trio controller with Riing Trio fans, but I assume its only because of the rgb, speed control will work the same.

Melvasul, 2019-03-03:
Thanks for the configuration, is there a way to check the real-time speed?
I've found no way to read them through your software or HWInfo(e.g.)

> BTW, the detected controller is a Riing Plus controller and not Riing Trio controller. Riing Trio fans have a different controller because of the 3 rgb zones. Thermaltake says you have to use Riing [...]

My fans are the
"Riing Plus 12 RGB Radiator Fan TT Premium Edition (3 Fan Pack)"
Maybe something is different?

> Interesting, you mean each effect you select in the official software stops when you exit it? Because I have Riing Trio fans and there are some effects that are built into the controller so you dont need the software running.
I mostly tested few RGB effects like 
-) static colour
-) pulsing colour

all of them restored to the basic colour cycling mode.

alialooooosh055455, 2019-03-03:
@Melvasul

devcompl, 2019-03-03:
> Thanks for the configuration, is there a way to check the real-time speed?
> I've found no way to read them through your software or HWInfo(e.g.)

For now I guess you can only use the show hardware info option, but that requires the service to not be running. I'll add proper logging so you can see everything when you run it in console mode.

> My fans are the
> "Riing Plus 12 RGB Radiator Fan TT Premium Edition (3 Fan Pack)"
> Maybe something is different?

Oh, so you have Riing Plus fans, I though you have Riing Trio because of the issue title. So if you want I could add a rgb effect to the fans, like static color or maybe temperature based color. Also you can have a separate boot and shutdown profile, so when you boot/shutdown you can have different fan speed and effects.

> Interesting, you mean each effect you select in the official software stops when you exit it? Because I have Riing Trio fans and there are some effects that are built into the controller so you dont need the software running.
> all of them restored to the basic colour cycling mode.

There is a couple of effects that do not require the app to be running, but they reset to the default when you restart the computer. You can change those defaults and that is what I call boot profile.

Melvasul, 2019-03-03:
> edit: I pushed v0.4.0.0-dev which includes basic logging.

Perfect I will try it asap 👍 

  
> Also you can have a separate boot and shutdown profile, so when you boot/shutdown you can have different fan speed and effects.

Why should I be interested in having a shutdown profile? Shouldn't I be interested only in a boot one?

Ok understood, I'm only interested in a fading effect with colour blue(pure blue)

devcompl, 2019-03-03:
> The only thing I would add is to put the menu in a loop with a 0 option to exit from the TTController.Service.exe file

Yeah, the menu needs a redesign.

> And a "pause" command to make the window not close immediately after it has shown the info

There is a pause after the show hardware info option, it waits for user input and only then exits. 

> Why should I be interested in having a shutdown profile? Shouldn't I be interested only in a boot one?

You could have a separate rgb effect for boot and shutdown. It also sets a static speed on the fans while the service is not running and there is no control of the fans. So for example when you stop the service it will trigger the shutdown profile and set the fans to 50%.

> I'm only interested in a fading effect with colour blue(pure blue)

Ok, added a blue fading effect, boot profile with green fading effect and 35% fan speed, and shutdown profile with red fading effect and 35% fan speed. You should be able to tweak/remove what you want.
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Pulse",
            "Speed": "Slow",
            "Colors": [
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 35,
      "EffectType": "Pulse",
      "EffectSpeed": "Slow",
      "EffectColors": [
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0]
      ]
    },
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 35,
      "EffectType": "Pulse",
      "EffectSpeed": "Slow",
      "EffectColors": [
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0]
      ]
    }
  ],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 70
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 5000
}
```

Melvasul, 2019-03-03:
Everything works but I found something interesting:
The first and second configuration you gave me did not behave as expected but instead, they were statically at 60%(the cpu temp reader was correct though)

Thanks for everything ^^

devcompl, 2019-03-03:
The first config was not working because the port identifiers where for made for Riing Trio controller (`[9802, 8501, 1]` instead of `[9802, 8101, 1]`).

The second one should be working 100% but without controlling the rgb effects. Maybe you did not restart the service after you changed the config? 

Anyway, glad it is working now. Closing.

closed 2019-03-03 by devcompl

alialooooosh055455, 2019-03-04:
@Melvasul

AlphaDE, 2019-04-10:
Hi,

with the current version, when I copy&paste exactly the last config, having the same ports and adapting to my hardware (intelcpu instead of amdcpu)

Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 35, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 35, Rpm: 616, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 35, Rpm: 620, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
Intel Core i7-4790K:
        /intelcpu/0/temperature/0:
                Name: CPU Core #1
                Value: 32
Intel Core i7-4790K:
        /intelcpu/0/temperature/1:
                Name: CPU Core #2
                Value: 34
Intel Core i7-4790K:
        /intelcpu/0/temperature/2:
                Name: CPU Core #3
                Value: 32
Intel Core i7-4790K:
        /intelcpu/0/temperature/3:
                Name: CPU Core #4
                Value: 32
Intel Core i7-4790K:
        /intelcpu/0/temperature/4:
                Name: CPU Package
                Value: 34
NVIDIA GeForce GTX 1080 Ti:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 40
WDC WD20EFRX-68EUZN0:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 28
-------------------------------

the program crashes:

17:26:41.8214: ================================================================
17:26:41.8324: Initializing...
17:26:42.0394: Loading plugins...
[... 36 lines cut ...]
   bei TTController.Service.Config.Converter.AbstractPluginConverter`2.ReadJson(JsonReader reader, Type objectType, TPlugin existingValue, Boolean hasExistingValue, JsonSerializer serializer)
   bei Newtonsoft.Json.JsonConverter`1.ReadJson(JsonReader reader, Type objectType, Object existingValue, JsonSerializer serializer)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
[... cut ...]
   bei Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   bei Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   bei Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   bei TTController.Service.Manager.ConfigManager.LoadOrCreateConfig()
   bei TTController.Service.TTService.Initialize()
   bei TTController.Service.Program.<>c.<Main>b__2_1()
   bei TTController.Service.Program.Main(String[] args)

AlphaDE, 2019-04-10:
Are there any rules to create the GUID?

Melvasul, 2019-05-14:
I used to use V 0.4, now we are at 0.7, how can we translate the same things we did with the previous version to the new one?
I looked into the repository and I've not found the answer to this :/

MoshiMoshi0, 2019-05-14:
You have to look at the main page for current config layout. While I know its not explained the best, it should help you make the changes you want. 

Here is the corrected config assuming the last one did not change:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "PulseEffect",
          "Config": {
            "BrightnessStep": 0.025,
            "Colors": [
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 35,
      "EffectType": "Pulse_Slow",
      "EffectColors": [
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0],
        [0, 255, 0]
      ]
    },
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 35,
      "EffectType": "Pulse_Slow",
      "EffectColors": [
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0],
        [255, 0, 0]
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 70
      }
    }
  ]
}
```

Switched from old `DefaultEffect` to custom `PulseEffect`. It should look a little nicer and give you more control over the speed with `BrightnessStep`