MoshiMoshi0/TTController #4: Service doesn't start after updating.

Zeonymous, 2019-03-04:
Okay.

So, I'd come to figure out my issue with the pump. Easy enough, right?

Not so much.

My issue is as follow, and I'll admit to my own stupidity beforehand.
I'd tried to install the later version, from a previous one.
I'd deleted everything but the .config file, with the service stopped, I believe, but I ran into a bit of a snag;
When I restarted the service window, TTController.Service.exe hangs up on "Start" after installing.

[1]     Manage Service
[2]     Run in console
[3]     Show hardware info
[1, 2, 3]: 1

-----------------

[1]     Start
[2]     Uninstall
[1, 2]: 1

Then nothing happens. It hangs up, and the service does not run, nor is it able to run. I'm not sure what I'd done wrong, but this is the second time coming to the same conclusion.

My steps taken were to go into the regedit console, and delete the TTController service.

This allowed for a fresh install, but it stops at the same spot, still.

Call me a fool, but I felt like this was the only way to get it started without spilling a bunch of nonsense here, which I seem to have ended up doing anyway, haha.

I can speak volumes to this being a more reliable service than the previous "Official" service, at least in regards to the RGB not locking up, and any time I started the GUI, it seemed as though the service was running fine, until I messed with things.

Additional info:
When attempting to start the service from Services, the error reads:

"Windows could not start the Thermaltake Controller service on Local Computer.
Error 1064: An exception occurred in the service when handling the control request."

The service was previously not "Local" and was my user's login information. I'm not sure if this is the same issue I had where I was 'forced' to jump through hoops because of the PIN. I'll try to work back to where it was working previously, in the menetime. I'd kept the previous version's files.

This way I could at least get the proper login permissions, since deleting the Controler Service may have messed that up a bit.

Zeonymous, 2019-03-04:
Alright, I'm an idiot, as usual.. Lol.

So, the config file was the problem. I installed from the previous version (Non Custom User) so this time it installed locally.
(This would have saved me a big headache some time ago)
Anyway, the config file was what was causing the hangup, maybe a password was changed, or something had been messed up with the permissions of the file, I'm not sure, really. This is all new territory to me.

What I do know is, my previous config file (No big deal losing it, there was not much in the file to begin with) was what was causing the hangup issues.

I've since reinstalled, and the GUI is functioning properly now.

Crisis adverted. I also learned something today.

Zeonymous, 2019-03-04:
Okay, I've gotten the service working, it seems.

However, I don't know how to code. More or less, from the config examples, I don't exactly understand what goes where.

Any kind of crash-course in git coding you have to offer to help better understand how to get into all of this?

Here was my very simple previous config, which was used with 1.7 IIRC.

{
  "ProfileMap": [
    {
      "Key": {
        "Guid": "a4eadc2f-5c42-47cb-ac2f-64b7637f821a",
        "Name": "RGB Normal",
        "ProfileType": "Normal",
        "Points": [
          {
            "Temperature": 50.0,
            "Speed": 20
          },
          {
            "Temperature": 60.0,
            "Speed": 25
          },
          {
            "Temperature": 70.0,
            "Speed": 40
          },
          {
            "Temperature": 80.0,
            "Speed": 75
          },
          {
            "Temperature": 85.0,
            "Speed": 100
          }
        ],
        "SpeedMode": "Manual",
        "RgbMode": "Wave",
        "RgbSpeed": "Extreme",
        "RgbColors": [
          "#00FF1D",
          "#00FFB6",
          "#00E9FF",
          "#006EFF",
          "#000CFF",
          "#7700FF",
          "#D800FF",
          "#FF00B2",
          "#FF0010",
          "#FF4300",
          "#FFAE00",
          "#FFFA00"
        ],
        "ManualFanSpeed": 100,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureSensorIdentifiers": [],
        "TemperatureMixFunction": "Max"
      },
      "Value": [
        689174312,
        689174314,
        689174315,
        689174316,
        -831959983,
        -831959982,
        -831959981,
        -831959980,
        -831959979
      ]
    },
    {
      "Key": {
        "Guid": "197d1bda-8e1f-4718-8d4c-5b8981c7db08",
        "Name": "RGB Boot",
        "ProfileType": "Boot",
        "Points": [
          {
            "Temperature": 50.0,
            "Speed": 20
          },
          {
            "Temperature": 60.0,
            "Speed": 25
          },
          {
            "Temperature": 70.0,
            "Speed": 40
          },
          {
            "Temperature": 80.0,
            "Speed": 75
          },
          {
            "Temperature": 85.0,
            "Speed": 100
          }
        ],
        "SpeedMode": "Manual",
        "RgbMode": "Wave",
        "RgbSpeed": "Normal",
        "RgbColors": [
          "#00FF1D",
          "#00FFB6",
          "#00E9FF",
          "#006EFF",
          "#000CFF",
          "#7700FF",
          "#D800FF",
          "#FF00B2",
          "#FF0010",
          "#FF4300",
          "#FFAE00",
          "#FFFA00"
        ],
        "ManualFanSpeed": 100,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureMixFunction": "Max"
      },
      "Value": [
        689174312,
        689174314,
        689174315,
        689174316,
        -831959983,
        -831959982,
        -831959981,
        -831959980,
        -831959979
      ]
    },
    {
      "Key": {
        "Guid": "93d836b0-59c3-4388-ac2a-497268ebd599",
        "Name": "RGB Shutdown",
        "ProfileType": "Shutdown",
        "Points": [
          {
            "Temperature": 50.0,
            "Speed": 20
          },
          {
            "Temperature": 60.0,
            "Speed": 25
          },
          {
            "Temperature": 70.0,
            "Speed": 40
          },
          {
            "Temperature": 80.0,
            "Speed": 75
          },
          {
            "Temperature": 85.0,
            "Speed": 100
          }
        ],
        "SpeedMode": "Manual",
        "RgbMode": "Wave",
        "RgbSpeed": "Extreme",
        "RgbColors": [
          "#00FF1D",
          "#00FFB6",
          "#00E9FF",
          "#006EFF",
          "#000CFF",
          "#7700FF",
          "#D800FF",
          "#FF00B2",
          "#FF0010",
          "#FF4300",
          "#FFAE00",
          "#FFFA00"
        ],
        "ManualFanSpeed": 100,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureMixFunction": "Max"
      },
      "Value": [
        689174312,
        689174314,
        689174315,
        689174316,
        -831959983,
        -831959982,
        -831959981,
        -831959980,
        -831959979
      ]
    }
  ],
  "TemperatureTimerInterval": 250.0,
  "FanTimerInterval": 2500.0
}

One of controller 2's color were off. Was just attempting to mess about with the colors to see if I could get it to do what I wanted it to do before updating the service.

I wouldn't know where to start, in plugging these values. I'll do some research on it, but this will be my first time in attempts.

If you have any resources that would help explain to a total novice, this would be super helpful.

alialooooosh055455, 2019-03-04:
@Zeonymous

alialooooosh055455, 2019-03-04:
:grinning:

alialooooosh055455, 2019-03-04:
**
### **

alialooooosh055455, 2019-03-04:
****

alialooooosh055455, 2019-03-04:
__

alialooooosh055455, 2019-03-04:
@alialooooosh055455 
- ### **_~~:relaxed:~~ _ **

alialooooosh055455, 2019-03-04:
😀

Zeonymous, 2019-03-04:
Okay, so this is my first shot at this. Let me know if this is going to turn my computer into a fireball?

```
{
  "Profiles": [
  {
  "Name": "RGB Normal",
  "Guid": "a4eadc2f-5c42-47cb-ac2f-64b7637f821a",
  "Ports": ["<PortIdentifier>"
    [9802, 8101, 5],
    [9802, 8101, 4],
    [9802, 8101, 3],
	[9802, 8101, 2],
    [9802, 8101, 1],
  ],
}
  "SpeedControllers": [
  {
  "Type": "PwmSpeedController", 
  "Config": {
    "Speed": 50,
    "Trigger": {
      "AlwaysTrigger": {}
    }
  }
}
],
  "Effects": [
  {
  "Type": "Wave",
  "Config": {
    "Type": "Wave",
    "Speed": "Extreme",
    "Colors": [
    ["#00FF1D",
     "#00FFB6",
     "#00E9FF",
     "#006EFF",
     "#000CFF",
     "#7700FF",
     "#D800FF",
     "#FF00B2",
     "#FF0010",
     "#FF4300",
     "#FFAE00",
     "#FFFA00"]
  ]
    ],
    "Trigger": {
	"AlwaysTrigger": {}
	}
  }
}
]
}
],
  "ComputerStateProfiles": [Normal],
  "PortConfig": [{
  "Key": [
	[9802, 8101, 5],
	[9802, 8101, 4],
    [9802, 8101, 3],
	[9802, 8101, 2],
    [9802, 8101, 1],
]
  "Value": {
    "Name": "Controller1",
    "LedCount": 12,
    "LedRotation": 10,
    "LedReverse": false
  }
}],
  "CriticalTemperature": {90},
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 33
}
```

devcompl, 2019-03-04:
You can't translate the old config to the new one like that as its completely different. Tell me what you want and I'll try to make a config for you. 

Please attach the output of the Show hardware info option, with the service is off, and a description of what is connected to which port on the controllers.

BTW if the service hangs or crashes you can try stopping it and running it in console mode. It also has some logging for testing.

@alialooooosh055455 why are you spamming mentions everywhere without saying anything? Are you a bot or something?

alialooooosh055455, 2019-03-04:
O

alialooooosh055455, 2019-03-04:
~~**
### 
## 
# ** ~~

Zeonymous, 2019-03-04:
Basically, just a simple wave, all components, with the second component on controller 1 having 6 nodes, instead of 12.

So, there is the PSU attached to the USB header on the motherboard.
Then in this order, on the first controller are
1. Rear Exhaust.
2. Pump LED
3,4,5. are simply the next 3 fans which are exhausting the top mounted radiator.

Then, controller 2, is:
1.2.3.: All 3 frontal intake fans.

Pretty simple setup. They're all doing the same thing, aside from the pump. Which is IIRC, PWM mode, based on temp (The case is huge, I don't believe I've ever seen them rev to full) all running the "Wave" config RGB setup.

Both RiingPlus, and TTcontroller service are stopped. This is the "show hardware" info:

```
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 100, Rpm: 1364, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 100, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 100, Rpm: 1621, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 100, Rpm: 1611, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 100, Rpm: 1609, Unknown: 255]
        Identifier: [9802, 8101, 5]

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 100, Rpm: 1519, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 100, Rpm: 1507, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 100, Rpm: 1504, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 100, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 100, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Sensors
-------------------------------
AMD Ryzen Threadripper 2950X 16-Core Processor:
        /amdcpu/0/temperature/0:
                Name: Core (Tctl)
                Value: 64.25001
        /amdcpu/0/temperature/1:
                Name: Core (Tdie)
                Value: 64.25001
NVIDIA GeForce RTX 2080 Ti:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 27
Samsung SSD 860 EVO 1TB:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 25
Samsung SSD 860 EVO 1TB:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 24
```

The service only hung up when I tried using the old GUI's config. It seems fine now. Starts and stops when requested.

Not worried about that guy. Trolls gonna troll.

devcompl, 2019-03-04:
Ok, I think this is what you want:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "d6b35392-429c-48d8-af11-2105ae51e800",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100],
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Wave",
            "Speed": "Extreme",
            "Colors": [
              [0, 255, 29],
              [0, 255, 182],
              [0, 233, 255],
              [0, 110, 255],
              [0, 12, 255],
              [119, 0, 255],
              [216, 0, 255],
              [255, 0, 178],
              [255, 0, 16],
              [255, 67, 0],
              [255, 174, 0],
              [255, 250, 0]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [
    {
      "Key": [9802, 8101, 2],
      "Value": {
        "LedCount": 6,
      }
    }
  ],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

BTW I just realized that I there is a bug with `LedCount` so it still sends all colors instead of clipping the list. I'll release a fix shortly.

You can try changing the `Colors` array to six values to test the 6 led device while I push the fix, but obviously the fans will be half off.

Zeonymous, 2019-03-04:
Right on. Awesome. I'll download the new release, get that all set up, and copy this into the confg.
So neat, and orderly! I really need to figure out how to do this style of coding so I can better understand it.

Zeonymous, 2019-03-04:
Okay, I've gotten the service working, it seems.

However, I don't know how to code. More or less, from the 

[image]

So, I do have a slight issue. Trying to update, IIRC, removing the previous files, and unpacking into the folder is the correct method, right?

These files specifically (With RiingPlus and TTController services stopped) report this error message ^

Zeonymous, 2019-03-04:
Okay, so I ran it. Looks like something fishy happened?

What I see, (I'll try to grab a video later) is that all the fans are using 6/12 LEDs.
The pump is also using 6/12 LEDs, but they're all technically using all of the LEDS, just that the latter 6 are "off", or just not reporting a color.

Same goes for the pump, it's cycling through 6 colors, and 6 "Black" or "Off" rather than cycling through only 6.

What it looks like, is the LedCount: 6 ended up copying this value for all, and the colors of all of the fans, and pump are as follows:
Green
Teal
Teal
Teal
Blue
Purple
"Off"
"Off"
"Off"
"Off"
"Off"
"Off"

I'm wondering why the:
```    {
      "Key": [9802, 8101, 2],
      "Value": {
        "LedCount": 6,
      }```
Is seemingly counting the same for all the fans, and not just the second one?

Is it because:
````
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
```
"[9802, 8101, 2]," is up here, as well?

devcompl, 2019-03-04:
> So neat, and orderly! I really need to figure out how to do this style of coding so I can better understand it.

It's not really coding, this is called json. A lot of text editors allow you to make it neat and also highlight any errors you made. You can even find some json editors online.

> Okay, so I ran it. Looks like something fishy happened?

Yeah... I was able to reproduce the issue and fix it quickly, sorry about that. Really hard to debug this stuff without having access to more TT devices. Try `0.4.2.0-dev` release.

You can also remove the coma in `"LedCount": 6,`, I dont think this is valid json but I tried it and it is being parsed without errors.

Zeonymous, 2019-03-04:
Oh, right on. I'll go grab the new one, and reload the previous config, delete that comma, and report back with what happens from here.

Zeonymous, 2019-03-04:
Right on.

So, the fix for 4.2.0 did the trick, and removing the comma didn't change anything, with or without.

Although, the previous issue from before has persisted.

I went ahead and too some video, as I'd not know how to troubleshoot it offhand.

https://youtu.be/e2zW4VS1JeQ

What it seems to be doing is limit the "Active LEDs" without disabling their 12 in-total cycle.
Half of the LEDs are off, but it's not exactly how they were intended to be "Inactive" nodes. Rather just making 6/12 active, rather than 6/6 active.

devcompl, 2019-03-05:
> What it seems to be doing is limit the "Active LEDs" without disabling their 12 in-total cycle.
> Half of the LEDs are off, but it's not exactly how they were intended to be "Inactive" nodes. Rather just making 6/12 active, rather than 6/6 active.

I guess they must be somehow setting how many leds does a port have, but I dont know what is the command. Or I bet they have the controller pre-programmed to 12 leds but they just duplicate the leds in software so it looks correct 🤦‍♂️ That would be very Thermaltake thing to do.

To check this I would need you to capture usb data with wireshark. So I can determine if they set the led count with some command or just duplicate colors.

Or I could just ignore their buggy rgb effects and create them from scratch. And I think this is a better option, less problems and way more configurable effects. Only downside is it would use a slight amount of cpu.

Zeonymous, 2019-03-05:
The main issue being is having to deal with their terrible software to figure out how the manage it on their own.

I've got Wireshark installed now, not exactly sure how to access the controller.

Rebooting to finalize the Wireshark install, if you'd still want me to try to figure out what exactly the issue is, let me know.

That would be VERY ironic if it "Duplicated" the 6 colors on the latter end, rather than a raw signal of just 6 colors. Know what... I wouldn't put it past them, either.

LMK, and I'll go on ahead finding the information.

devcompl, 2019-03-05:
> I've got Wireshark installed now, not exactly sure how to access the controller.

Ok, so when you are installing wireshark make sure to also install usbpcap (the installer asks if you want to install it), if you didnt I think you have to uninstall and reinstall again with that option selected.

After you have wireshark and usbpcap installed stop the service, launch wireshark and double click on `USBPCap1` on the main screen. Dont worry if wireshark hangs for a long time because there is a bug with usbpcap, it will load after some time.

When you see some lines scrolling, start the official software and "play with it". Like change rgb modes a few times on the pump port, then change the type from pump to fan a couple of times etc. 
But don't do this too long or the exported file will be very large.

Then click on the red square to stop the capture and export the data to a file using `File -> Save As...`. Upload it here or somewhere where I can download it, you can delete it after if you want.

Zeonymous, 2019-03-05:
[TTTesting.zip](https://github.com/devcompl/TTController/files/2929104/TTTesting.zip)

That hopefully does the trick.

I didn't get to change from pump to fan, but in this file, I changed every one of the nodes a few times.

devcompl, 2019-03-05:
> I didn't get to change from pump to fan, but in this file, I changed every one of the nodes a few times.
What do you mean by nodes? I wanted to check if there is some command they send when you change the device type to set the led count.

It seems they are not duplicating the led colors but always appending the same 6 colors. So yeah, the controller seems to be set to 12 leds.

Try this:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "d6b35392-429c-48d8-af11-2105ae51e800",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100],
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Wave",
            "Speed": "Extreme",
            "Colors": [
              [255, 0, 255],
              [255, 0, 0],
              [255, 255, 0],
              [0, 255, 0],
              [0, 255, 255],
              [0, 0, 255],
              [235, 255, 0],
              [195, 255, 0],
              [155, 255, 0],
              [195, 255, 0],
              [235, 255, 0],
              [255, 240, 0]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

The fans should be half color, half yellowish color. This is the exact data they are sending. Please tell me how the pump leds look, do they also cycle thru the yellowish colors?

If they do I guess the simplest way to solve this is to duplicate the colors like I said. Because I have no idea how they are setting the led count.
```json
{
  "Profiles": [
    {
      "Name": "Fans",
      "Guid": "d6b35392-429c-48d8-af11-2105ae51e800",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100],
            ],
            "Sensors": [
              "/amdcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Wave",
            "Speed": "Extreme",
            "Colors": [
              [0, 255, 29],
              [0, 255, 182],
              [0, 233, 255],
              [0, 110, 255],
              [0, 12, 255],
              [119, 0, 255],
              [216, 0, 255],
              [255, 0, 178],
              [255, 0, 16],
              [255, 67, 0],
              [255, 174, 0],
              [255, 250, 0]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "2e1753be-41c2-4d2b-b232-42699d9f95be",
      "Ports": [
        [9802, 8101, 2]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Wave",
            "Speed": "Extreme",
            "Colors": [
              [0, 255, 29],
              [0, 233, 255],
              [0, 12, 255],
              [216, 0, 255],
              [255, 0, 16],
              [255, 174, 0],
              [0, 255, 29],
              [0, 233, 255],
              [0, 12, 255],
              [216, 0, 255],
              [255, 0, 16],
              [255, 174, 0]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

Zeonymous, 2019-03-05:
Okay, so the first json config  has all of the fans at the half yellow, as suggested, and the same happens for the pump, as well.
Cycling though the 6 RGB, and then the 6 Yellow.

The second json won't seem to run, and the "Run in console" command crashes the .exe.

Alright. I'll run the Wireshark again, and specifically do the one thing I didn't before.
I will set the pump to the Riing Plus Initially.
So, the start of this will show Controller 1, port 2, as a fan, at first.
(Note: this makes the pump function as we've experienced up until this point.)
I'll start Wireshark, change it to the pump, and stop wireshark. I'll send that data here:
[PumpOff_PumpOn.zip](https://github.com/devcompl/TTController/files/2932534/PumpOff_PumpOn.zip)

If you need me to send anything, I think I have a better grasp on Wireshark, so if you need me to do anything specific, I'll be happy to do so.

Side note:
I let the ThermalTake software run last night, as a test to see if the newer 1.2.7 would crash overnight...
And it did.
Lol.
(Not just one controller, mind you. Everything, including the software.)

devcompl, 2019-03-05:
> Okay, so the first json config has all of the fans at the half yellow, as suggested, and the same happens for the pump, as well.
> Cycling though the 6 RGB, and then the 6 Yellow.

Then I really have no idea how they do the 6 led devices, I'll take a look at the new wireshark file.

> The second json won't seem to run, and the "Run in console" command crashes the .exe.

Oh god, yea found another bug. Pushed 0.4.3.0-dev with a fix. 

It also includes a fix for detection of dpsg. Please check if show hardware info now detects the psu, ofc make sure the official dpsg app is not running.

Zeonymous, 2019-03-05:
Can do, I'll try that now. I haven't turned off DPSG because it normally functions properly, and gives me usage stats, and stuff.
Is there a way to decouple the fans/leds from DPSG without losing all of the information it provides via the exe?

Issue: I'll look past it for now, but I'm having persistent issues updating, because everything else works, but I can't edit, change, delete the file called "OpenHardwareMonitorLib.sys" meaning I can't replace it either.
It say's it's open in another program, but I can't for the life of me figure out what program that is.

devcompl, 2019-03-05:
> Is there a way to decouple the fans/leds from DPSG without losing all of the information it provides via the exe?

Not really, because it would mean that 2 applications are writing to the same controller and they would get conflicting outputs. I am just curious if it will be detected correctly.

> I'll look past it for now, but I'm having persistent issues updating, because everything else works, but I can't edit, change, delete the file called "OpenHardwareMonitorLib.sys" meaning I can't replace it either.
It say's it's open in another program, but I can't for the life of me figure out what program that is.

Yea I have this problem sometimes, dont know what is causing it. But this file is a driver created by the LibreHardwareMonitor project, I guess somehow used to read the sensors. You dont really need to remove/replace it since it will never change.

Zeonymous, 2019-03-05:
Okay, that did it, but the first json did not affect the DPSG LEDs (With the yellow) it seems.

However, the second one works with 0.4.3.0-dev

(Pump included, as well)

The only change I think I'd personally make to it now, would be pulling it back from "Extreme" to "Fast" as to more reasonably flow with the fans speed.

Also, on that topic, in the ThermalTake software, extreme is a lot faster, like rebelliously fast. (Not personally preferable, btw.)

Just noting that there is a step-up from your service's extreme, to ThermalTake's "extreme" speed.
Not sure if this bothers you at all, but it's something I figured I'd point out, so you're aware.

devcompl, 2019-03-05:
> Okay, that did it, but the first json did not affect the DPSG LEDs (With the yellow) it seems.

It will not affect it because it is not in the config. Dpsg is not fully supported yet and just wanted to see if the dpsg controller is detected at all. You can check it in "show hardware info", it should list 3 controllers, one of them should be "Dpsg controller".

The effects and static speed controller should work, but I was unable to figure out the data returned from the dpsg controller to read speed/rpm/watts/and voltage stuff. So the pwm control will not work without a fan speed reading.

> The only change I think I'd personally make to it now, would be pulling it back from "Extreme" to "Fast" (For the pump) as to more reasonably flow with the fans speed.

Yeah, just change the effect speed from "Extreme" to "Fast" in the config.

> Also, on that topic, in the ThermalTake software, extreme is a lot faster, like ridiculously fast. (Not personally preferable, btw.)
> Just noting that there is a step-up from your service's extreme, to ThermalTake's "extreme" speed.
Not sure if this bothers you at all, but it's something I figured I'd point out, so you're aware.

I bet they also abandoned the built in effects and just started to create them in the software by just sending all colors, 30 times a second (instead of this being done on the controller itself). That is why there is a speed difference. And looking at the wireshark data that is exactly what they are doing, for the pump they are sending 12 colors.

> If you want to get the PSU detection sorted, I'll be happy to assist, even if I don't have a specific use for it, as the current json mimics the PSU's format for me.
Would this mean that there would have to be a toggle in the json when the PSU detection is figured out, so that there are no conflicts when I do end up using the DPSG program?

There is no need for a toggle, if you dont include the port identifier for dpsg in a profile it will not conflict with the official dpsg software. The only command that will be send is to initialize it, that should not break the official software but I dont have one so I cant test.

If you see the dpsg controller in the show hardware info, you can add its port identifier to the pump profile.

I think it would be:
```json
        [9802, 8101, 2],
        [9802, 9001]
```

This will make it also control the psu rgb. If it works I guess I would only need to figure out how to read the data and it would be fully supported.

Zeonymous, 2019-03-05:
> It will not affect it because it is not in the config. Dpsg is not fully supported yet and just wanted to see if the dpsg controller is detected at all. You can check it in "show hardware info", it should list 3 controllers, one of them should be "Dpsg controller".

The "Show Hardware" shows this:
```
Controllers
-------------------------------
2019-03-05 16:44:10.5561|INFO|TTController.Service.Manager.DeviceManager|Creating Device Manager...
2019-03-05 16:44:10.6519|INFO|TTController.Service.Manager.DeviceManager|Detected controller: Dpsg Controller [0x264A, 0x2329]
2019-03-05 16:44:10.8638|INFO|TTController.Service.Manager.DeviceManager|Detected controller: Riing Plus Controller [0x264A, 0x1FA5]
2019-03-05 16:44:10.8638|INFO|TTController.Service.Manager.DeviceManager|Detected controller: Riing Plus Controller [0x264A, 0x1FA6]
Name: Dpsg Controller
VendorId: 9802
ProductId: 9001
Ports:
        Id: 0
        Data: [PortId: 0, Rpm: 0, Temperature: 0, WATTS: 0]
        Identifier: [9802, 9001]
```
So, up, it sees it.

> The effects and static speed controller should work, but I was unable to figure out the data returned from the dpsg controller to read speed/rpm/watts/and voltage stuff. So the pwm control will not work without a fan speed reading.

Okay, so you need to figure out how to get that info read in the service?

> Yeah, just change the effect speed from "Extreme" to "Fast" in the config.

Right on.

> I bet they also abandoned the built in effects and just started to create them in the software by just sending all colors, 30 times a second (instead of this being done on the controller itself). [...]

How very ThermalTake of them. There's no wonder the app is always crashing.

 
> There is no need for a toggle, if you dont include the port identifier for dpsg in a profile it will not conflict with the official dpsg software. The only command that will be send is to initialize it, that should not break the official software but I dont have one so I cant test.

So it is possible, but you'd just need the info for what the port is controlling specifically?
Not sure if I follow, entirely. I feel like it would possibly be overwritten (RGB/Fan Speeds) by the official software 

> If you see the dpsg controller in the show hardware info, you can add its port identifier to the pump profile.
>
> I think it would be:
>
> ```json
>         [9802, 8101, 2],
>         [9802, 9001]
> ```
>
> This [...]

Cool, I'll try that. See how it goes.

Yup, that gives the PSU the Pump's 12 RGBs.

devcompl, 2019-03-05:
Cool! Still a lot of stuff to do but at least its partially working.

> Okay, so you need to figure out how to get that info read in the service?
> So it is possible, but you'd just need the info for what the port is controlling specifically?
Not sure if I follow, entirely. I feel like it would possibly be overwritten (RGB/Fan Speeds) by the official software

Basically, I know all of the commands to read the data, like fan speed/temperature/voltage etc., but I haven't yet figured out how to convert the returned data to real values, for ex. this is temperature from your wireshark data: `[0x70, 0x10]`. I also need to figure out how to expose the temperature sensor to the plugins.

While I work on dpsg stuff you can keep using the official dpsg software, but remember to remove `[9802, 9001]` from your config so the service does not interfere with the official software. 
The service should not break the official software while it detects dpsg controller but if it does, please create a new issue for that.

Anyway, we are getting kinda off topic in this issue. 
I am closing this since you have a working config, if you have more config questions you can reply here or just create a new issue.

closed 2019-03-05 by devcompl