MoshiMoshi0/TTController #6: Vinasis fron TT Forum

Vinasis01, 2019-03-28:
**Describe your desired configuration**
Hello, appreciate you taking the time to help, I have the thermaltake level 20 XT case which is made for water cooling, I got it for air cooling and have 12 fans installed in it of the Riing Plus series.
I would like to set some of the fans to zero spin and some to 75% or 90% speed while keeping the rgb to test which configuration would be better for cooling since this case isn't the normal case design, the motherboard is flat not on the wall of the case.
I tried the version 1.7 but each time i restart I have to do each fan manually which I think via the github version would be easier.
For now I am planning to use the RGB Spectrum effect.
I tried to write it on my own but sadly I am not that well versed in writing these codes, I am good with hardware assembly and stuff.
Another way is to open the case and manually detaching different cables from the controllers to test "cable management for 12 fans and other parts was hell".
I got lost while trying to figure what/how to write the .json file...

Thanks.

**Hardware info**
```
Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 907, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 910, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 50, Rpm: 904, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 50, Rpm: 922, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 50, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8103
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 886, Unknown: 255]
        Identifier: [9802, 8103, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 883, Unknown: 255]
        Identifier: [9802, 8103, 2]

        Id: 3
        Data: [PortId: 3, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8103, 3]

        Id: 4
        Data: [PortId: 4, Speed: 50, Rpm: 511, Unknown: 255]
        Identifier: [9802, 8103, 4]

        Id: 5
        Data: [PortId: 5, Speed: 50, Rpm: 635, Unknown: 255]
        Identifier: [9802, 8103, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 30, Rpm: 527, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 30, Rpm: 518, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 30, Rpm: 529, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 30, Rpm: 534, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
AMD Ryzen Threadripper 1950X 16-Core Processor:
        /amdcpu/0/temperature/0:
                Name: Core (Tctl)
                Value: 70.5
        /amdcpu/0/temperature/1:
                Name: Core (Tdie)
                Value: 43.5
NVIDIA GeForce RTX 2080 Ti:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 57
ST8000AS0002-1NA17Z:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 35
WDC WD80EZAZ-11TDBA0:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 37
WDC WD2002FAEX-00MJRA0:
        /hdd/2/temperature/0:
                Name: Temperature
                Value: 38
WDC WD2002FAEX-00MJRA0:
        /hdd/3/temperature/0:
                Name: Temperature
                Value: 38
Samsung SSD 860 EVO M.2 2TB:
        /hdd/5/temperature/0:
                Name: Temperature
                Value: 42
-------------------------------
```

**Current configuration**
```json
If applicable, add contents of your current configuration here.
```

devcompl, 2019-03-29:
Please use 0.5.1.0-dev because I found a small bug.

I think this config should do what you want:
```json
{
  "Profiles": [
    {
      "Name": "On",
      "Guid": "73f9a2b2-cf11-40db-a052-844d0c8d6516",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3],
        [9802, 8102, 4],
        [9802, 8103, 1],
        [9802, 8103, 2],
        [9802, 8103, 4],
        [9802, 8103, 5],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1.0,
            "Brightness": 1.0,
            "HueStep": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Off",
      "Guid": "189c57ca-e5eb-4bd4-ae7f-37d169ad3ebb",
      "Ports": [
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 0,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1.0,
            "Brightness": 1.0,
            "HueStep": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

To change the off fans, cut the desired ports from "On" profile and paste them to "Off" profile ports.

Vinasis01, 2019-03-29:
Hello, 
**Thank you, you saved me from the regret I had due to Thermaltake support being none existent.**
I followed your directions and it worked great, I just wanted to ask If I wanted to further segregate the fan work like have 3 sets, as set running at 75% another at 90% and the last one at zero, would it look like this?

{
  "Profiles": [
    {
      "Name": "On1",
      "Guid": "73f9a2b2-cf11-40db-a052-844d0c8d6516",
      "Ports": [
	[9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8103, 1],
        [9802, 8103, 2],        
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
      "Name": "On2",
      "Guid": "73f9a2b2-cf11-40db-a052-844d0c8d6516",
      "Ports": [
	[9802, 8103, 4],
        [9802, 8103, 5],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 90,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Off",
      "Guid": "189c57ca-e5eb-4bd4-ae7f-37d169ad3ebb",
      "Ports": [
	[9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
	[9802, 8102, 3],
        [9802, 8102, 4],
        
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 0,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}

devcompl, 2019-03-29:
Yea pretty much, but you have some syntax errors (missing '{') and you also need to change the guid for each profile to be unique.

Here is it corrected:

```json
{
  "Profiles": [
    {
      "Name": "On1",
      "Guid": "73f9a2b2-cf11-40db-a052-844d0c8d6516",
      "Ports": [
	[9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8103, 1],
        [9802, 8103, 2],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "On2",
      "Guid": "9c28427d-0a7b-48cb-b935-70c068201eb5",
      "Ports": [
        [9802, 8103, 4],
        [9802, 8103, 5],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 90,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Off",
      "Guid": "189c57ca-e5eb-4bd4-ae7f-37d169ad3ebb",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8102, 3],
        [9802, 8102, 4],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 0,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

You can get new guids [here](https://www.guidgen.com/), and test your json [here](https://jsoneditoronline.org/).

Vinasis01, 2019-03-29:
Thanks a lot, appreciate your assistance <3 I will start testing different things, hope someday you can add a reload function for when the pc restarts, even if you don't add one it's so ease via manage services to do once on every restart.

I am just glad that you're helping in ways thermaltake themselves aren't doing.

Respect <3

devcompl, 2019-03-29:
> hope someday you can add a reload function for when the pc restarts, even if you don't add one it's so easy via manage services to do once on every restart.

What do you mean by reload on restart?

Vinasis01, 2019-03-29:
Everything seems fine with it now, I did shut down my PC and started it the effects and speeds were at default but then again I shut > start again and it worked, tried 4-5 times full shut down and start after that and they were working fine.

devcompl, 2019-03-30:
It should always load the config when booting, after restart or after resume from sleep. Don't know why it didnt, maybe you had official software running or the old service? 

While developing I think I never encountered this problem. If it happens again please make a bug report issue.

Since you have your config working I will close this. If you need any help with the config feel free to create another issue.

closed 2019-03-30 by devcompl