MoshiMoshi0/TTController #7: Config reader crash

AlphaDE, 2019-04-10:
Hi,

with the current version, when I copy&paste exactly the last config from "Need help configurating 3 Riing Plus 12 fans", having the same ports (1-3 used for pump and 2 fans, 4 and 5 free) and adapting to my hardware (intelcpu instead of amdcpu)

```
{
 "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/intelcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "AlwaysTrigger": {}
            }
          }
        }      ],
      "Effects": [
      ]
    }
  ],
  "ComputerStateProfiles": [
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/0": 70,
    "/intelcpu/0/temperature/1": 70,
    "/intelcpu/0/temperature/2": 70,
    "/intelcpu/0/temperature/3": 70,
    "/intelcpu/0/temperature/4": 70
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 1000
}

```
with the following hardware

```
Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 35, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 1]
[... 98 lines cut ...]
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bei Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   bei Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   bei Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   bei Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   bei TTController.Service.Manager.ConfigManager.LoadOrCreateConfig()
   bei TTController.Service.TTService.Initialize()
   bei TTController.Service.Program.<>c.<Main>b__2_1()
   bei TTController.Service.Program.Main(String[] args)

```

MoshiMoshi0, 2019-04-10:
That config was made for an earlier version so its now invalid.

The ```"AlwaysTrigger": {}``` line should be changed to ```"Type": "AlwaysTrigger"```

So the full config would be:
```json
{
 "Profiles": [
    {
      "Name": "Default",
      "Guid": "458ad61c-e033-4c84-b27d-b57492716450",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100]
            ],
            "Sensors": [
              "/intelcpu/0/temperature/0"
            ],
            "SensorMixFunction": "Maximum",
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }      ],
      "Effects": []
    }
  ],
  "ComputerStateProfiles": [],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/0": 70,
    "/intelcpu/0/temperature/1": 70,
    "/intelcpu/0/temperature/2": 70,
    "/intelcpu/0/temperature/3": 70,
    "/intelcpu/0/temperature/4": 70
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 1000
}
```

AlphaDE, 2019-04-10:
Thx for the quick reply, I assume it's the same for the Effects section of that issue... do you have an overview over accepted sets?

I tried the original one... and I assume that Type and/or Trigger need to be different, too?

```
      "Effects": [
        {
          "Type": "DefaultEffect",
          "Config": {
            "Type": "Pulse",
            "Speed": "Slow",
            "Colors": [
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255],
              [0, 0, 255]
            ],
            "Trigger": {
              "OneTimeTrigger": {}
            }
          }
        }
      ]

```

AlphaDE, 2019-04-10:
Btw. this one from the main page leads also to crash of the config reader.

```
        {
          "Type": "RippleEffect",
          "Config": {
            "Length": 4,
            "Color": [
              [255, 0, 0]
            ],
            "TickInterval": 2,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }

```

Recommendation: Skip the section and print the faulty statement if that's feasible.

AlphaDE, 2019-04-10:
Is it possible to exclude a single port from FanSpeed (like the liquid pump) but keep it for effects?

MoshiMoshi0, 2019-04-10:
> Thx for the quick reply, I assume it's the same for the Effects section of that issue... do you have an overview over accepted sets?
> 
> I tried the original one... and I assume that Type and/or Trigger need to be different, too?

Yes, the trigger format is wrong, but also the ```DefaultEffect``` no longer exists. Each of the default Thermaltake effects was rewritten as a separate plugin for more configuration options. If you want to use the controller builtin effects you can use ```RawEffect``` which uses one of the effects from the "Available effect types" list.

> Btw. this one from the main page leads also to crash of the config reader.

Whoops, thanks for spotting this.
It should be ```"Color": [255, 0, 0],```.

> Recommendation: Skip the section and print the faulty statement if that's feasible.

Nicer logging of config errors is already added on develop branch, will be in the next release.

> Is it possible to exclude a single port from FanSpeed (like the liquid pump) but keep it for effects?

Yea just create another profile.
Basing on your other issue it would look something like this:

```json
{
  "Profiles": [
    {
      "Name": "Fans",
      "Guid": "5212f5c7-a69e-49fd-86c0-ef7412eea6f0",
      "Ports": [
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 20],
              [40, 25],
              [50, 40],
              [60, 75],
              [70, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "TemperatureEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/4"],
            "ColorGradient": [
              [30, [0, 0, 255]],
              [60, [255, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "c162d280-273d-4b65-a52a-d7a1ff1e41d1",
      "Ports": [
        [9802, 8101, 1]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "TemperatureEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/4"],
            "ColorGradient": [
              [30, [0, 0, 255]],
              [60, [255, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 50
    }
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/4": 70
  },
  "PortConfig": []
}
```

Btw what pump do you have? Was its speed changing with the fans? In the "show hardware info" output you can see its RPM is 0, so I assumed it is not speed controlled.

AlphaDE, 2019-04-10:
What an outstanding support... 

I recently replaced my 2014 Thermaltake Water 3.0 Extreme with Thermaltake Floe Dual Ring RGB 240 Premium, so I have the Floe Riing Water Pump (which as pump is running as a constant speed) and the two 120mm Riing Plus RGB fans for the cooling element.

The old software for the Water 3.0 was not updated anymore, it was really loud when the fans ran at 50+% and it started to ignore the config (or fought with the BIOS over control).

With the Floe, I had to notice that after a resuming from hibernation, TT software did not restore the original status, so I had to end and restart the software. Looking for a solution, I found your tool mentioned in the TT forums which is exactly the solution I am looking for. Nice, clean minimal service, running in the background and doing, what it should do.

Anything I need to add for resume after hibernation or does the service resets the controller properly?

The pump does not show any RPM in the TT software, only the fans do.

AlphaDE, 2019-04-10:
On what base do you create the GUIDs?

MoshiMoshi0, 2019-04-10:
> so I have the Floe Riing Water Pump (which as pump is running as a constant speed)

Well if the pump is at constant speed and is not speed controllable than there is no reason for 2 profiles. The service will still send commands to the controller to change the speed on pump port but if the pump does not understand that it needs to change the speed it will simply ignore it.

In the official software you can set device type for each port, and so you can force the controller to think that on a port that u have a pump connected there is a fan, that would mean the software is sending speed update commands to the pump, just like the service would with 1 profile.

> Anything I need to add for resume after hibernation or does the service resets the controller properly?

Nope it should work correctly, ofc each setup is different so no guarantees.

> On what base do you create the GUIDs?

I generate them [here](https://www.guidgen.com/). But you could just change one number to a different one, they just have to be in this format and unique.

Btw you might want to add
```json
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 50
    }
```
to the ```ComputerStateProfiles``` so when your pc is booting it will force the fans to a given speed. It overrides the default speed (and/or rgb) if there is no software controlling the fans. The service needs to run once to save the boot profile settings to the controller.

AlphaDE, 2019-04-10:
The TT software reports an error if one tries to convert the TT Floe Riing (Pump) into a Riing Plus (Fan) and does not allow it. That's why I asked how to exclude it here from the pwmspeed part completely.

Your program is provided for free and as is, so if it breaks, I just keep both parts :) 

Ok, added and restarted the service. Thanks for that.

Guess a FAQ / examples section where every user can send you his config with a description (processor type/cores and what it does, as well as the version of TTController it was used with) could help new users understanding how to configure or just by grabbing the matching example and going from there.

Nevertheless, great piece of software and exactly what I needed. Thanks for providing it and supporting it.

MoshiMoshi0, 2019-04-11:
> The TT software reports an error if one tries to convert the TT Floe Riing (Pump) into a Riing Plus (Fan) and does not allow it.

Oh, thats interesting. I wonder how they detect the pump. What does the error say?

> Guess a FAQ / examples section where every user can send you his config with a description (processor type/cores and what it does, as well as the version of TTController it was used with) could help [...]

Yea the plan is to create a configurator app to create the configs so you would not have to touch json at all, but I have not yet figured out a good way to do this.

AlphaDE, 2019-04-11:
It shows this.

[image]

Basically, if it should be more than a simple json editor that would need sets of effects to achieve and then the app would offer a form for each effect to fill in the values, as text, as graph (for example for the fan speed curve) or as other control item.

Perhaps you could that even with config snippets that are read by the config app and then the config app generates the input form for that snippet.

MoshiMoshi0, 2019-04-11:
Could you stop the service then run [this](https://github.com/MoshiMoshi0/TTController/files/3068274/debug.zip) and post the output here?

AlphaDE, 2019-04-11:
W fe 33
R fe 33 fc 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 50
R 33 50 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 51 01
R 33 51 01 ff 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 51 02
R 33 51 02 ff 32 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 51 03
R 33 51 03 ff 32 bc 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 51 04
R 33 51 04 ff 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W 33 51 05
R 33 51 05 ff 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Here we go, Pump on 1, Fans on 2 and 3, 4 and 5 not used.

MoshiMoshi0, 2019-04-11:
Thanks, so nothing additional is returned for the pump port. 

I think that the warning you see is just a no/low RPM warning. Since the pump does not report its RPM when you change the pump port to a fan it expects the RPM to be non 0. So the controller doesn't know that the pump is connected to that port.

Im gonna close this issue since we are getting off topic.
If you need more help with configuration please make a new issue.

closed 2019-04-11 by MoshiMoshi0