MoshiMoshi0/TTController #8: Set color i.a.w. CPU temp

AlphaDE, 2019-04-10:
With this hardware 

`Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 30, Rpm: 514, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 30, Rpm: 517, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
Intel Core i7-4790K:
        /intelcpu/0/temperature/0:
                Name: CPU Core #1
                Value: 35
Intel Core i7-4790K:
        /intelcpu/0/temperature/1:
                Name: CPU Core #2
                Value: 34
Intel Core i7-4790K:
        /intelcpu/0/temperature/2:
                Name: CPU Core #3
                Value: 30
Intel Core i7-4790K:
        /intelcpu/0/temperature/3:
                Name: CPU Core #4
                Value: 32
Intel Core i7-4790K:
        /intelcpu/0/temperature/4:
                Name: CPU Package
                Value: 35
NVIDIA GeForce GTX 1080 Ti:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 42
WDC WD20EFRX-68EUZN0:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 28
-------------------------------
Press any key to continue...
`

I would like to implement the same function TT software offers, to set the color of the water pump and my two fans (ID 0  and 1,2) from blue to red i.a.w. the current CPU temperature.

Any hints how to do this?

MoshiMoshi0, 2019-04-10:
I think this should do what you want:

```json
{
  "Profiles": [
    {
      "Name": "Fans",
      "Guid": "5212f5c7-a69e-49fd-86c0-ef7412eea6f0",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 20],
              [40, 25],
              [50, 40],
              [60, 75],
              [70, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "TemperatureEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/4"],
            "ColorGradient": [
              [30, [0, 0, 255]],
              [60, [255, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 50
    }
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/4": 70
  },
  "PortConfig": []
}
```
You can configure the ```CurvePoints``` to your desired pwm speed curve for the fans, the format is ```[temperature, speed]```.

The color gradient can be configured in ```ColorGradient``` property, the format is ```[temperature, [r, g, b]]```. You can have more than two colors and it will blend between them depending on the temperature.

AlphaDE, 2019-04-10:
Works like a charm... now I can deactivate the TT software.

THANKS!

closed 2019-04-10 by AlphaDE