MoshiMoshi0/TTController #9: After resume from hibernation, service does not run anymore

AlphaDE, 2019-04-12:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional content**
If applicable, add any content that can help explain your problem.

MoshiMoshi0, 2019-04-12:
Please check the windows event log for crash events around the time you resume from sleep and post them. There should be 2 or 3 events.
Also attach the log if it was created in the ```Logs``` folder.

AlphaDE, 2019-04-12:
The service "Thermaltake Controller" terminated unexptectedly (happened a total of 3 times).

```
2019-04-12 16:36:42.9895|INFO|TTController.Service.TTService|================================================================
2019-04-12 16:36:43.0135|INFO|TTController.Service.TTService|Initializing...
2019-04-12 16:36:43.5615|INFO|TTController.Service.TTService|Loading plugins...
2019-04-12 16:36:43.5615|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.BlinkEffect [1.0.0.0]
2019-04-12 16:36:43.5615|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.ByLedColorEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.FlowEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.FullColorEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PulseEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.RawEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.RippleEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SnakeEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: CSCore [1.2.1.2]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SoundEffect [1.0.0.0]
2019-04-12 16:36:43.5715|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SpectrumEffect [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.TemperatureEffect [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.WaveEffect [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PwmSpeedController [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.StaticSpeedController [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.AlwaysTrigger [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.LogicTrigger [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.OneTimeTrigger [1.0.0.0]
2019-04-12 16:36:43.5885|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.ProcessTrigger [1.0.0.0]
2019-04-12 16:36:43.6025|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PulseTrigger [1.0.0.0]
2019-04-12 16:36:43.6025|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.TemperatureTrigger [1.0.0.0]
2019-04-12 16:36:43.6405|INFO|TTController.Service.Manager.ConfigManager|Creating Config Manager...
2019-04-12 16:36:43.6405|INFO|TTController.Service.Manager.ConfigManager|Loading config...
2019-04-12 16:36:44.4145|INFO|TTController.Service.Manager.ConfigManager|Loading done...
2019-04-12 16:36:44.4195|INFO|TTController.Service.Utils.DataCache|Creating DataCache...
2019-04-12 16:36:44.4195|INFO|TTController.Service.Manager.SensorManager|Creating Sensor Manager...
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.SensorManager|Detected sensors: 35
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.TemperatureManager|Creating Temperature Manager...
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/0
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/1
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/2
2019-04-12 16:36:45.7314|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/3
2019-04-12 16:36:45.7444|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/4
2019-04-12 16:36:45.7444|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /nvidiagpu/0/temperature/0
2019-04-12 16:36:45.7444|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /hdd/0/temperature/0
2019-04-12 16:36:45.7444|INFO|TTController.Service.Manager.EffectManager|Creating Effect Manager...
2019-04-12 16:36:45.7444|INFO|TTController.Service.Manager.SpeedControllerManager|Creating Speed Controller Manager...
2019-04-12 16:36:45.7754|INFO|TTController.Service.Manager.DeviceManager|Creating Device Manager...
2019-04-12 16:36:45.8154|INFO|TTController.Service.Manager.DeviceManager|Detected controller: Riing Plus Controller [0x264A, 0x1FA5]
2019-04-12 16:36:45.8244|INFO|TTController.Service.TTService|Applying config...
2019-04-12 16:36:45.8244|INFO|TTController.Service.Manager.EffectManager|Adding effect: TemperatureEffect [5212f5c7-a69e-49fd-86c0-ef7412eea6f0]
2019-04-12 16:36:45.8244|INFO|TTController.Service.Manager.SpeedControllerManager|Adding speed controller: PwmSpeedController [5212f5c7-a69e-49fd-86c0-ef7412eea6f0]
2019-04-12 16:36:45.8244|INFO|TTController.Service.Manager.TemperatureManager|Enabling sensor: /intelcpu/0/temperature/4
[... cut ...]
2019-04-12 19:23:23.8707|INFO|TTController.Service.Manager.EffectManager|Adding effect: TemperatureEffect [c162d280-273d-4b65-a52a-d7a1ff1e41d1]
2019-04-12 19:23:23.8947|INFO|TTController.Service.Manager.TimerManager|Creating Timer Manager...
2019-04-12 19:23:23.8947|INFO|TTController.Service.Manager.TimerManager|Starting 4 timers...
2019-04-12 19:23:23.9097|INFO|TTController.Service.TTService|Initializing done!
2019-04-12 19:23:23.9097|INFO|TTController.Service.TTService|================================================================

```

MoshiMoshi0, 2019-04-13:
It seems like the OpenHardwareMonitor is causing the crash. Please download and replace your current ```TTController.Service.exe``` with the attached one, then try to reproduce the issue. 
Post new logs as they will contain more information about the exception.

[TTController.Service.exe.zip](https://github.com/MoshiMoshi0/TTController/files/3075517/TTController.Service.exe.zip)

AlphaDE, 2019-04-13:
Downloaded and will try.

AlphaDE, 2019-04-13:
```
2019-04-12 20:10:20.6185|INFO|TTController.Service.TTService|================================================================
2019-04-12 20:10:20.6435|INFO|TTController.Service.TTService|Initializing...
2019-04-12 20:10:21.2635|INFO|TTController.Service.TTService|Loading plugins...
2019-04-12 20:10:21.2635|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.BlinkEffect [1.0.0.0]
2019-04-12 20:10:21.2635|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.ByLedColorEffect [1.0.0.0]
2019-04-12 20:10:21.2635|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.FlowEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.FullColorEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PulseEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.RawEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.RippleEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SnakeEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: CSCore [1.2.1.2]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SoundEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.SpectrumEffect [1.0.0.0]
2019-04-12 20:10:21.2755|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.TemperatureEffect [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.WaveEffect [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PwmSpeedController [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.StaticSpeedController [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.AlwaysTrigger [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.LogicTrigger [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.OneTimeTrigger [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.ProcessTrigger [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.PulseTrigger [1.0.0.0]
2019-04-12 20:10:21.2905|INFO|TTController.Service.TTService|Loading assembly: TTController.Plugin.TemperatureTrigger [1.0.0.0]
2019-04-12 20:10:21.3355|INFO|TTController.Service.Manager.ConfigManager|Creating Config Manager...
2019-04-12 20:10:21.3385|INFO|TTController.Service.Manager.ConfigManager|Loading config...
2019-04-12 20:10:22.0155|INFO|TTController.Service.Manager.ConfigManager|Loading done...
2019-04-12 20:10:22.0155|INFO|TTController.Service.Utils.DataCache|Creating DataCache...
2019-04-12 20:10:22.0155|INFO|TTController.Service.Manager.SensorManager|Creating Sensor Manager...
2019-04-12 20:10:23.5945|INFO|TTController.Service.Manager.SensorManager|Detected sensors: 35
2019-04-12 20:10:23.5945|INFO|TTController.Service.Manager.TemperatureManager|Creating Temperature Manager...
2019-04-12 20:10:23.6055|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/0
2019-04-12 20:10:23.6055|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/1
2019-04-12 20:10:23.6055|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/2
2019-04-12 20:10:23.6055|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/3
2019-04-12 20:10:23.6195|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /intelcpu/0/temperature/4
2019-04-12 20:10:23.6195|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /nvidiagpu/0/temperature/0
2019-04-12 20:10:23.6195|INFO|TTController.Service.Manager.TemperatureManager|Valid sensor identifier: /hdd/0/temperature/0
2019-04-12 20:10:23.6195|INFO|TTController.Service.Manager.EffectManager|Creating Effect Manager...
2019-04-12 20:10:23.6195|INFO|TTController.Service.Manager.SpeedControllerManager|Creating Speed Controller Manager...
2019-04-12 20:10:23.6505|INFO|TTController.Service.Manager.DeviceManager|Creating Device Manager...
2019-04-12 20:10:23.6875|INFO|TTController.Service.Manager.DeviceManager|Detected controller: Riing Plus Controller [0x264A, 0x1FA5]
2019-04-12 20:10:23.7025|INFO|TTController.Service.TTService|Applying config...
2019-04-12 20:10:23.7025|INFO|TTController.Service.Manager.EffectManager|Adding effect: TemperatureEffect [5212f5c7-a69e-49fd-86c0-ef7412eea6f0]
2019-04-12 20:10:23.7025|INFO|TTController.Service.Manager.SpeedControllerManager|Adding speed controller: PwmSpeedController [5212f5c7-a69e-49fd-86c0-ef7412eea6f0]
2019-04-12 20:10:23.7025|INFO|TTController.Service.Manager.TemperatureManager|Enabling sensor: /intelcpu/0/temperature/4
[... cut ...]
2019-04-12 20:59:02.3228|FATAL|TTController.Service.TTService|System.InvalidOperationException: Die Sammlung wurde geändert. Der Enumerationsvorgang kann möglicherweise nicht ausgeführt werden.
   bei System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   bei System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   bei System.Collections.Generic.List`1.Enumerator.MoveNext()
   bei System.Linq.Enumerable.<SelectManyIterator>d__17`2.MoveNext()
   bei TTController.Service.Manager.TemperatureManager.EnableSensors(IEnumerable`1 identifiers)
   bei TTController.Service.TTService.Initialize()
   bei TTController.Service.TTService.OnStart(String[] args)

```

MoshiMoshi0, 2019-04-13:
Ok, I think I fixed it. Please try this:

[TTController.Service.exe.zip](https://github.com/MoshiMoshi0/TTController/files/3076131/TTController.Service.exe.zip)

The problem was that there are two Resume from sleep events and both were trying to initialize the service at the same time causing the crash. Funny thing is on my machine the timing was slightly different and they were not interfering with each other, thats why I didnt catch this bug before.

AlphaDE, 2019-04-13:
This time it did not crash, but the fans stayed at 50% until I started TTController.exe from the commandline to check the status of the service. Log doesn't mention that, I check that.

Btw. any plans to support for example the NVidia color/fan control (AORUS)?

MoshiMoshi0, 2019-04-13:
> This time it did not crash, but the fans stayed at 50% until I started TTController.exe from the commandline to check the status of the service. 

You mean the service was not running after resume and you had to start it via "Manage Service -> Start" ?

> Log doesn't mention that, I check that.

What are the logs now?

> Btw. any plans to support for example the NVidia color/fan control (AORUS)?

Not really, I dont have one and I would have to disassemble the AORUS software to get the correct data to send. Also I dont know how would you even send data to the GPU, I dont think it would work with HidApi (usb) so this would probably require a lot of code to be rewritten.

AlphaDE, 2019-04-13:
No, the service was running, but it looked like the system remained on 50% instead of going back to temperature based control (colors)... but I just integrated a 200mm riing plus case fan and for that purpose, hibernated the system and checked it with OpenHardwareMonitor after startup... CPU was/is a little warmer (40°C instead of 30°C) which leads to a different color. So all good and working ok.

Regarding NVidia: Understandable... and with the new fan, I hardly can see the colored name tag of the graphics card, so I will disable AORUS.

MoshiMoshi0, 2019-04-13:
Ok, awesome. Thanks for reporting this.

I will leave this issue open for now until the next release just in case you find any more problems with sleep/resume.

AlphaDE, 2019-04-13:
Ok. I keep an eye...

AlphaDE, 2019-04-14:
Works reliable so far... now with 5 fans :)

MoshiMoshi0, 2019-04-15:
Ok, I pushed a new release with the fixes so I will close this.
You dont have to update because the latest version attached in this post and the 0.5.4.0-dev version are the same.

closed 2019-04-15 by MoshiMoshi0

AlphaDE, 2019-04-18:
Works fine.