MoshiMoshi0/TTController #10: Multiple Config

HaiMeister, 2019-04-17:
**Describe your desired configuration**
Hello, I am trying to set up two controllers with 3 fans on each. I can get your sample config working, however I cant seem to figure out how to set up different devs with different settings. I am trying to set one of the fans color to change based on temperature. The rest I just want static which I have figured out. I also want to set them to run pwm but when I change the config the service doesnt start. If you have a detailed config setup for each setting I can probably figure it out from there.

Also, I got this from the TT forum and it looks like you used to have a gui for it, does that still exist? If not no problem just curious.

P.S.
Thanks for your work, this is really awesome as my computer crashes when the OEM software runs.

**Hardware info**
```
Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 958, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 952, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 50, Rpm: 948, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 882, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 894, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 50, Rpm: 888, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 30, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 30, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
Intel Core i7-8700K:
        /intelcpu/0/temperature/0:
                Name: CPU Core #1
                Value: 35
Intel Core i7-8700K:
        /intelcpu/0/temperature/1:
                Name: CPU Core #2
                Value: 33
Intel Core i7-8700K:
        /intelcpu/0/temperature/2:
                Name: CPU Core #3
                Value: 48
Intel Core i7-8700K:
        /intelcpu/0/temperature/3:
                Name: CPU Core #4
                Value: 35
Intel Core i7-8700K:
        /intelcpu/0/temperature/4:
                Name: CPU Core #5
                Value: 36
Intel Core i7-8700K:
        /intelcpu/0/temperature/5:
                Name: CPU Core #6
                Value: 36
Intel Core i7-8700K:
        /intelcpu/0/temperature/6:
                Name: CPU Package
                Value: 48
AMD Radeon (TM) R9 390 Series:
        /atigpu/0/temperature/0:
                Name: GPU Core
                Value: 42
SanDisk SD6SB2M-512G-1006:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 35
Samsung SSD 850 EVO 500GB:
        /hdd/2/temperature/0:
                Name: Temperature
                Value: 29.
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "188a0675-06cd-4bb0-9d4e-e7367e8c6912",
      "Ports": [
		[9802, 8101, 1],
		[9802, 8101, 2],
                [9802, 8101, 3],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 50,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 255, 0],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    }
	
  ],
  "PortConfig": [
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/8": 90
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

MoshiMoshi0, 2019-04-17:
> I also want to set them to run pwm but when I change the config the service doesnt start.

Do you mind posting that not working config, and preferably the log that it created?
I want to see if your config was the problem or there is a bug.

> If you have a detailed config setup for each setting I can probably figure it out from there.

I know that the documentation is lacking but I have not figured out a good way to do it other than manually. I would like the documentation to be automatically generated. 

> Also, I got this from the TT forum and it looks like you used to have a gui for it, does that still exist? If not no problem just curious.

That was the old version. Due to the complexity of this one I dont know if it is worth it to implement. I was thinking about writing a html configuration form that should at least help with writing the config. But a gui to configure the service while its running (like the old version) would be very complex because of the plugin api.

Anyways I think this is the config you want:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "188a0675-06cd-4bb0-9d4e-e7367e8c6912",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/6"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 255, 0],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Temperature",
      "Guid": "188a0675-06cd-4bb0-9d4e-e7367e8c6913",
      "Ports": [
        [9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/6"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "TemperatureEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/6"],
            "ColorGradient": [
                [40, [16, 16, 128]],
                [60, [16, 16, 16]],
                [80, [128, 16, 16]]
            ],
            "Trigger": {
                "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "Speed": 35
    },
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "Speed": 35
    }
  ],
  "CriticalTemperature": {
    "/intelcpu/0/temperature/6": 90
  }
}
```

You can configure your desired pwm curve in ```CurvePoints``` with format ```[temperature, speed]``` and the temperature color gradient in ```ColorGradient``` with format ```[temperature, [r, g, b]]```

HaiMeister, 2019-04-17:
Okay I was just missing some arguments it looks like. I will try this tonight.

Okay I figured it was complex for the GUI. Thought about trying to create on myself but  idk how to do it without the args for each setting.

Is there maybe a way to have it output all the args that aren't populated when you change something?

And maybe instead of having a program that changes the config live, have an html or hta that pulls your hardware info, then it gives you options to select for each Dev and output a config that you can copy and paste into it. That would probably be the easiest way to do it.

MoshiMoshi0, 2019-04-17:
> Thought about trying to create on myself but idk how to do it without the args for each setting.

You would need to get all types that extend ```EffectConfigBase```/```SpeedControllerConfigBase```/```TriggerConfigBase``` and read the available properties.

You can find the config class in each plugin project folder.

And you would need to use for example WCF for client <-> service communication.

> Is there maybe a way to have it output all the args that aren't populated when you change something?

Not really, you would need to recompile the project with 
```csharp
NullValueHandling = NullValueHandling.Include,
DefaultValueHandling = DefaultValueHandling.Include,
```
in ```ConfigManager``` class, and then save the config after loading. That would save all config properties with default values. 

> And maybe instead of having a program that changes the config live, have an html or hta that pulls your hardware info, then it gives you options to select for each Dev and output a config that you can copy and paste into it. That would probably be the easiest way to do it.

Yea the html form that would generate the config would be easiest. But I would even skip the hardware info stuff and have the user fill it manually. 

I got it somewhat working by converting the class structure into a json schema and then using a json schema to html5 form generator. It has potential but there were a few bugs in the generator so I left it for now.

HaiMeister, 2019-04-17:
Gotcha. Okay well I'll let you know how it goes. And post any issues thanks!

HaiMeister, 2019-04-18:
That config worked perfectly for what I needed I tweaked it a little to change the pwm/color of the single fan to the temp of the GPU.

I am going to mess around with an html config page a little idk how itll turn out though! 

Thanks for everything!

closed 2019-04-18 by MoshiMoshi0