MoshiMoshi0/TTController #11: Configuration Help

Vinasis01, 2019-04-22:
**Describe your desired configuration**
Hello, hope you're doing well and are healthy, if you have the time please  I need help sitting a single color to my fans at full bright.

I think the color  number would be (255,165,0).

**Hardware info**
```
Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 75, Rpm: 1318, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 75, Rpm: 1254, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 50, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8103
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 75, Rpm: 1245, Unknown: 255]
        Identifier: [9802, 8103, 1]

        Id: 2
        Data: [PortId: 2, Speed: 75, Rpm: 1269, Unknown: 255]
        Identifier: [9802, 8103, 2]

        Id: 3
        Data: [PortId: 3, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8103, 3]

        Id: 4
        Data: [PortId: 4, Speed: 75, Rpm: 887, Unknown: 255]
        Identifier: [9802, 8103, 4]

        Id: 5
        Data: [PortId: 5, Speed: 75, Rpm: 855, Unknown: 255]
        Identifier: [9802, 8103, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
AMD Ryzen Threadripper 1950X 16-Core Processor:
        /amdcpu/0/temperature/0:
                Name: Core (Tctl)
                Value: 64.25
AMD Ryzen Threadripper 1950X 16-Core Processor:
        /amdcpu/0/temperature/1:
                Name: Core (Tdie)
                Value: 37.25
NVIDIA GeForce RTX 2080 Ti:
        /nvidiagpu/0/temperature/0:
                Name: GPU Core
                Value: 52
ST8000AS0002-1NA17Z:
        /hdd/0/temperature/0:
                Name: Temperature
                Value: 31
WDC WD80EZAZ-11TDBA0:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 32
WDC WD2002FAEX-00MJRA0:
        /hdd/2/temperature/0:
                Name: Temperature
                Value: 31
WDC WD2002FAEX-00MJRA0:
        /hdd/3/temperature/0:
                Name: Temperature
                Value: 32
Samsung SSD 860 EVO M.2 2TB:
        /hdd/5/temperature/0:
                Name: Temperature
                Value: 39
-------------------------------
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "On",
      "Guid": "73f9a2b2-cf11-40db-a052-844d0c8d6516",
      "Ports": [
	[9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8103, 1],
        [9802, 8103, 2],
        [9802, 8103, 4],
        [9802, 8103, 5],
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Off",
      "Guid": "189c57ca-e5eb-4bd4-ae7f-37d169ad3ebb",
      "Ports": [
	[9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
	[9802, 8102, 3],
        [9802, 8102, 4],
        
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 0,
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [],
  "PortConfig": [],
  "CriticalTemperature": {
    "/amdcpu/0/temperature/0": 85
  },
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

MoshiMoshi0, 2019-04-22:
Just change the effects to this:

```json
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 165, 0],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
```

Vinasis01, 2019-04-23:
Thanks for your continuous help bro

closed 2019-04-23 by Vinasis01