MoshiMoshi0/TTController #12: PresentationFontCache.exe causing high CPU usage

HaiMeister, 2019-04-23:
**Describe the bug**
After installing and running TTC everything seems normal, after a reboot a process called PresentationFontCache.exe shows up and uses 10-20% CPU causing my temp to rise quite a bit. Not sure if this is just normal for how the service runs but seems kinda weird.
uninstalling and reinstalling results in the same effect. I had V5.4 then updated to V6.0 and had same results.

The TTC.Service is using less than 1% at all times. 

Not sure if you know a way to optimize this or not.

**Hardware**
Controllers
-------------------------------
Name: Riing Plus Controller
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 35, Rpm: 770, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 35, Rpm: 768, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 35, Rpm: 758, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus Controller
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 35, Rpm: 694, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 35, Rpm: 727, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 35, Rpm: 693, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 30, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 30, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Sensors
-------------------------------
Intel Core i7-8700K:
        /intelcpu/0/temperature/0:
                Name: CPU Core #1
                Value: 35
Intel Core i7-8700K:
        /intelcpu/0/temperature/1:
                Name: CPU Core #2
                Value: 35
Intel Core i7-8700K:
        /intelcpu/0/temperature/2:
                Name: CPU Core #3
                Value: 38
Intel Core i7-8700K:
        /intelcpu/0/temperature/3:
                Name: CPU Core #4
                Value: 34
Intel Core i7-8700K:
        /intelcpu/0/temperature/4:
                Name: CPU Core #5
                Value: 49
Intel Core i7-8700K:
        /intelcpu/0/temperature/5:
                Name: CPU Core #6
                Value: 37
Intel Core i7-8700K:
        /intelcpu/0/temperature/6:
                Name: CPU Package
                Value: 49
AMD Radeon (TM) R9 390 Series:
        /atigpu/0/temperature/0:
                Name: GPU Core
                Value: 36
SanDisk SD6SB2M-512G-1006:
        /hdd/1/temperature/0:
                Name: Temperature
                Value: 35
Samsung SSD 850 EVO 500GB:
        /hdd/2/temperature/0:
                Name: Temperature
                Value: 30
-------------------------------

rickboy, 2019-04-23:
> **Describe the bug**
> After installing and running TTC everything seems normal, after a reboot a process called PresentationFontCache.exe shows up and uses 10-20% CPU causing my temp to rise quite a [...]

The PresentationFontCache.exe service is part of the .NET framework and is used by Windows Presentation Foundation (WPF) applications.

If you Google for "presentationfontcache cpu" you will see a number of pages with people having this issue and some possible fixes (just be careful what you do). 

The TTController is not a WPF application, so unfortunately, I don't believe there is anything that can be done.

HaiMeister, 2019-04-23:
Alright I will look into it more. Thanks.

closed 2019-04-23 by HaiMeister