MoshiMoshi0/TTController #13: Waiting For Service error

extayam, 2019-04-28:
Application: RiingPlus.Service.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ComponentModel.Win32Exception
   at System.ServiceProcess.ServiceProcessInstaller.GetAccountSid(System.String)
   at System.ServiceProcess.ServiceProcessInstaller.Install(System.Collections.IDictionary)
   at System.Configuration.Install.Installer.Install(System.Collections.IDictionary)
   at System.Configuration.Install.Installer.Install(System.Collections.IDictionary)
   at System.Configuration.Install.AssemblyInstaller.Install(System.Collections.IDictionary)
   at System.Configuration.Install.Installer.Install(System.Collections.IDictionary)
   at System.Configuration.Install.TransactedInstaller.Install(System.Collections.IDictionary)

Exception Info: System.InvalidOperationException
   at System.Configuration.Install.ManagedInstallerClass.InstallHelper(System.String[])
   at RiingPlus.Service.Program.InstallService()
   at RiingPlus.Service.Program.ManageService()
   at RiingPlus.Service.Program.Main(System.String[])

Managed to install from the zip file. But after login it says "Waiting for service". This is the log from the event viewer. Is there any fix to this ? Thanks for any help in advance

MoshiMoshi0, 2019-04-28:
It seems like you are using the old version but it is no longer supported.

Please try the new one from the releases.

extayam, 2019-04-28:
Is it also the zipped one too? Coz tried the one from github but im a total noob in pc software. Ill try when i got home thanks man keep it up good job 👍👍👍

MoshiMoshi0, 2019-04-28:
Yea it is also zipped. 

If you get to trying it make sure you stop the official software. You could first check for any errors by running it in the console instead of installing it. There is a "Run in console" option in the menu when you launch the service exe.

extayam, 2019-04-28:
Hi, I tried running the latest version TTController_0.6.0.0-dev, but as u suggested when i run in console it stuck at "Creating Device Manager".

In General tab it says  "The Thermaltake Controller service is marked as an interactive service.  However, the system is configured to not allow interactive services.  This service may not function properly."

- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="Service Control Manager" Guid="{555908d1-a6d7-4695-8e1e-26931d2012f4}" EventSourceName="Service Control Manager" /> 
  <EventID Qualifiers="49152">7030</EventID> 
  <Version>0</Version> 
  <Level>2</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8080000000000000</Keywords> 
  <TimeCreated SystemTime="2019-04-28T15:17:52.919904500Z" /> 
  <EventRecordID>5313</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="704" ThreadID="5192" /> 
  <Channel>System</Channel> 
  <Computer>DESKTOP-1IFSF9T</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data Name="param1">Thermaltake Controller</Data> 
  </EventData>
  </Event>

Any tips ? Is it because of my hardware?

MoshiMoshi0, 2019-04-28:
Could you list what thermaltake devices you have?

> In General tab it says "The Thermaltake Controller service is marked as an interactive service. However, the system is configured to not allow interactive services. This service may not function properly."

Yes this is what it says for me too but it works just fine.

extayam, 2019-04-29:
Im using the Riing 12 Radiator Fan TT Premium Edition 3 fan pack.

> [TTController_0.6.0-alpha.27.f01ab074a4ddd974b8abc9cafeaefc675d57eaec.zip](https://github.com/MoshiMoshi0/TTController/files/3125347/TTController_0.6.0-alpha.27.f01ab074a4ddd974b8abc9cafeaefc675d57eaec.zip)
Tried using this, but seems when I run in console it get stuck at Searching for "Riing" Controllers.
When i run Hardware Info there is nothing listed. Is it because of the controller? But my fan is running but only with the default red color.

MoshiMoshi0, 2019-04-29:
Please post the whole initialization log. You can select text in the console and right click to copy.

extayam, 2019-04-29:
3:43:16.5235: ================================================================
23:43:16.5655: Initializing...
23:43:16.6115: Loading plugins...
23:43:16.6185: Loading assembly: TTController.Plugin.BlinkEffect [0.6.0.0]
23:43:16.6385: Loading assembly: TTController.Plugin.ByLedColorEffect [0.6.0.0]
23:43:16.6385: Loading assembly: TTController.Plugin.FlowEffect [0.6.0.0]
23:43:16.6565: Loading assembly: TTController.Plugin.FullColorEffect [0.6.0.0]
23:43:16.6565: Loading assembly: TTController.Plugin.PulseEffect [0.6.0.0]
23:43:16.6665: Loading assembly: TTController.Plugin.RawEffect [0.6.0.0]
23:43:16.6665: Loading assembly: TTController.Plugin.RippleEffect [0.6.0.0]
23:43:16.6665: Loading assembly: TTController.Plugin.SnakeEffect [0.6.0.0]
23:43:16.6665: Loading assembly: CSCore [1.2.1.2]
23:43:16.6805: Loading assembly: TTController.Plugin.SoundEffect [0.6.0.0]
23:43:16.6805: Loading assembly: TTController.Plugin.SpectrumEffect [0.6.0.0]
23:43:16.6805: Loading assembly: TTController.Plugin.TemperatureEffect [0.6.0.0]
23:43:16.6805: Loading assembly: TTController.Plugin.WaveEffect [0.6.0.0]
23:43:16.6955: Loading assembly: TTController.Plugin.CopySpeedController [0.6.0.0]
23:43:16.6955: Loading assembly: TTController.Plugin.PwmSpeedController [0.6.0.0]
23:43:16.6955: Loading assembly: TTController.Plugin.StaticSpeedController [0.6.0.0]
23:43:16.6955: Loading assembly: TTController.Plugin.AlwaysTrigger [0.6.0.0]
23:43:16.6955: Loading assembly: TTController.Plugin.LogicTrigger [0.6.0.0]
23:43:16.7095: Loading assembly: TTController.Plugin.OneTimeTrigger [0.6.0.0]
23:43:16.7095: Loading assembly: TTController.Plugin.ProcessTrigger [0.6.0.0]
23:43:16.7095: Loading assembly: TTController.Plugin.PulseTrigger [0.6.0.0]
23:43:16.7095: Loading assembly: TTController.Plugin.TemperatureTrigger [0.6.0.0]
23:43:16.7365: Creating Config Manager...
23:43:16.7415: Loading config...
23:43:17.0016: Loading done...
23:43:17.0116: Creating DataCache...
23:43:17.0116: Creating Sensor Manager...
23:43:18.9508: Detected sensors: 54
23:43:18.9618: Creating Temperature Manager...
23:43:18.9618: Valid sensor identifier: /amdcpu/0/temperature/0
23:43:18.9778: Valid sensor identifier: /amdcpu/0/temperature/1
23:43:18.9778: Valid sensor identifier: /atigpu/0/temperature/0
23:43:18.9918: Valid sensor identifier: /hdd/0/temperature/0
23:43:18.9918: Valid sensor identifier: /hdd/1/temperature/0
23:43:18.9918: Valid sensor identifier: /hdd/2/temperature/0
23:43:19.0108: Creating Effect Manager...
23:43:19.0108: Creating Speed Controller Manager...
23:43:19.0238: Creating Device Manager...
23:43:19.0648: Searching for "Dpsg" controllers
23:43:19.0978: Searching for "Riing" controllers

Here is the log as you requested

MoshiMoshi0, 2019-04-29:
Hmm thats wierd.

Could you edit the TTController.Service.exe.config file and change this line
```xml
        <logger name="*" minLevel="Debug" writeTo="console" enabled="true" />
```
to 
```xml
        <logger name="*" minLevel="Trace" writeTo="console" enabled="true" />
```

and post the output again?

extayam, 2019-04-29:
00:03:05.8443: ================================================================
00:03:05.8853: Initializing...
00:03:05.9303: Loading plugins...
00:03:05.9303: Loading assembly: TTController.Plugin.BlinkEffect [0.6.0.0]
00:03:05.9483: Loading assembly: TTController.Plugin.ByLedColorEffect [0.6.0.0]
00:03:05.9483: Loading assembly: TTController.Plugin.FlowEffect [0.6.0.0]
00:03:05.9483: Loading assembly: TTController.Plugin.FullColorEffect [0.6.0.0]
00:03:05.9651: Loading assembly: TTController.Plugin.PulseEffect [0.6.0.0]
00:03:05.9651: Loading assembly: TTController.Plugin.RawEffect [0.6.0.0]
00:03:05.9761: Loading assembly: TTController.Plugin.RippleEffect [0.6.0.0]
00:03:05.9761: Loading assembly: TTController.Plugin.SnakeEffect [0.6.0.0]
00:03:05.9761: Loading assembly: CSCore [1.2.1.2]
00:03:05.9761: Loading assembly: TTController.Plugin.SoundEffect [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.SpectrumEffect [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.TemperatureEffect [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.WaveEffect [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.CopySpeedController [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.PwmSpeedController [0.6.0.0]
00:03:05.9911: Loading assembly: TTController.Plugin.StaticSpeedController [0.6.0.0]
00:03:06.0091: Loading assembly: TTController.Plugin.AlwaysTrigger [0.6.0.0]
00:03:06.0091: Loading assembly: TTController.Plugin.LogicTrigger [0.6.0.0]
00:03:06.0091: Loading assembly: TTController.Plugin.OneTimeTrigger [0.6.0.0]
00:03:06.0091: Loading assembly: TTController.Plugin.ProcessTrigger [0.6.0.0]
00:03:06.0091: Loading assembly: TTController.Plugin.PulseTrigger [0.6.0.0]
00:03:06.0231: Loading assembly: TTController.Plugin.TemperatureTrigger [0.6.0.0]
00:03:06.0411: Creating Config Manager...
00:03:06.0411: Loading config...
00:03:06.2571: Loading done...
00:03:06.2571: Creating DataCache...
00:03:06.2751: Creating Sensor Manager...
00:03:07.9688: Detected sensors: 54
00:03:07.9848: Creating Temperature Manager...
00:03:07.9938: Valid sensor identifier: /amdcpu/0/temperature/0
00:03:07.9938: Valid sensor identifier: /amdcpu/0/temperature/1
00:03:07.9938: Valid sensor identifier: /atigpu/0/temperature/0
00:03:08.0068: Valid sensor identifier: /hdd/0/temperature/0
00:03:08.0068: Valid sensor identifier: /hdd/1/temperature/0
00:03:08.0068: Valid sensor identifier: /hdd/2/temperature/0
00:03:08.0068: Creating Effect Manager...
00:03:08.0248: Creating Speed Controller Manager...
00:03:08.0248: Creating Device Manager...
00:03:08.0638: Searching for "Dpsg" controllers
00:03:08.0798: Searching for "Riing" controllers
00:03:08.0918: W[9802, 8001] 0, 254, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

Edited the file as you instructed and here is the output. Sorry man for taking your time :"D

MoshiMoshi0, 2019-04-29:
Nah, no worries 😄 

I think you are the first one with the first gen controllers so they have not been tested. I have to find out how they initialize those, because it hangs waiting for a reply from the controller. 

I'll post a release for you to test one I figure it out.

extayam, 2019-04-29:
Oh how do you know that mine was a first gen controllers? Sure man ill wait thanks for your tips and help :D

MoshiMoshi0, 2019-04-29:
Because your controller has the lowest product id (8001) that is supported by the official software.
There is also 8101 that is second gen, 8501 that is riing trio and 8601 that is an unreleased controller.

It seems like there is nothing special they do and it should work just fine, so I dont really know why it is hanging. Are you sure that any official software is not running? How about the old version from the forums? Do you have the controllers dip switches correctly set? Is the official software working if you use it?

I've added a timeout so it does not hang but just returns an error, post log again running this:
[TTController_0.6.0-alpha.30.f14b7f7669c319f6457000f90a06f9d606850681.zip](https://github.com/MoshiMoshi0/TTController/files/3128364/TTController_0.6.0-alpha.30.f14b7f7669c319f6457000f90a06f9d606850681.zip)

extayam, 2019-04-29:
Oh i see noted. Yeah all of the official software I installed, they only run in task manager and there is no window or anything to be seen. The error I get in my first post here is from the zip file you give in the forum. The controller dip is the default setting, its set to no.1.

01:17:15.4499: ================================================================
01:17:15.4968: Initializing...
01:17:15.5437: Loading plugins...
01:17:15.5437: Loading assembly: TTController.Plugin.BlinkEffect [0.6.0.0]
01:17:15.5593: Loading assembly: TTController.Plugin.ByLedColorEffect [0.6.0.0]
01:17:15.5593: Loading assembly: TTController.Plugin.FlowEffect [0.6.0.0]
01:17:15.5749: Loading assembly: TTController.Plugin.FullColorEffect [0.6.0.0]
01:17:15.5749: Loading assembly: TTController.Plugin.PulseEffect [0.6.0.0]
01:17:15.5749: Loading assembly: TTController.Plugin.RawEffect [0.6.0.0]
01:17:15.5749: Loading assembly: TTController.Plugin.RippleEffect [0.6.0.0]
01:17:15.6062: Loading assembly: TTController.Plugin.SnakeEffect [0.6.0.0]
01:17:15.6062: Loading assembly: CSCore [1.2.1.2]
01:17:15.6218: Loading assembly: TTController.Plugin.SoundEffect [0.6.0.0]
01:17:15.6218: Loading assembly: TTController.Plugin.SpectrumEffect [0.6.0.0]
01:17:15.6218: Loading assembly: TTController.Plugin.TemperatureEffect [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.WaveEffect [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.CopySpeedController [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.PwmSpeedController [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.StaticSpeedController [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.AlwaysTrigger [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.LogicTrigger [0.6.0.0]
01:17:15.6374: Loading assembly: TTController.Plugin.OneTimeTrigger [0.6.0.0]
01:17:15.6531: Loading assembly: TTController.Plugin.ProcessTrigger [0.6.0.0]
01:17:15.6531: Loading assembly: TTController.Plugin.PulseTrigger [0.6.0.0]
01:17:15.6843: Loading assembly: TTController.Plugin.TemperatureTrigger [0.6.0.0]
01:17:15.6843: Creating Config Manager...
01:17:15.6999: Loading config...
01:17:15.6999: Config does not exist! Creating default config...
01:17:15.7156: Saving config...
01:17:16.3250: Saving done...
01:17:16.3406: Loading done...
01:17:16.3406: Creating DataCache...
01:17:16.3562: Creating Sensor Manager...
01:17:20.8566: Detected sensors: 54
01:17:20.8566: Creating Temperature Manager...
01:17:20.8722: Valid sensor identifier: /amdcpu/0/temperature/0
01:17:20.8722: Valid sensor identifier: /amdcpu/0/temperature/1
01:17:20.8722: Valid sensor identifier: /atigpu/0/temperature/0
01:17:20.8722: Valid sensor identifier: /hdd/0/temperature/0
01:17:20.8722: Valid sensor identifier: /hdd/1/temperature/0
01:17:20.8878: Valid sensor identifier: /hdd/2/temperature/0
01:17:20.8878: Creating Effect Manager...
01:17:20.8878: Creating Speed Controller Manager...
01:17:20.9035: Creating Device Manager...
01:17:20.9972: Searching for "Dpsg" controllers
01:17:21.1468: Searching for "Riing" controllers
01:17:21.1624: Found 1 controller [9802, 8001]
01:17:21.1780: W[9802, 8001] 0, 254, 51, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

Here is the log with the new zip file you give :D

MoshiMoshi0, 2019-04-29:
> Yeah all of the official software I installed, they only run in task manager and there is no window or anything to be seen

Oh, so they dont show up in the task bar? Did you kill them in the task manager before running the ttcontroller?

If even the official software hangs then either you controller is dead or there is some motherboard issue, try maybe using another usb header on the motherboard?

extayam, 2019-04-30:
Yeah I did kill all of them before running ur program, and I even uninstalled them.
I tried switching to another working usb header which im using for my pc casing front usb, and still the same

MoshiMoshi0, 2019-04-30:
But does the controller work with the official software? Or was working before?

Because if it doesnt, I dont think there is anything I can do to fix this.

extayam, 2019-04-30:
Yeah its not working too i guess. Coz the app wont even load, only show in the task manager. My fan is all running but only with the default red LED. Nvm then I guess I just have to buy another. Should have research earlier about their software problem. Rip money. Thanks man for your help and sorry for taking much of your time.

MoshiMoshi0, 2019-04-30:
Yeah I think its either broken, or somethings up with the usb stuff ☹️ 

You can try sending it for RMA but dont know how long it will take knowing TT.

I'm closing this issue but feel free to open another one if you need help. Or comment here that you got the official software working and I'll reopen.

closed 2019-04-30 by MoshiMoshi0