MoshiMoshi0/TTController #14: Certain RGB values failing

jlaineinat0r, 2019-04-30:
### Prerequisites
* [ x] The controller is recognized without issues in the official software
* [x ] I made sure no official software or any other software (e.g. Aida64) that can communicate with the controllers was running when the issue with the service occurred

**Describe the issue**
I've been tinkering around with colors and am running into variants that cause the controller software to crash, one example:

With this config:
      "Effects": [
        {
          "Type": "RippleEffect",
          "Config": {
            "Length": 5,
            "Color": [5, 95, 173],
            "TickInterval": 5,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]

I end up with an appcrash: 
Application: TTController.Service.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.OverflowException
   at System.Convert.ToByte(Int32)
   at TTController.Common.LedColor.FromHsv(Double, Double, Double)
   at TTController.Plugin.RippleEffect.RippleEffect.GenerateColors(System.Collections.Generic.List`1<TTController.Common.PortIdentifier>, TTController.Common.ICacheProvider)
   at TTController.Service.TTService.DeviceRgbTimerCallback()
   at TTController.Service.Manager.TimerManager+<>c__DisplayClass3_0.<RegisterTimer>b__0(System.Object, TimerElapsedEventArgs)
   at TTController.Service.Manager.TimerManager+Timer.ExecuteTimer()
   at System.Threading.ThreadHelper.ThreadStart_Context(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()

But this particular hue works:
      "Effects": [
        {
          "Type": "RippleEffect",
          "Config": {
            "Length": 5,
            "Color": [32, 147, 248],
            "TickInterval": 5,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]

MoshiMoshi0, 2019-04-30:
Yes there was a bug in RippleEffect, thanks for reporting.

I'll push a new release soon with the fix.

MoshiMoshi0, 2019-04-30:
Pushed 0.6.1.0-dev release, it should fix this issue

closed 2019-04-30 by jlaineinat0r