MoshiMoshi0/TTController #15: Configuration Help

jlaineinat0r, 2019-05-10:
**Describe your desired configuration**
Can you validate this layout for me? I've attempted to upgrade and can't seem to get the changes to function - .61 is causing high CPU usage for me (~5%) so hoping moving to .7 may remedy.

**Hardware info**
```
Controllers
-------------------------------
Name: Riing Plus
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 53, Rpm: 970, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 53, Rpm: 980, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 53, Rpm: 983, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 100, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 53, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SnakeEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll

Sensors
-------------------------------
Intel Core i9-9900K:
        CPU Core #1 (/intelcpu/0/temperature/0): 35
        CPU Core #2 (/intelcpu/0/temperature/1): 36
        CPU Core #3 (/intelcpu/0/temperature/2): 32
        CPU Core #4 (/intelcpu/0/temperature/3): 32
        CPU Core #5 (/intelcpu/0/temperature/4): 32
        CPU Core #6 (/intelcpu/0/temperature/5): 33
        CPU Core #7 (/intelcpu/0/temperature/6): 32
        CPU Core #8 (/intelcpu/0/temperature/7): 39
        CPU Package (/intelcpu/0/temperature/8): 39
NVIDIA GeForce GTX 1080 Ti:
        GPU Core (/nvidiagpu/0/temperature/0): 27
WDC WD30EFRX-68AX9N0:
        Temperature (/hdd/0/temperature/0): 37
WDC WD30EFRX-68AX9N0:
        Temperature (/hdd/1/temperature/0): 37
Samsung SSD 850 EVO 500GB:
        Temperature (/hdd/2/temperature/0): 26
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Main",
      "Guid": "01c14011-919a-478e-8941-0949115b2887",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
	[9802, 8101, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 40],
              [35, 50],
              [40, 55],
              [45, 60],
              [50, 65],
              [55, 70],
              [60, 75],
              [65, 80],
              [70, 85],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8","/nvidiagpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RippleEffect",
          "Config": {
            "Length": 5,
            "Color": [0, 1, 26],
            "TickInterval": 5,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "4522b673-2f58-469c-bbad-6fa2a3487724",
      "Ports": [
        [9802, 8101, 4]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "ColorGradient": [
              [30, [52, 52, 52]],
              [70, [105, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 5]],
      "Config": {
      "Name": "CPU Block",
      "LedCount": 12,
      "LedCountHandling" : "DoNothing",
      "LedRotation": 10,
      "LedReverse": false
    }
    },
    {
      "Ports": [[9802, 8101, 4]],
      "Config": {
      "Name": "Pump",
      "LedCount": 12,
      "LedCountHandling" : "DoNothing",
      "LedRotation": 0,
      "LedReverse": false
    }
    },
    {
      "Ports": [[9802, 8101, 3]],
      "Config": {
      "Name": "Top Fan",
      "LedCount": 9,
      "LedCountHandling" : "DoNothing",
      "LedRotation": 0,
      "LedReverse": true
    }
    },
    {
      "Ports": [[9802, 8101, 2]],
      "Config": {
      "Name": "Middle Fan",
      "LedCount": 9,
      "LedCountHandling" : "DoNothing",
      "LedRotation": 2,
      "LedReverse": true
    }
    },
    {
      "Ports": [[9802, 8101, 1]],
      "Config": {
      "Name": "Bottom Fan",
      "LedCount": 9,
      "LedCountHandling" : "DoNothing",
      "LedRotation": 4,
      "LedReverse": true
    }
    }
  ],
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

MoshiMoshi0, 2019-05-10:
Yes it does fix it, its caused by hidlibrary. That is why I switched to custom forks of the thirdparty libraries.

But what is the problem? I only see that for some reason `LedCountHandling` is always deserialized to `Trim`. Other than that it seems everything is working correctly.

jlaineinat0r, 2019-05-10:
Sorry, should have posted this with the rest. :)

This was the app crash I couldn't work around on my device.

```
Application: TTController.Service.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at TTController.Service.TTService.<DeviceSpeedTimerCallback>b__19_0(OpenHardwareMonitor.Hardware.Identifier)
   at System.Linq.Enumerable.Any[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]](System.Collections.Generic.IEnumerable`1<System.__Canon>, System.Func`2<System.__Canon,Boolean>)
   at TTController.Service.TTService.DeviceSpeedTimerCallback()
   at TTController.Service.Manager.TimerManager+<>c__DisplayClass3_0.<RegisterTimer>b__0(System.Object, TimerElapsedEventArgs)
   at TTController.Service.Manager.TimerManager+Timer.ExecuteTimer()
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()
```

MoshiMoshi0, 2019-05-10:
Fixed, try this build:
[TTController_0.7.0.0-dev.d004df47aaffb161577961fcc816b483f90cd086.zip](https://github.com/MoshiMoshi0/TTController/files/3166426/TTController_0.7.0.0-dev.d004df47aaffb161577961fcc816b483f90cd086.zip)

If it works without issues it will be released as 0.7.0.1-dev

jlaineinat0r, 2019-05-10:
Works, updated version also cut CPU consumption by about 90% - all good news!

closed 2019-05-10 by jlaineinat0r