MoshiMoshi0/TTController #16: Razer support

FearNaBoinne, 2019-05-16:
I would be great if a module could be created to enable sync with Razer Chroma, just like the crappy official TT app

MoshiMoshi0, 2019-05-16:
I dont have anything from razer so I dont know how the sync works, but do you mean you want TTController to set rgb for razer products, or you want the razer software to send rgb colors to TTController?

The former is too much work and is out of scope for this project, that would require a big rewrite, if not complete rewrite.
The latter should be doable with a custom effect.

FearNaBoinne, 2019-05-16:
Respond to Razer control... Just like the latest version of the TT software does... But like everything else, TT have done this half-arsed. Aside from still not fixing the hibernation issue, it also requires you to MANUALLY turn on Razer sync at each restart of the software...

MoshiMoshi0, 2019-05-16:
Should be doable. 

Seems like its part of the "Razer Chroma Connected Devices Program" and its only for other device manufacturers. From a quick search I could not find any libraries/apis for that so I might have to dig into the TT software.

I will update if I find anything interesting.

FearNaBoinne, 2019-05-16:
RGB.net (another project) have an app RGBsync+ that works with it... Don't know whether there's source for it available... Will have a look and let you know!

FearNaBoinne, 2019-05-16:
https://github.com/DarthAffe/RGB.NET
and
https://github.com/DarthAffe/RGBSyncPlus

MoshiMoshi0, 2019-05-16:
Hmm can't find anything specific to chroma connect.
Thermaltake software does not use "RzChromaSDK.dll" like RGB.NET.

Is this how you enable RBGsync+/RBG.NET in razer synapse?:
https://www.youtube.com/watch?v=X9mxPcXLo6A&t=18s

Btw could you download DbgView or DebugView++ and post the debug log from the TT software? Start it before you enable the razer sync, then try enabling it and changing few effects in synapse.

FearNaBoinne, 2019-05-17:
Yes that's how it works, and I will run the debugger for ya...

Anything special you want me to look for (I'm a (former) developer, and speak C/C++/C#, Python, PHP and a few other dialects, but Java isn't one of my strong points, so I know my way around debuggers and code...)?

MoshiMoshi0, 2019-05-17:
Well, I would like to see the debug output of the TT software with DebugView++, they log a lot of stuff there. 

And if you want to go debugging then if you can find where RGB.NET receives color data from razer synapse that would be cool because im not seeing it. From what I can tell RGB.NET/RGBSyncPlus is only used for pushing effects to razer synapse not receiving them.

FearNaBoinne, 2019-05-17:
At 15.98 seconds (approx) I enabled Razer Sync...

FearNaBoinne, 2019-05-17:
Looking at the debug log, I did not see anything related to the Razer color stuff, so I did another debug log with only TT RGB Plus.exe events, and changed color profiles on the Razer side, which would push updated info to TT. 
Up to 29 second is it running the Razer Fire emulation program (an orangy-red flickering animation) on the three fans connected, at 29-ish I switched to a situation aware profile (which turned them white-ish) and around 35 seconds I returned to the Fire profile...
I do NOT see anything reflecting that in the TT Debug Logs (attached), so it looks like they forgot to add that to the debug output...  :(

[TT only.zip](https://github.com/MoshiMoshi0/TTController/files/3190690/TT.only.zip)

Anything else you want me to try?

MoshiMoshi0, 2019-05-17:
They seem to log that they enable/connect to the chroma broadcast but not the data received.

This is a long shot but try this:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3192490/RzTest.zip)

I was not able to test this since I cant get razer synapse app to start. This should hopefully show the app in synapse and maybe capture the broadcasted events.

MoshiMoshi0, 2019-05-17:
Sorry, try this:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3192549/RzTest.zip)

I think there is some kind of serial code used for initialization.

FearNaBoinne, 2019-05-18:
I will test soon, busy weekend...  ;)

FearNaBoinne, 2019-05-21:
Just ran it:
PS D:\Stack\Downloads> .\RzTest.exe
MSG: 36, LP: 98299212, WP: 0, HWND: 6357326
MSG: 129, LP: 98299200, WP: 0, HWND: 6357326
MSG: 131, LP: 98299180, WP: 0, HWND: 6357326
MSG: 1, LP: 98299200, WP: 0, HWND: 6357326
Init: 59109745109237760 / 0
PS D:\Stack\Download>

Also, a new "device" has shown up in Synapse under "Connect"... 
[image]

(And in studio, obviously...)
[image]

MoshiMoshi0, 2019-05-21:
Yea if you remove the keys if will "uninstall"

Are there any logs in `C:\Program Files (x86)\Razer\ChromaBroadcast\Logs`?
Also what happens if you swap `HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Razer\ChromaBroadcast\RzTest.exe\Index` from 9 to 8? Also change thermaltakes from 8 to 9.

That is a strange return value.

FearNaBoinne, 2019-05-21:
Log: [RzTest.log](https://github.com/MoshiMoshi0/TTController/files/3203900/RzTest.log)

The two switch positions in the Chroma Studio, until I disable/reenable the TT sync, which reestablishes itself as #8, which then automagically deletes the RZtest entry...

Output from both attempts:
PS D:\Stack\Download> .\RzTest.exe
MSG: 36, LP: 100396492, WP: 0, HWND: 933064
MSG: 129, LP: 100396480, WP: 0, HWND: 933064
MSG: 131, LP: 100396460, WP: 0, HWND: 933064
MSG: 1, LP: 100396480, WP: 0, HWND: 933064
Init: 85286917943328768 / 0
PS D:\Stack\Download> .\RzTest.exe
MSG: 36, LP: 102428700, WP: 0, HWND: 1445498
MSG: 129, LP: 102428688, WP: 0, HWND: 1445498
MSG: 131, LP: 102428668, WP: 0, HWND: 1445498
MSG: 1, LP: 102428688, WP: 0, HWND: 1445498
Init: 81064793292668928 / 0

Do you want me to assign an actual activity to the RZtest device in Chroma Studio?

FearNaBoinne, 2019-05-21:
[TT RGB Plus.log](https://github.com/MoshiMoshi0/TTController/files/3203921/TT.RGB.Plus.log)

Here's the TT log, just for comparison...

FearNaBoinne, 2019-05-21:
When I assign the fire effect to the RzTest device, it animates it similar as the TT RGB Plus device...

MoshiMoshi0, 2019-05-21:
Hmm, this is going to be hard to debug.

Do this: disable the TT sync, remove the TT key under ChromaBroadcast from the registry, change RzTest.exe Index to 8, and then test RzTest.exe

Also I think the return values are just garbage because I set the return type to long, but it looks like broadcast api uses int. Use this:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3204048/RzTest.zip)

Should be working? 🤷‍♂

FearNaBoinne, 2019-05-22:
I removed both entries and ran the new RzTest (twice actually, as the first time my AV intercepted it, and I wasn't certain it ran correctly)

D:\Stack> .\RzTest.exe
MSG: 36, LP: 102820796, WP: 0, HWND: 1446294
MSG: 129, LP: 102820784, WP: 0, HWND: 1446294
MSG: 131, LP: 102820764, WP: 0, HWND: 1446294
MSG: 1, LP: 102820784, WP: 0, HWND: 1446294
Init: SUCCESS / 0
RegisterEventNotification: SUCCESS

Log:
[RzTest.log](https://github.com/MoshiMoshi0/TTController/files/3206340/RzTest.log)

Registry section after running the new RzTest
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3206352/RzTest.zip)

Hope you find something useful!

MoshiMoshi0, 2019-05-22:
Ok so it seems to be working, now you can test what happens when you assign effects to RzTest. You should see messages with rgb colors.

FearNaBoinne, 2019-05-22:
Where should I see the messages exactly?

FearNaBoinne, 2019-05-22:
Ok, update: When I run RzTest with Taskmanager open, I see Windows Problem Reporter run along in the background, before RzTest terminates... I am assuming now - from your 'messages' statement - that it should continue to run and show me stuff?

Anyway, here's the Problem Reporter detail (in Dutch, sorry):

Bron
Sandbox

Overzicht
Werkt niet meer

Datum
‎22-‎5-‎2019 11:31

Status
Verzonden rapport

Beschrijving
Pad naar toepassing met fout:	D:\Stack\RzTest.exe

Probleem met handtekening
Naam van probleemgebeurtenis:	BEX
Naam van de toepassing:	RzTest.exe
Versie van toepassing:	1.0.0.0
Tijdstempel van toepassing:	5ce44aff
Naam van foutmodule:	StackHash_2beb
Versie van foutmodule:	0.0.0.0
Tijdstempel van foutmodule:	00000000
Uitzonderingsmarge:	PCH_AB_FROM_ntdll+0x0006AE8C
Uitzonderingscode:	c0000005
Uitzonderingsgegevens:	00000008
Versie van besturingssysteem:	10.0.17134.2.0.0.256.48
Landinstelling-id:	1043
Aanvullende informatie 1:	2beb
Aanvullende informatie 2:	2beba6fb4680d73a8c78ca7c24ccdb46
Aanvullende informatie 3:	1ca5
Aanvullende informatie 4:	1ca5ddcf4f5843895cc5b9d12f08468c

Extra informatie over het probleem
Bucket-id:	0dbbbdd8030c424affe422cf55951236 (2297999983755203126)

Does this help any?

FearNaBoinne, 2019-05-22:
(Given that it is a BEX event, I would hazard a guess that somewhere you are using an incorrect size buffer/variable to store feedback from the Razer SDK?)

MoshiMoshi0, 2019-05-22:
There should be another event that shows the stacktrace, I need to know which line is causing the crash.

The console should be running all the time and it will output the messages. To exit you have to press any key twice (first dispose/uninit, second exit)

FearNaBoinne, 2019-05-22:
All events in chronological order:

----------------
Toepassing: RzTest.exe
Framework-versie: v4.0.30319
Beschrijving: het proces is beëindigd als gevolg van een onverwerkte uitzondering.
Uitzonderingsinformatie: uitzonderingscode c0000005, uitzonderingsadres 003A09D2
----------------
Naam van toepassing met fout: RzTest.exe, versie: 1.0.0.0, tijdstempel: 0x5ce44aff
Naam van module met fout: unknown, versie: 0.0.0.0, tijdstempel: 0x00000000
Uitzonderingscode: 0xc0000005
Foutmarge: 0x003a09d2
Id van proces met fout: 0x1e08
Starttijd van toepassing met fout: 0x01d510812b9bc283
Pad naar toepassing met fout: D:\Stack\RzTest.exe
Pad naar module met fout: unknown
Rapport-id: 8f6ecf4a-e957-439e-86d6-dc84b6e28dec
Volledige pakketnaam met fout: 
Relatieve toepassings-id van pakket met fout: 
----------------
Foutbucket 2297999983755203126, type 5
Naam van gebeurtenis: BEX
Antwoord: Niet beschikbaar
Id van CAB-bestand: 0

Handtekening van probleem:
P1: RzTest.exe
P2: 1.0.0.0
P3: 5ce44aff
P4: StackHash_2beb
P5: 0.0.0.0
P6: 00000000
P7: PCH_AB_FROM_ntdll+0x0006AE8C
P8: c0000005
P9: 00000008
P10: 

Toegevoegde bestanden:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER1D58.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER28A3.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER297F.tmp.xml

Deze bestanden zijn mogelijk hier beschikbaar:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_RzTest.exe_d32c6598e6befc9bb7d61c874e7ec195a4553bec_866f5d7c_09222db3

Analysesymbool: 
Opnieuw zoeken naar oplossing: 0
Rapport-id: 8f6ecf4a-e957-439e-86d6-dc84b6e28dec
Rapportstatus: 268435456
Opgedeelde bucket: 0dbbbdd8030c424affe422cf55951236
GUID van CAB-bestand: 0
----------------

Finally the WER report:
[Report.zip](https://github.com/MoshiMoshi0/TTController/files/3206785/Report.zip)

There are no dump files saved, AFAICT...

FearNaBoinne, 2019-05-22:
All events in chronological order:

:arrow_forward:
Toepassing: RzTest.exe
Framework-versie: v4.0.30319
Beschrijving: het proces is beëindigd als gevolg van een onverwerkte uitzondering.
Uitzonderingsinformatie: uitzonderingscode c0000005, uitzonderingsadres 003A09D2
:arrow_forward:
Naam van toepassing met fout: RzTest.exe, versie: 1.0.0.0, tijdstempel: 0x5ce44aff
Naam van module met fout: unknown, versie: 0.0.0.0, tijdstempel: 0x00000000
Uitzonderingscode: 0xc0000005
Foutmarge: 0x003a09d2
Id van proces met fout: 0x1e08
Starttijd van toepassing met fout: 0x01d510812b9bc283
Pad naar toepassing met fout: D:\Stack\RzTest.exe
Pad naar module met fout: unknown
Rapport-id: 8f6ecf4a-e957-439e-86d6-dc84b6e28dec
Volledige pakketnaam met fout: 
Relatieve toepassings-id van pakket met fout: 
:arrow_forward:
Foutbucket 2297999983755203126, type 5
Naam van gebeurtenis: BEX
Antwoord: Niet beschikbaar
Id van CAB-bestand: 0

Handtekening van probleem:
P1: RzTest.exe
P2: 1.0.0.0
P3: 5ce44aff
P4: StackHash_2beb
P5: 0.0.0.0
P6: 00000000
P7: PCH_AB_FROM_ntdll+0x0006AE8C
P8: c0000005
P9: 00000008
P10: 

Toegevoegde bestanden:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER1D58.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER28A3.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER297F.tmp.xml

Deze bestanden zijn mogelijk hier beschikbaar:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_RzTest.exe_d32c6598e6befc9bb7d61c874e7ec195a4553bec_866f5d7c_09222db3

Analysesymbool: 
Opnieuw zoeken naar oplossing: 0
Rapport-id: 8f6ecf4a-e957-439e-86d6-dc84b6e28dec
Rapportstatus: 268435456
Opgedeelde bucket: 0dbbbdd8030c424affe422cf55951236
GUID van CAB-bestand: 0
:arrow_forward:

Finally the WER report:
[Report.zip](https://github.com/MoshiMoshi0/TTController/files/3206785/Report.zip)

There are no dump files saved, AFAICT...

MoshiMoshi0, 2019-05-22:
Hmm, well I need the stacktrace to know where it breaks. Was there an exception shown in the console windows? What was the last printed line?

FearNaBoinne, 2019-05-22:
no error or exception.

Last line was 
RegisterEventNotification: SUCCESS

After that the prompt was shown. 
Actually the output in comment https://github.com/MoshiMoshi0/TTController/issues/16#issuecomment-494502543 was verbatim from the console, and shows exactly the sequence. 

Below the verbatim output from the last 4 attempts (note that #2 stuck around long enough to post the unregister and uninit messages)

PS D:\Stack> .\RzTest.exe
MSG: 36, LP: 102820796, WP: 0, HWND: 1446294
MSG: 129, LP: 102820784, WP: 0, HWND: 1446294
MSG: 131, LP: 102820764, WP: 0, HWND: 1446294
MSG: 1, LP: 102820784, WP: 0, HWND: 1446294
Init: SUCCESS / 0
RegisterEventNotification: SUCCESS
PS D:\Stack> .\RzTest.exe
MSG: 36, LP: 87486268, WP: 0, HWND: 14680938
MSG: 129, LP: 87486256, WP: 0, HWND: 14680938
MSG: 131, LP: 87486236, WP: 0, HWND: 14680938
MSG: 1, LP: 87486256, WP: 0, HWND: 14680938
Init: SUCCESS / 0
RegisterEventNotification: SUCCESS
UnregisterEventNotification: SUCCESS
UnInit: SUCCESS
PS D:\Stack> .\RzTest.exe
MSG: 36, LP: 102166412, WP: 0, HWND: 4918734
MSG: 129, LP: 102166400, WP: 0, HWND: 4918734
MSG: 131, LP: 102166380, WP: 0, HWND: 4918734
MSG: 1, LP: 102166400, WP: 0, HWND: 4918734
Init: SUCCESS / 0
RegisterEventNotification: SUCCESS
PS D:\Stack> .\RzTest.exe
MSG: 36, LP: 90172940, WP: 0, HWND: 3803602
MSG: 129, LP: 90172928, WP: 0, HWND: 3803602
MSG: 131, LP: 90172908, WP: 0, HWND: 3803602
MSG: 1, LP: 90172928, WP: 0, HWND: 3803602
Init: SUCCESS / 0
RegisterEventNotification: SUCCESS
PS D:\Stack>

FearNaBoinne, 2019-05-22:
By being fast (and using 1 CLI and 2 Explorer Windows next to eachother, run, Ctrl-A, Ctrl-C and Ctrl-V combo's) I was able to grab these from the WER Temp folder before they were deleted again:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3206884/RzTest.zip)
These go with this WER report:
[AppCrash_RzTest.exe_d32c6598e6befc9bb7d61c874e7ec195a4553bec_866f5d7c_69828951.zip](https://github.com/MoshiMoshi0/TTController/files/3206889/AppCrash_RzTest.exe_d32c6598e6befc9bb7d61c874e7ec195a4553bec_866f5d7c_69828951.zip)

Hope this helps a bit more!

MoshiMoshi0, 2019-05-22:
Nah, that is all generic exception stuff.

I've changed the only thing that can throw buffer overflow:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3206900/RzTest.zip)

Btw do you have visual studio setup? I think it would be much faster if I could send you the project and you would run it.

FearNaBoinne, 2019-05-22:
No Visual Studio, but I can set it up, plenty of spare space on my drive!  ;)

Will test the latest RzTest

FearNaBoinne, 2019-05-22:
Same result!  :(

I will do a reboot, just to make certain everything is clear, and if that doesn't help, I will go grab VS

MoshiMoshi0, 2019-05-22:
You can also try this: 
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3207040/RzTest.zip)

Changed the same thing to a different method.

MoshiMoshi0, 2019-05-22:
And source if above also doesn't work:
[Sandbox.zip](https://github.com/MoshiMoshi0/TTController/files/3207071/Sandbox.zip)

FearNaBoinne, 2019-05-22:
[image]

[image]

[image]

So the app remains running since reboot, but no messages... It identifies itself as TT, and shows up like that in Synapse, but with a generic icon instead of the TT icon...

FearNaBoinne, 2019-05-22:
Downloading the VS IDE now!

FearNaBoinne, 2019-05-22:
PS: What components do I need for the project?

MoshiMoshi0, 2019-05-22:
I guess you only need c#, you can leave the installer at defaults and it will be fine.

If it does not crash anymore than I would think it is working, its just not getting any messages. But dunno why.

FearNaBoinne, 2019-05-23:
I did apply and save the effect... Even change profiles to enforce it to be (re-)applied...

FearNaBoinne, 2019-05-23:
I will try to have a look later today at the source and how it responds

MoshiMoshi0, 2019-05-27:
Ok, I got it working. 

Try this:
[RzTest.zip](https://github.com/MoshiMoshi0/TTController/files/3224762/RzTest.zip)

Im only getting the message "Chroma Broadcast Offline" but I guess that is because I dont have any razer devices. I had to modify the api dll assembly to skip the device present check.

FearNaBoinne, 2019-05-28:
Methinks you were successful... 

As you probably noted, Razer supports up to 5 LEDs for the broadcast connection, so below some output from the 'fire' emulation, with 5 values each broadcast (1 for each virtualized LED):

D:\Stack> .\RzTest.exe
Init: SUCCESS
RegisterEventNotification: SUCCESS
LEN: 5
[195, 110, 0]
[71, 34, 0]
[195, 110, 0]
[84, 40, 0]
[157, 75, 0]
Chroma Broadcast Online
LEN: 5
[189, 106, 0]
[51, 24, 0]
[189, 106, 0]
[60, 28, 0]
[145, 69, 0]
LEN: 5
[189, 106, 0]
[51, 24, 0]
[189, 106, 0]
[60, 28, 0]
[145, 69, 0]
LEN: 5
[195, 110, 0]
[52, 25, 0]
[195, 110, 0]
[85, 41, 0]
[138, 66, 0]
LEN: 5
[202, 114, 0]
[54, 26, 0]
[202, 114, 0]
[111, 53, 0]
[132, 63, 0]
LEN: 5
[209, 118, 0]
[56, 27, 0]
[209, 118, 0]
[136, 65, 0]
[125, 60, 0]
LEN: 5
[216, 121, 0]
[58, 27, 0]
[216, 121, 0]
[162, 78, 0]
[119, 57, 0]
LEN: 5
[216, 121, 0]
[58, 27, 0]
[216, 121, 0]
[162, 78, 0]
[119, 57, 0]
LEN: 5
[195, 110, 0]
[78, 37, 0]
[195, 110, 0]
[132, 63, 0]
[122, 58, 0]
LEN: 5
[175, 98, 0]
[99, 47, 0]
[175, 98, 0]
[102, 49, 0]
[125, 60, 0]
LEN: 5
[155, 87, 0]
[119, 57, 0]
[155, 87, 0]
[72, 34, 0]
[128, 61, 0]
LEN: 5
[135, 76, 0]
[140, 67, 0]
[135, 76, 0]
[42, 20, 0]
[132, 63, 0]
LEN: 5
[135, 76, 0]
[140, 67, 0]
[135, 76, 0]
[42, 20, 0]
[132, 63, 0]
LEN: 5
[137, 77, 0]
[147, 70, 0]
[137, 77, 0]
[77, 37, 0]
[118, 56, 0]
LEN: 5
[140, 79, 0]
[154, 74, 0]
[140, 79, 0]
[113, 54, 0]
[105, 50, 0]
LEN: 5
[142, 80, 0]
[161, 77, 0]
[142, 80, 0]
[148, 71, 0]
[92, 44, 0]
LEN: 5
[145, 81, 0]
[168, 81, 0]
[145, 81, 0]
[184, 88, 0]
[79, 38, 0]
LEN: 5
[145, 81, 0]
[168, 81, 0]
[145, 81, 0]
[184, 88, 0]
[79, 38, 0]
LEN: 5
[155, 87, 0]
[168, 81, 0]
[155, 87, 0]
[155, 74, 0]
[87, 41, 0]
LEN: 5
[165, 93, 0]
[169, 81, 0]
[165, 93, 0]
[127, 61, 0]
[96, 46, 0]
LEN: 5
[175, 98, 0]
[169, 81, 0]
[175, 98, 0]
[99, 47, 0]
[104, 50, 0]
LEN: 5
[185, 104, 0]
[170, 82, 0]
[185, 104, 0]
[71, 34, 0]
[113, 54, 0]
LEN: 5
[185, 104, 0]
[170, 82, 0]
[185, 104, 0]
[71, 34, 0]
[113, 54, 0]
LEN: 5
[190, 107, 0]
[138, 66, 0]
[190, 107, 0]
[103, 49, 0]
[93, 44, 0]
LEN: 5
[196, 110, 0]
[107, 51, 0]
[196, 110, 0]
[135, 65, 0]
[73, 35, 0]
LEN: 5
[201, 113, 0]
[76, 36, 0]
[201, 113, 0]
[167, 80, 0]
[53, 25, 0]
LEN: 5
[207, 116, 0]
[45, 21, 0]
[207, 116, 0]
[199, 95, 0]
[33, 15, 0]
LEN: 5
[207, 116, 0]
[45, 21, 0]
[207, 116, 0]
[199, 95, 0]
[33, 15, 0]
LEN: 5
[163, 92, 0]
[60, 28, 0]
[163, 92, 0]
[163, 78, 0]
[41, 19, 0]
LEN: 5
[120, 67, 0]
[76, 36, 0]
[120, 67, 0]
[128, 61, 0]
[49, 23, 0]
LEN: 5
[77, 43, 0]
[92, 44, 0]
[77, 43, 0]
[93, 44, 0]
[57, 27, 0]
LEN: 5
[34, 19, 0]
[108, 52, 0]
[34, 19, 0]
[58, 27, 0]
[65, 31, 0]
LEN: 5
[34, 19, 0]
[108, 52, 0]
[34, 19, 0]
[58, 27, 0]
[65, 31, 0]
LEN: 5
[69, 38, 0]
[92, 44, 0]
[69, 38, 0]
[84, 40, 0]
[94, 45, 0]
LEN: 5
[105, 59, 0]
[76, 36, 0]
[105, 59, 0]
[110, 53, 0]
[123, 59, 0]
LEN: 5
[141, 79, 0]
[60, 28, 0]
[141, 79, 0]
[136, 65, 0]
[152, 73, 0]
LEN: 5
[177, 99, 0]
[45, 21, 0]
[177, 99, 0]
[163, 78, 0]
[181, 87, 0]
LEN: 5
[177, 99, 0]
[45, 21, 0]
[177, 99, 0]
[163, 78, 0]
[181, 87, 0]
LEN: 5
[161, 90, 0]
[69, 33, 0]
[161, 90, 0]
[169, 81, 0]
[175, 84, 0]
LEN: 5
[145, 81, 0]
[93, 44, 0]
[145, 81, 0]
[176, 84, 0]
[170, 82, 0]
LEN: 5
[129, 72, 0]
[117, 56, 0]
[129, 72, 0]
[182, 87, 0]
[164, 79, 0]
LEN: 5
[114, 64, 0]
[141, 68, 0]
[114, 64, 0]
[189, 91, 0]
[159, 76, 0]
LEN: 5
[114, 64, 0]
[141, 68, 0]
[114, 64, 0]
[189, 91, 0]
[159, 76, 0]
LEN: 5
[127, 71, 0]
[131, 63, 0]
[127, 71, 0]
[183, 88, 0]
[173, 83, 0]
LEN: 5
[140, 79, 0]
[122, 58, 0]
[140, 79, 0]
[177, 85, 0]
[188, 90, 0]
LEN: 5
[153, 86, 0]
[113, 54, 0]
[153, 86, 0]
[171, 82, 0]
[203, 97, 0]
LEN: 5
[166, 93, 0]
[104, 50, 0]
[166, 93, 0]
[166, 80, 0]
[218, 105, 0]
LEN: 5
[166, 93, 0]
[104, 50, 0]
[166, 93, 0]
[166, 80, 0]
[218, 105, 0]
LEN: 5
[174, 98, 0]
[122, 58, 0]
[174, 98, 0]
[169, 81, 0]
[176, 84, 0]
LEN: 5
[182, 102, 0]
[141, 68, 0]
[182, 102, 0]
[172, 82, 0]
[134, 64, 0]
LEN: 5
[190, 107, 0]
[160, 77, 0]
[190, 107, 0]
[175, 84, 0]
[92, 44, 0]
LEN: 5
[199, 112, 0]
[179, 86, 0]
[199, 112, 0]
[178, 85, 0]
[51, 24, 0]
LEN: 5
[199, 112, 0]
[179, 86, 0]
[199, 112, 0]
[178, 85, 0]
[51, 24, 0]
LEN: 5
[169, 95, 0]
[171, 82, 0]
[169, 95, 0]
[148, 71, 0]
[57, 27, 0]
LEN: 5
[140, 79, 0]
[163, 78, 0]
[140, 79, 0]
[119, 57, 0]
[63, 30, 0]
LEN: 5
[110, 62, 0]
[155, 74, 0]
[110, 62, 0]
[90, 43, 0]
[69, 33, 0]
LEN: 5
[81, 45, 0]
[148, 71, 0]
[81, 45, 0]
[61, 29, 0]
[76, 36, 0]
LEN: 5
[81, 45, 0]
[148, 71, 0]
[81, 45, 0]
[61, 29, 0]
[76, 36, 0]
LEN: 5
[73, 41, 0]
[121, 58, 0]
[73, 41, 0]
[91, 43, 0]
[110, 53, 0]
LEN: 5
[65, 36, 0]
[94, 45, 0]
[65, 36, 0]
[122, 58, 0]
[144, 69, 0]
LEN: 5
[57, 32, 0]
[67, 32, 0]
[57, 32, 0]
[153, 73, 0]
[178, 85, 0]
LEN: 5
[49, 27, 0]
[40, 19, 0]
[49, 27, 0]
[184, 88, 0]
[212, 102, 0]
LEN: 5
[49, 27, 0]
[40, 19, 0]
[49, 27, 0]
[184, 88, 0]
[212, 102, 0]
LEN: 5
[79, 44, 0]
[54, 26, 0]
[79, 44, 0]
[153, 73, 0]
[186, 89, 0]

I will try to set up a profile that will allow me to map a single color to each LED, so we can figure out how TT maps them...  ;)

MoshiMoshi0, 2019-05-28:
Nice! Although the blue is always 0, I hope its just because of the fire effect.

You will want to use `LedCountHandling.Lerp` with this anyways to stretch the 5 colors to 12. 

Here is a plugin for the service: 
[TTController.Plugin.RazerConnectEffect.dll.zip](https://github.com/MoshiMoshi0/TTController/files/3228099/TTController.Plugin.RazerConnectEffect.dll.zip)
Put it in the `Plugins` folder.

The config should look something like this:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-516cad5d53c1",
      "Ports": [
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8101, 2],
        [9802, 8101, 1]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8101, 2],
        [9802, 8101, 1]
      ],
      "Config": {
        "LedCountHandling": "Lerp"
      }
    }
  ]
}
```

FearNaBoinne, 2019-05-28:
I do not understand the Razer Ambient Awareness (which I wanted to do separate colors through)... When I do a hard black/white 50/50 and select that as the region, I get muddled greys, even when I choose "sharp"...

Will test some more...

FearNaBoinne, 2019-05-28:
[image]
leads to 
[image]
leads to
[image]
and gives:
D:\Stack> .\RzTest.exe
Init: SUCCESS
RegisterEventNotification: SUCCESS
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
Chroma Broadcast Online
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]
LEN: 5
[200, 214, 56]
[239, 243, 16]
[200, 214, 56]
[122, 154, 139]
[83, 125, 179]

(Hope it makes sense)

FearNaBoinne, 2019-05-28:
I will get the latest installed and test it... GTG finish some work now first ;)

FearNaBoinne, 2019-05-28:
Something is wrong, either in the config file, or in it's interpretation:

2019-05-28 16:55:51.4215|INFO|TTController.Service.TTService|================================================================
2019-05-28 16:55:51.4670|INFO|TTController.Service.TTService|Initializing...
2019-05-28 16:55:51.4874|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
2019-05-28 16:55:51.5154|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
2019-05-28 16:55:51.5154|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
2019-05-28 16:55:51.5154|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
2019-05-28 16:55:51.5154|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
2019-05-28 16:55:51.5154|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
2019-05-28 16:55:51.5314|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [1.0.0.0]
2019-05-28 16:55:51.5314|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
2019-05-28 16:55:51.5314|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
2019-05-28 16:55:51.5314|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SnakeEffect [0.7.0.0]
2019-05-28 16:55:51.5583|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
2019-05-28 16:55:51.5623|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
2019-05-28 16:55:51.5623|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
2019-05-28 16:55:51.5893|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
2019-05-28 16:55:51.5928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
2019-05-28 16:55:51.5928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
2019-05-28 16:55:51.5928|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
2019-05-28 16:55:51.6077|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
2019-05-28 16:55:51.6077|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
2019-05-28 16:55:51.6077|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
2019-05-28 16:55:51.6077|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
2019-05-28 16:55:51.6222|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
2019-05-28 16:55:51.6222|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
2019-05-28 16:55:51.6222|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
2019-05-28 16:55:51.6222|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
2019-05-28 16:55:51.6377|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
2019-05-28 16:55:51.6377|INFO|TTController.Service.Manager.ConfigManager|Creating Config Manager...
2019-05-28 16:55:51.6377|INFO|TTController.Service.Manager.ConfigManager|Loading config...
2019-05-28 16:55:52.0131|FATAL|TTController.Service.Manager.ConfigManager|Reeks bevat geen overeenkomend element
2019-05-28 16:55:52.0131|FATAL|TTController.Service.Manager.ConfigManager|System.InvalidOperationException: Reeks bevat geen overeenkomend element
   bij System.Linq.Enumerable.First[TSource](IEnumerable`1 source, Func`2 predicate)
   bij TTController.Service.Config.Converter.AbstractPluginConverter`2.ReadJson(JsonReader reader, Type objectType, TPlugin existingValue, Boolean hasExistingValue, JsonSerializer serializer)
   bij Newtonsoft.Json.JsonConverter`1.ReadJson(JsonReader reader, Type objectType, Object existingValue, JsonSerializer serializer)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
[... cut ...]
2019-05-28 16:55:52.0131|INFO|TTController.Service.TTService|================================================================
2019-05-28 16:55:52.0131|INFO|TTController.Service.TTService|Finalizing...
2019-05-28 16:55:52.0131|INFO|TTController.Service.Manager.ConfigManager|Disposing ConfigManager...
2019-05-28 16:55:52.0286|INFO|TTController.Service.TTService|Disposing done!
2019-05-28 16:55:52.0286|INFO|TTController.Service.TTService|================================================================

("Reeks bevat geen overeenkomend element" translates to "Series does not contain a matching element")

FearNaBoinne, 2019-05-28:
Config as reported by TTController:

Controllers
-------------------------------
Name: Riing Plus
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 40, Rpm: 761, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 40, Rpm: 736, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 40, Rpm: 782, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SnakeEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll

Sensors
-------------------------------
Intel Core i5-4670K:
        CPU Core #1 (/intelcpu/0/temperature/0): 41
        CPU Core #2 (/intelcpu/0/temperature/1): 42
        CPU Core #3 (/intelcpu/0/temperature/2): 41
        CPU Core #4 (/intelcpu/0/temperature/3): 38
        CPU Package (/intelcpu/0/temperature/4): 41
NVIDIA GeForce GTX 1070 Ti:
        GPU Core (/nvidiagpu/0/temperature/0): 53
SSD PLUS 480GB:
        Temperature (/hdd/0/temperature/0): 28
ST1000DM003-1ER162:
        Temperature (/hdd/1/temperature/0): 32
WDC WDS500G2B0B-00YS70:
        Temperature (/hdd/2/temperature/0): 33
-------------------------------

MoshiMoshi0, 2019-05-28:
Whoops, typo in class name so the service was unable to find it. Fixed:
[TTController.Plugin.RazerConnectEffect.dll.zip](https://github.com/MoshiMoshi0/TTController/files/3228357/TTController.Plugin.RazerConnectEffect.dll.zip)

From the config you can remove ```[9802, 8101, 4]``` and ```[9802, 8101, 5]``` since it seems you have 3 fans.

Also the picture of the fan you posted was with `Copy` or `Lerp` LedCountHandling?

FearNaBoinne, 2019-05-28:
The fan image was from the TT RGB Plus software, so you had an idea of what *it* does...

I will test with the updated DLL

FearNaBoinne, 2019-05-28:
I feel so bad, but...

```2019-05-28 17:19:21.0097|INFO|TTController.Service.TTService|================================================================
2019-05-28 17:19:21.0547|INFO|TTController.Service.TTService|Initializing...
2019-05-28 17:19:21.0801|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
2019-05-28 17:19:21.1136|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
2019-05-28 17:19:21.1136|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
2019-05-28 17:19:21.1271|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
2019-05-28 17:19:21.1271|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
2019-05-28 17:19:21.1271|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
2019-05-28 17:19:21.1410|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [1.0.0.0]
2019-05-28 17:19:21.1410|FATAL|TTController.Service.TTService|System.IO.FileLoadException: Kan bestand of assembly TTController.Plugin.RazerConnectEffect, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null of een van de afhankelijkheden hiervan niet laden. Bewerking wordt niet ondersteund. (Uitzondering van HRESULT: 0x80131515)
Bestandsnaam: TTController.Plugin.RazerConnectEffect, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null ---> System.NotSupportedException: Er is een poging gedaan een assembly te laden vanuit een netwerklocatie. Hierdoor kan de assembly in een sandbox worden geplaatst in eerdere versies van .NET Framework. In deze versie van .NET Framework wordt CAS-beleid niet standaard ingeschakeld. Deze laadbewerking kan daarom gevaarlijk zijn. Als deze laadbewerking niet is bedoeld om de assembly in een sandbox te plaatsen, moet u de schakeloptie loadFromRemoteSources inschakelen. Zie http://go.microsoft.com/fwlink/?LinkId=155569 voor meer informatie.
   bij System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   bij System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   bij System.Reflection.Assembly.Load(AssemblyName assemblyRef)
   bij TTController.Service.Utils.PluginLoader.<>c.<Load>b__3_0(Assembly a)
   bij System.Linq.Enumerable.WhereSelectEnumerableIterator`2.MoveNext()
   bij System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   bij System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   bij TTController.Service.TTService.Initialize()
   bij TTController.Service.TTService.OnStart(String[] args)

2019-05-28 17:19:21.1580|INFO|TTController.Service.TTService|================================================================
2019-05-28 17:19:21.1580|INFO|TTController.Service.TTService|Finalizing...
2019-05-28 17:19:21.1580|INFO|TTController.Service.TTService|Disposing done!
2019-05-28 17:19:21.1580|INFO|TTController.Service.TTService|================================================================
```

Using config:
```{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-516cad5d53c1",
      "Ports": [
        [9802, 8101, 3],
        [9802, 8101, 2],
        [9802, 8101, 1]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 3],
        [9802, 8101, 2],
        [9802, 8101, 1]
      ],
      "Config": {
        "LedCountHandling": "Lerp"
      }
    }
  ]
}```

MoshiMoshi0, 2019-05-28:
Umm, why would you get this error now and not before...

Try this:
* rightclick the dll
* click properties
* click unblock

FearNaBoinne, 2019-05-28:
Yes, it was blocked... Not sure why it marked the last update that way, but that resolved one issue, to show the next one...
*ducks and hides*

```
2019-05-28 17:35:29.0625|INFO|TTController.Service.TTService|================================================================
2019-05-28 17:35:29.0959|INFO|TTController.Service.TTService|Initializing...
2019-05-28 17:35:29.1179|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
2019-05-28 17:35:29.1468|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
2019-05-28 17:35:29.1468|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
2019-05-28 17:35:29.1533|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
2019-05-28 17:35:29.1533|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
2019-05-28 17:35:29.1533|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
2019-05-28 17:35:29.1533|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [1.0.0.0]
2019-05-28 17:35:29.1533|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
2019-05-28 17:35:29.1698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
2019-05-28 17:35:29.1698|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SnakeEffect [0.7.0.0]
2019-05-28 17:35:29.1888|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
2019-05-28 17:35:29.1888|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
2019-05-28 17:35:29.1888|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
2019-05-28 17:35:29.2162|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
2019-05-28 17:35:29.2162|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
2019-05-28 17:35:29.2162|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
2019-05-28 17:35:29.2162|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
2019-05-28 17:35:29.2317|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
2019-05-28 17:35:29.2317|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
2019-05-28 17:35:29.2317|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
2019-05-28 17:35:29.2317|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
2019-05-28 17:35:29.2317|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
2019-05-28 17:35:29.2487|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
2019-05-28 17:35:29.2487|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
2019-05-28 17:35:29.2487|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
2019-05-28 17:35:29.2487|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
2019-05-28 17:35:29.2686|INFO|TTController.Service.Manager.ConfigManager|Creating Config Manager...
2019-05-28 17:35:29.2686|INFO|TTController.Service.Manager.ConfigManager|Loading config...
2019-05-28 17:35:29.6780|FATAL|TTController.Service.Manager.ConfigManager|Het doel van een aanroep heeft een uitzondering veroorzaakt.
2019-05-28 17:35:29.6780|FATAL|TTController.Service.Manager.ConfigManager|System.Reflection.TargetInvocationException: Het doel van een aanroep heeft een uitzondering veroorzaakt. ---> System.ArgumentNullException: Waarde kan niet null zijn.
Parameternaam: ptr
   bij System.Runtime.InteropServices.Marshal.GetDelegateForFunctionPointer(IntPtr ptr, Type t)
   bij TTController.Plugin.RazerConnectEffect.RzChromaBroadcastNative.Load()
   bij TTController.Plugin.RazerConnectEffect.RazerConnectEffect..ctor(RazerConnectEffectConfig config)
   --- Einde van intern uitzonderingsstackpad ---
   bij System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   bij System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   bij System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   bij System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
[... cut ...]
2019-05-28 17:35:29.6860|INFO|TTController.Service.TTService|Finalizing...
2019-05-28 17:35:29.7059|INFO|TTController.Service.Manager.ConfigManager|Disposing ConfigManager...
2019-05-28 17:35:29.7059|INFO|TTController.Service.TTService|Disposing done!
2019-05-28 17:35:29.7059|INFO|TTController.Service.TTService|================================================================
```

Is this line correct:
TTController.Plugin.RazerConnectEffect.RazerConnectEffect..ctor(RazerConnectEffectConfig config)

MoshiMoshi0, 2019-05-28:
Nah no worries, this is what happens when you cant fully test the code you are writing 😛 

It was loading 32bit dll but the service is 64bit. This should maybe work now:
[TTController.Plugin.RazerConnectEffect.dll.zip](https://github.com/MoshiMoshi0/TTController/files/3228566/TTController.Plugin.RazerConnectEffect.dll.zip)

FearNaBoinne, 2019-05-28:
```
19:01:20.9062: ================================================================
19:01:20.9506: Initializing...
19:01:20.9716: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
19:01:21.0125: Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
19:01:21.0165: Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
19:01:21.0165: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
19:01:21.0165: Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
19:01:21.0165: Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
19:01:21.0325: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [1.0.0.0]
19:01:21.0325: Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
19:01:21.0325: Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
19:01:21.0444: Loading plugin assembly: TTController.Plugin.SnakeEffect [0.7.0.0]
19:01:21.0619: Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
19:01:21.0619: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
19:01:21.0619: Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
19:01:21.0884: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
19:01:21.0914: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
19:01:21.0914: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
19:01:21.0914: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
19:01:21.0914: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
19:01:21.0914: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
19:01:21.1093: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
19:01:21.1093: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
19:01:21.1093: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
19:01:21.1093: Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
19:01:21.1223: Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
19:01:21.1223: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
19:01:21.1223: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
19:01:21.1393: Creating Config Manager...
19:01:21.1393: Loading config...

Onverwerkte uitzondering: System.Reflection.TargetInvocationException: Het doel van een aanroep heeft een uitzondering veroorzaakt. ---> System.AccessViolationException: Poging tot het lezen of schrijven van beveiligd geheugen. Dit duidt er vaak op dat ander geheugen is beschadigd.
   bij TTController.Plugin.RazerConnectEffect.RzChromaBroadcastNative.Init(Guid guid)
   bij TTController.Plugin.RazerConnectEffect.RazerConnectEffect..ctor(RazerConnectEffectConfig config)
   --- Einde van intern uitzonderingsstackpad ---
   bij System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   bij System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   bij System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   bij System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   bij System.Activator.CreateInstance(Type type, Object[] args)
   bij TTController.Service.Config.Converter.AbstractPluginConverter`2.ReadJson(JsonReader reader, Type objectType, TPlugin existingValue, Boolean hasExistingValue, JsonSerializer serializer)
   bij Newtonsoft.Json.JsonConverter`1.ReadJson(JsonReader reader, Type objectType, Object existingValue, JsonSerializer serializer)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.DeserializeConvertable(JsonConverter converter, JsonReader reader, Type objectType, Object existingValue)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   bij Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
[... cut ...]
   bij Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   bij TTController.Service.Manager.ConfigManager.LoadOrCreateConfig()
   bij TTController.Service.TTService.Initialize()
   bij TTController.Service.Program.<>c.<Main>b__2_1()
   bij TTController.Service.Program.Main(String[] args)
```

"The target of a call has caused an exception. ---> System.AccessViolationException: Attempting to read or write protected memory. This often indicates that other memory is corrupted."

FearNaBoinne, 2019-05-28:
PS: The log stopped after "Loading config...", so I had to run it in console and grab the output before it all crashed...

MoshiMoshi0, 2019-05-28:
I tried to make 64bit work but I think there is something different with 64 and 32 bit api. I changed the service to compile as 32bit. 

Try this build:
[TTController_0.7.0.0-razer-connect.1.dbf94f07f2f39a77912cebac8d94f4d896fa61a9.zip](https://github.com/MoshiMoshi0/TTController/files/3229301/TTController_0.7.0.0-razer-connect.1.dbf94f07f2f39a77912cebac8d94f4d896fa61a9.zip)

FearNaBoinne, 2019-05-28:
```
23:18:29.3908: ================================================================
23:18:29.4182: Initializing...
23:18:29.8356: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
23:18:29.8645: Loading plugin assembly: TTController.Plugin.ByLedColorEffect [0.7.0.0]
23:18:29.8645: Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
23:18:29.8645: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
23:18:29.8750: Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
23:18:29.8750: Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
23:18:29.8750: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.7.0.0]
23:18:29.8750: Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
23:18:29.8910: Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
23:18:29.8910: Loading plugin assembly: TTController.Plugin.SnakeEffect [0.7.0.0]
23:18:29.9135: Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
23:18:29.9135: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
23:18:29.9214: Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
23:18:29.9399: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
23:18:29.9399: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
23:18:29.9399: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
23:18:29.9524: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
23:18:29.9693: Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
23:18:29.9693: Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
23:18:29.9693: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
23:18:29.9693: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
23:18:29.9898: Creating Config Manager...
23:18:30.0742: Loading config...
23:18:30.2883: Creating DataCache...
23:18:30.2963: Creating Sensor Manager...
23:18:30.2963: Initializing Open Hardware Monitor...
23:18:31.7425: Detected 38 sensors
23:18:31.7425: Creating Effect Manager...
23:18:31.7425: Creating Speed Controller Manager...
23:18:31.7500: Creating Device Manager...
23:18:31.7925: Searching for "Dpsg" controllers
23:18:32.0221: Searching for "Riing" controllers
23:18:32.0471: Searching for "Riing Plus" controllers
23:18:32.0765: Initialized "Riing Plus" controller [0x264A, 0x1FA5]
23:18:32.0765: Searching for "Riing Trio" controllers
23:18:32.1025: Adding effect: RazerConnectEffect [10af9207-7e67-4581-9d13-516cad5d53c1]
23:18:32.1110: Creating Timer Manager...
23:18:32.1110: Starting 4 timers...
23:18:32.1110: Initializing done!
23:18:32.1110: ================================================================

Onverwerkte uitzondering: System.AccessViolationException: Poging tot het lezen of schrijven van beveiligd geheugen. Dit duidt er vaak op dat ander geheugen is beschadigd.
```
It pops up some more info, but closes immediately after, so I cannot grab it...

MoshiMoshi0, 2019-05-28:
This is so unnecessarily difficult... I should have bought a broken mice or keyboard and we would be done already.

But I think this is the last one:
[TTController_0.7.0.0-razer-connect.1.dbf94f07f2f39a77912cebac8d94f4d896fa61a9.zip](https://github.com/MoshiMoshi0/TTController/files/3229819/TTController_0.7.0.0-razer-connect.1.dbf94f07f2f39a77912cebac8d94f4d896fa61a9.zip)

[21 more replies]

closed 2019-05-29 by FearNaBoinne