MoshiMoshi0/TTController #17: Programm doesnt start

Pobblebonk2102, 2019-05-25:
### Prerequisites

> Replace `[ ]` to `[x]` to check

* [ x] The controller is recognized without issues in the official software
* [ x] I made sure no official software or any other software that can communicate with the controllers (e.g. Aida64) was running when the issue with the service occurred

**Describe the issue**
A clear and concise description of what the issue is.

[image]

The programm wont start and it shows the error in the picture.
English translation:
Unhandled exception: System.InvalidOperationException: The Service TTController cannot start on the computer ---> System.ComponentModel.Win32Exception: The Service doesnt answer in time to start- or control-request.
--- End of intern exception stack monitoring ---
at System.ServiceProcess.ServiceController.Start<String[] args>
at TTController.Service.Program.<ManageService>g__StartService:3_1<>
at TTController.Service.Program.<>c<ManageService>b__3_2<>
at TTController.Service.Program.ManageService<>
at TTController.Service.Program.Main<String[] args>

OS: Windows 7 64-bit

I hope there is a solution for this. I would definitely choose your program over the shitty one from TT.

MoshiMoshi0, 2019-05-25:
Please edit the `TTController.Service.exe.Config` file and change:
`<logger name="*" minLevel="Debug" writeTo="console" enabled="true" />`
to
`<logger name="*" minLevel="Trace" writeTo="console" enabled="true" />`
then in the `Main Menu` select `run in console` and post the log output.

Pobblebonk2102, 2019-05-25:
[image]
[image]
[image]

MoshiMoshi0, 2019-05-25:
It seems to initialize and finalize correctly and it is not hanging for a long time anywhere. 

Did you maybe move the exe after installing? If so you have to uninstall and install again.

Pobblebonk2102, 2019-05-25:
I didnt move the .exe after installing. I also tried uninstalling and installing again but that doesnt help :/

MoshiMoshi0, 2019-05-25:
Could you try starting the service from `services.msc`? If it also wont start you could try changing the service account by right clicking on the service and going to "Log On" in properties. There you can set a custom account to run the service as instead of Local System, make sure the account has admin rights.

MoshiMoshi0, 2019-05-25:
Also check if "allow service to interact with desktop" was checked in the service properties.

Pobblebonk2102, 2019-05-25:
Starting trough services.msc with or without letting it run on a custom account fails with the error message "Error 1053: The Service doesnt anwer in time to start- or control-request" (translated).
"Allow service to interact with desktop" was checked.

MoshiMoshi0, 2019-05-25:
Are there logs created in `Logs` folder? Post the newest one when you try to start it.

This is a really weird issue. Seems like its something wrong with windows.

Pobblebonk2102, 2019-05-25:
[current.log](https://github.com/MoshiMoshi0/TTController/files/3219555/current.log)

That is the latest log file after trying to start trough services.msc .

Pobblebonk2102, 2019-05-25:
I tried it again because why not and for some reason it started without an error message right now.
Log file after successful start:
[current.log](https://github.com/MoshiMoshi0/TTController/files/3219556/current.log)

MoshiMoshi0, 2019-05-25:
Oh... Well it fails to start because the config is empty or invalid. Interesting that it works fine when you run it in console.

Did you try to configure the service? What is your current config? If you dont know how to configure it I can make one for you.

Pobblebonk2102, 2019-05-25:
I didnt configure the service yet. Current config.json:
{
  "Profiles": [],
  "ComputerStateProfiles": [],
  "PortConfigs": [],
  "SensorConfigs": []
}
I will try to configure it following the examples given in the readme.

MoshiMoshi0, 2019-05-25:
Ok, if you need help with the config just make a "configuration help" issue. 

I will close this one once I fix that the service fails to start with empty config.

Pobblebonk2102, 2019-05-25:
Thanks for your help :)

MoshiMoshi0, 2019-05-25:
Unfortunately, im unable to reproduce this bug with empty config. So you had to somehow have an invalid config for this to happen. The exception you got only happens when there is no plugin with a name specified in the config. I'll add some log messages for that.

Anyways, you have it working so im closing this issue.

closed 2019-05-25 by MoshiMoshi0