MoshiMoshi0/TTController #18: Controller not showing up in CLI app

Anreill, 2019-05-31:
Using the hub included with this Riing Trio kit, just curious if it's supported or if I'm having a problem of some sort.

https://www.amazon.com/gp/product/B07FWJ1GHT

MoshiMoshi0, 2019-05-31:
Yes it is supported. 

Does the controller work with the official software? Did you make sure to not run both official software and TTController at the same time?

Also by "not showing" you mean in show hardware info?

Anreill, 2019-06-04:
Yeah, sorry about that.

Works fine with the official software, wasn't running both at the same time. Yes, hardware info in your CLI interface didn't list the controller.

I went ahead and got the TT Sync since it will run off the motherboard RGB headers, so no worries. Thanks for the quick response. If it would help to try stuff with the old one for dev purposes let me know and I can hook it back up.

MoshiMoshi0, 2019-06-04:
No problem. 

If you want you could edit `TTController.Service.exe.Config` file and change: 
`<logger name="*" minLevel="Debug" writeTo="console" enabled="true" />` 
to
 `<logger name="*" minLevel="Trace" writeTo="console" enabled="true" />`
then select "Run in console" in the menu and post the output after it initializes. Dont know why it would not list it.

closed 2019-06-08 by MoshiMoshi0