MoshiMoshi0/TTController #19: Configuration Help

xxnovosxx, 2019-06-20:
**Describe your desired configuration**
what I want as a configuration is this step:
CPU temp 15° - Fan 0%
CPU temp 30° - Fan 20%
CPU temp 40° - Fan 40%
CPU temp 50° - Fan 50%
CPU temp 60° - Fan 75%
CPU temp 65° - Fan 90%
CPU temp 70° - Fan 100%

Wave effect exept when the cpu hits 75C then red pulsating

**Hardware info**
Controllers
-------------------------------
Name: Riing Plus
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 30, Rpm: 492, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 30, Rpm: 494, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 30, Rpm: 540, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 0, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SnakeEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll

Sensors
-------------------------------
AMD Ryzen 7 1700X:
        Core (Tctl) (/amdcpu/0/temperature/0): 68
        Core (Tdie) (/amdcpu/0/temperature/1): 48
Radeon RX Vega:
        GPU Core (/atigpu/0/temperature/0): 53
ST3000DM001-1ER166:
        Temperature (/hdd/0/temperature/0): 38
WDC WDS240G2G0B-00EPW0:
        Temperature (/hdd/1/temperature/0): 34
-------------------------------

MoshiMoshi0, 2019-06-20:
This should do what you want I think, you didnt say what wave colors you want so I added white -> black gradient as a sample. You can easily tweak the colors/numbers, feel free to ask if you have problems.

```json
{
  "Profiles": [
    {
      "Name": "Fans",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100],
            ],
            "Sensors": ["/amdcpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "Colors": [
              [250, 250, 250],
              [230, 230, 230],
              [210, 210, 210],
              [190, 190, 190],
              [170, 170, 170],
              [150, 150, 150],
              [130, 130, 130],
              [110, 110, 110],
              [90, 90, 90],
              [70, 70, 70],
              [50, 50, 50],
              [30, 30, 30]
            ],
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c2",
      "Ports": [
        [9802, 8101, 4]
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/amdcpu/0/temperature/0"],
            "ColorGradient": [
              [15, [0, 0, 255]],
              [45, [255, 255, 0]],
              [70, [255, 0, 0]],
            ],
            "Trigger": {
              "Type": "PulseTrigger",
              "Config": {
                "On": 1000,
                "Off": 1000
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 75
      }
    }
  ]
}
```

xxnovosxx, 2019-06-20:
the pump all the time but i should have said like the breathing effect if it is possible and the fans rainbow wave sry my english is not so good..

MoshiMoshi0, 2019-06-21:
Unfortunately you cant mix two effects right now, you want a mix of TemperatureEffect and BreathingEffect.
The best you can do is to use the PulseTrigger to pulse the effect on and off.
But its possible to make the pulsing only on the red color, otherwise it would be smooth gradient.
Also you can change how long its on/off, so it could be a quick blink.

Here is the config with flow effect for the fans:
```json
{
  "Profiles": [
    {
      "Name": "Fans",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100],
            ],
            "Sensors": ["/amdcpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FlowEffect",
          "Config": {
            "FillStep": 0.05,
            "HueStep": 30,
            "Saturation": 1.0,
            "Brightness": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c2",
      "Ports": [
        [9802, 8101, 4]
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/amdcpu/0/temperature/0"],
            "ColorGradient": [
              [15, [0, 0, 255]],
              [45, [255, 255, 0]],
              [70, [255, 0, 0]],
            ],
            "Trigger": {
              "Type": "PulseTrigger",
              "Config": {
                "On": 1000,
                "Off": 1000
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 75
      }
    }
  ]
}
```

MoshiMoshi0, 2019-06-26:
Closing as I guess you have it working. 
If you need help again reply here or create a new issue.

closed 2019-06-26 by MoshiMoshi0