MoshiMoshi0/TTController #20: Configuration Help

ArtursGailis1995, 2019-06-22:
**Describe your desired configuration**
Thermaltake Riing Plus 14 LED RGB Fan TT Premium Edition Radiator 3-Fan Pack

All 3 fans used as case fans, connected to single controller, USB to MB, Power to PSU.

I tried original software but it is complete trash so I would like to switch to your software. Only thing working is temperature readings. Otherwise with every config it stays at speed 50 no matter what. Light effects also does not change and is stuck to RGB pulse. Cycling colors. No other effects seems to work. Tried sketch wit bootup/shutdown effect - also does not trigger. Official software removed for now (with it does not work too your software). Need to get at least PWM working.

**Hardware info**
```
Controllers
-------------------------------
Name: Riing
VendorId: 9802
ProductId: 8008
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 885, Unknown: 3]
        Identifier: [9802, 8008, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 865, Unknown: 3]
        Identifier: [9802, 8008, 2]

        Id: 3
        Data: [PortId: 3, Speed: 50, Rpm: 846, Unknown: 3]
        Identifier: [9802, 8008, 3]

        Id: 4
        Data: [PortId: 254, Speed: 0, Rpm: 0, Unknown: 3]
        Identifier: [9802, 8008, 4]

        Id: 5
        Data: [PortId: 254, Speed: 0, Rpm: 0, Unknown: 3]
        Identifier: [9802, 8008, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SnakeEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll

Sensors
-------------------------------
Intel Core i7-7700K:
        CPU Core #1 (/intelcpu/0/temperature/0): 48
        CPU Core #2 (/intelcpu/0/temperature/1): 48
        CPU Core #3 (/intelcpu/0/temperature/2): 45
        CPU Core #4 (/intelcpu/0/temperature/3): 48
        CPU Package (/intelcpu/0/temperature/4): 48
NVIDIA GeForce GTX 1070:
        GPU Core (/nvidiagpu/0/temperature/0): 43
WDC WD10EZEX-00BN5A0:
        Temperature (/hdd/0/temperature/0): 32
WDC WD5000AUDX-73H9TY0:
        Temperature (/hdd/1/temperature/0): 29
LITEON IT LST-32S9G-HP:
        Temperature (/hdd/2/temperature/0): 45
-------------------------------
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8008, 1],
        [9802, 8008, 2],
        [9802, 8008, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [15, 0],
              [30, 20],
              [40, 40],
              [50, 50],
              [60, 75],
              [65, 90],
              [70, 100],
            ],
            "Sensors": ["/intelcpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "Colors": [
              [250, 250, 250],
              [230, 230, 230],
              [210, 210, 210],
              [190, 190, 190],
              [170, 170, 170],
              [150, 150, 150],
              [130, 130, 130],
              [110, 110, 110],
              [90, 90, 90],
              [70, 70, 70],
              [50, 50, 50],
              [30, 30, 30]
            ],
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 70
      }
    }
  ]
}
```

ArtursGailis1995, 2019-06-22:
[image]
No matter the temperature, PWM value stays the same. Fans does not ramp up phisically too.

MoshiMoshi0, 2019-06-22:
Interesting. Is the official software able to control the speed of the fans and the rgb colors?

Could you edit `TTController.Service.exe.Config` file and change 
`<logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to
`<logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`
then run in console for a few seconds, press enter so it cleans up and then send the latest log file?

Also this is the first time i've seen the `Unknown` value to be something else than 255

ArtursGailis1995, 2019-06-22:
@MoshiMoshi0 Yes, with the official software everything works fine. I Hate the official software because every time computer starts up, it loses the configuration and I have to open software and turn on RGB and put it to PWM.

The requested LOG file is attached below. I work in IT, so if I am unable to fix this issue in multiple hours it means its either broken or I had too much beer.

[current.log](https://github.com/MoshiMoshi0/TTController/files/3316805/current.log)

MoshiMoshi0, 2019-06-22:
Are you sure you have Riing Plus fans? For some reasone your controller is detected as first gen Riing controller, not Riing Plus controller.

All data seems correct, and the controller responds with success messages, so im not sure what is wrong.

If you its possible, could you install Wireshark with usbpcap and capture the usb data? But use the official software while capturing. I want to see if I missed something.

ArtursGailis1995, 2019-06-22:
Ah, there is a chance that e-shop sold me wrong product. That makes sense then, as it was 50% off at least. In the listing it said Riing Plus 14.. Maybe it is not..

Here is the pcap attached (in zip because it did not allow me to attach pcap). 
[tt.zip](https://github.com/MoshiMoshi0/TTController/files/3317084/tt.zip)

MoshiMoshi0, 2019-06-22:
I dont see much that is different, but they seem to use different bytes for speed.
Try this build:
[TTController_0.7.1.0-alpha.22.416b4adaa45a2bcf9ef5efc92476a806aba7deee.zip](https://github.com/MoshiMoshi0/TTController/files/3317308/TTController_0.7.1.0-alpha.22.416b4adaa45a2bcf9ef5efc92476a806aba7deee.zip)

I hope this will make the speed work.

ArtursGailis1995, 2019-06-23:
@MoshiMoshi0 Yes, this time speed control seems to be working fine. Values do change in console and also I can hear fans ramping up/down on CPU temperature change.

Here is the PCAP file just in case you want to investigate further (some basic LED functions, like turning on and off, rainbow sweep effect, setting to static white color etc.) Started capture before running software. Ran software and changed fan speed, turned on RGB, saved, turned on some profiles. Controller should be connected on USB HUB number 1.3.1.

[tt_new_before_startup.zip](https://github.com/MoshiMoshi0/TTController/files/3317919/tt_new_before_startup.zip)

MoshiMoshi0, 2019-06-23:
Nice. 
It seems like you have first gen and first firmware (v1.0.0).

Looking at the page [here](https://www.thermaltake.com/riing-12-rgb-radiator-fan-tt-premium-edition.html) it looks like you only have 2 rgb effects? RGB Cycle/Flow and custom full color, is that correct other effects are working too?

If you have more effects could you do one last capture going thru all of the available effects while also changing the rgb effect speed and the effect custom colors?

If you only have those two, what custom color do you have set? Its showing as FFFF00 but I think they are using GRB format.

BTW the controller is 1.5.1.

MoshiMoshi0, 2019-06-26:
Can you test that RGB works with v0.7.2.0-alpha? As I would like to close this issue.

You can test with these effects:

Built in flow effect:
```json
{
    "Type": "RawEffect",
    "Config": {
        "EffectType": "Flow",
        "Trigger": {
            "Type": "OneTimeTrigger"
        }
    }
}
```

Full magenta color:
```json
{
    "Type": "FullColorEffect",
    "Config": {
        "Color": [255, 0, 144],
        "Trigger": {
            "Type": "OneTimeTrigger"
        }
    }
}
```

If those work you should be able use any other effect that uses `Full` effect type. For example `SensorEffect`.

ArtursGailis1995, 2019-06-29:
@MoshiMoshi0 Sorry for late reply, Im on holiday.

Yes, now the v0.7.2.0-alpha seems to be working fine. Only thing - its not creating magenta but Cyan color for me with your sample code.

MoshiMoshi0, 2019-06-29:
No worries.

When you can, try setting the color to `[255, 0, 0]` then `[0, 255, 0]` then `[0, 0, 255]`. Those should be red, green, blue.

If those are still cyan I would need another capture where you change the static color in the official software a couple of times, preferably to red, green, blue colors.

BTW does first config with the "Flow" effect work?

ArtursGailis1995, 2019-06-29:
[255, 0, 0] - Green
[0, 255, 0] - Red
[0, 0, 255] - Blue
[255, 255, 0] - Yellow
[255, 0, 255] - Cyan
[0, 255, 255] - Magenta
[0, 0, 0] - Black (Off)
[255, 255, 255] - White.

Yes, the Flow effect works.

MoshiMoshi0, 2019-06-29:
So it seems that they use RGB in the first gen but GRB in the other controllers.

Anyways, this should fix the colors:
[TTController_0.7.2.0-alpha.5.e6bf88d4fa2bb881d03b08f5dd079e70bcb5078d.zip](https://github.com/MoshiMoshi0/TTController/files/3342205/TTController_0.7.2.0-alpha.5.e6bf88d4fa2bb881d03b08f5dd079e70bcb5078d.zip)

If you confirm, I'll close this issue.

ArtursGailis1995, 2019-07-09:
Confirm.

closed 2019-07-09 by ArtursGailis1995