MoshiMoshi0/TTController #21: Help setting up.

astothialo, 2019-07-12:
Hi,

I'm having a hard time wrapping my head around how to set this up. I currently have 6 pure plus 12 fans and the tt rgb plus software from thermaltake creates massive dpc latency spikes so figured id give this a go.
Any chance you could please help me with a simple setup for the fans?
Just want a static colour of dark blue and for them to run quiet and increase their rpm if required by load etc

Would be greatly appreciated.

MoshiMoshi0, 2019-07-12:
Sorry for late reply, i didn't get the notification.

Could you go to `Main Menu -> Debug -> Report` and post it here?

astothialo, 2019-07-13:
is this what you wanted?

Controllers
-------------------------------
Name: Riing Plus
VendorId: 9802
ProductId: 8101
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 37, Rpm: 621, Unknown: 255]
        Identifier: [9802, 8101, 1]

        Id: 2
        Data: [PortId: 2, Speed: 37, Rpm: 645, Unknown: 255]
        Identifier: [9802, 8101, 2]

        Id: 3
        Data: [PortId: 3, Speed: 37, Rpm: 627, Unknown: 255]
        Identifier: [9802, 8101, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8101, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

Name: Riing Plus
VendorId: 9802
ProductId: 8102
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 37, Rpm: 632, Unknown: 255]
        Identifier: [9802, 8102, 1]

        Id: 2
        Data: [PortId: 2, Speed: 37, Rpm: 615, Unknown: 255]
        Identifier: [9802, 8102, 2]

        Id: 3
        Data: [PortId: 3, Speed: 37, Rpm: 644, Unknown: 255]
        Identifier: [9802, 8102, 3]

        Id: 4
        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 4]

        Id: 5
        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
        Identifier: [9802, 8102, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

-------------------------------
Sensors
-------------------------------
Intel Core i7-8700K:
        Temperature
                CPU Core #1 (/intelcpu/0/temperature/0): 34.0 °C
                CPU Core #2 (/intelcpu/0/temperature/1): 29.0 °C
                CPU Core #3 (/intelcpu/0/temperature/2): 30.0 °C
                CPU Core #4 (/intelcpu/0/temperature/3): 29.0 °C
                CPU Core #5 (/intelcpu/0/temperature/4): 27.0 °C
                CPU Core #6 (/intelcpu/0/temperature/5): 28.0 °C
                CPU Package (/intelcpu/0/temperature/6): 35.0 °C
NVIDIA GeForce GTX 1070:
        Temperature
                GPU Core (/nvidiagpu/0/temperature/0): 51.0 °C
ST3000DM001-1ER166:
        Temperature
                Temperature (/hdd/2/temperature/0): 23.0 °C
ST4000VX007-2DT166:
        Temperature
                Temperature (/hdd/3/temperature/0): 22.0 °C
-------------------------------
Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll
-------------------------------
Press any key to continue...

MoshiMoshi0, 2019-07-13:
Ok this should be the config you want:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "20af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [50, 20],
              [60, 25],
              [70, 40],
              [80, 75],
              [85, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 139],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "Speed": 35
    },
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3]
      ],
      "Speed": 35
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

The CurvePoints are in format `[temperature, speed]` you can change/add them how you want.

astothialo, 2019-07-13:
you sir are an absolute legend!
Works perfectly and my dpc latency is ZERO!
Thank you so much!!!!!!!!
Youve saved me from insanity haha the dpc latency from the TT rgb software was driving me up the wall to the point i was going to rip them out :/
Thank you x394989484984 :D

about the curvepoints you mentioned do I need to do anything with those?

MoshiMoshi0, 2019-07-13:
> about the curvepoints you mentioned do I need to do anything with those? WIll the fans kick in when under load as is?

You can configure them to make the pwm fan curve you want. 
This is how the curve looks now: https://i.imgur.com/vfC0T84.png

astothialo, 2019-07-13:
gotcha :)
Thank you again so very much!
Appreciate your time and effort :)

closed 2019-07-13 by MoshiMoshi0