MoshiMoshi0/TTController #23: Configuration Help

dartmoose, 2019-07-20:
**Describe your desired configuration**
Ultimately, replicate the Corsair Pong effect: https://twitter.com/corsair/status/1052350678713085952?lang=en

But for now:
All fans set to 50% speed, all LEDs set to Red

I'm using the basic profile below, but the LEDs seem not to respond or change color.  I have three Riing Trio 120s and two Riing Trio 140s on one Trio controller.  The fan speed works from the service (if I set it to 100 on the config it will ramp up correctly).  But there is no change at all in the colors.

Everything works with the official software.  Thank you by the way for this project, it is awesome!

**Hardware info**
```
Controllers
-------------------------------
Name: Riing Trio
VendorId: 9802
ProductId: 8501
Ports:
        Id: 1
        Data: [PortId: 1, Speed: 50, Rpm: 830, Unknown: 255]
        Identifier: [9802, 8501, 1]

        Id: 2
        Data: [PortId: 2, Speed: 50, Rpm: 849, Unknown: 255]
        Identifier: [9802, 8501, 2]

        Id: 3
        Data: [PortId: 3, Speed: 50, Rpm: 825, Unknown: 255]
        Identifier: [9802, 8501, 3]

        Id: 4
        Data: [PortId: 4, Speed: 50, Rpm: 824, Unknown: 255]
        Identifier: [9802, 8501, 4]

        Id: 5
        Data: [PortId: 5, Speed: 50, Rpm: 862, Unknown: 255]
        Identifier: [9802, 8501, 5]

Available effect types:
Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, ByLed, Full

-------------------------------
Sensors
-------------------------------
AMD Ryzen 9 3900X 12-Core Processor:
        Temperature
                Core (Tctl) (/amdcpu/0/temperature/0): 64.0 °C
                Core (Tdie) (/amdcpu/0/temperature/1): 64.0 °C
NVIDIA GeForce RTX 2080 Ti:
        Temperature
                GPU Core (/nvidiagpu/0/temperature/0): 48.0 °C
ST2000DM008-2FR102:
        Temperature
                Temperature (/hdd/0/temperature/0): 41.0 °C
-------------------------------
Plugins
-------------------------------
Valid plugins:
        TTController.Plugin.BlinkEffect.dll
        TTController.Plugin.ByLedColorEffect.dll
        TTController.Plugin.FlowEffect.dll
        TTController.Plugin.FullColorEffect.dll
        TTController.Plugin.PulseEffect.dll
        TTController.Plugin.RawEffect.dll
        TTController.Plugin.RazerConnectEffect.dll
        TTController.Plugin.RippleEffect.dll
        TTController.Plugin.SensorEffect.dll
        TTController.Plugin.SoundEffect.dll
        TTController.Plugin.SpectrumEffect.dll
        TTController.Plugin.WaveEffect.dll
        TTController.Plugin.CopySpeedController.dll
        TTController.Plugin.PwmSpeedController.dll
        TTController.Plugin.StaticSpeedController.dll
        TTController.Plugin.AlwaysTrigger.dll
        TTController.Plugin.LogicTrigger.dll
        TTController.Plugin.OneTimeTrigger.dll
        TTController.Plugin.ProcessTrigger.dll
        TTController.Plugin.PulseTrigger.dll
        TTController.Plugin.SensorTrigger.dll
        TTController.Plugin.DpsgController.dll
        TTController.Plugin.RiingController.dll
        TTController.Plugin.RiingPlusController.dll
        TTController.Plugin.RiingTrioController.dll
```

**Current configuration**
```
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [ 9802, 8501, 1 ],
        [ 9802, 8501, 2 ],
        [ 9802, 8501, 3 ],
        [ 9802, 8501, 4 ],
        [ 9802, 8501, 5 ]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 50,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [ 255, 0, 0 ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
         }
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [ "/amdcpu/0/temperature/0" ],
      "Config": {
        "CriticalValue": 75
      }
    }
  ]
}
```

MoshiMoshi0, 2019-07-20:
That is a cool effect. I'll add it.

Riing trio is a little bit tricky, you need to specify the correct number of leds in PortConfig, and in this case since you are using an effect that returns only one color, you need to copy that color 30 times using LedCountHandling.

Its on TODO to force FullColorEffect to return the colors copied automatically so you dont need to use LedCountHandling.

Add this above/below sensor configs and the color should be working:
```json
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "Config": {
        "LedCount": 30,
        "LedCountHandling": "Copy"
      }
    }
  ],
```

dartmoose, 2019-07-21:
Thanks, that worked perfectly!  All set now.

closed 2019-07-21 by dartmoose

MoshiMoshi0, 2019-09-15:
Hi, Ive added that corsair effect. It is available in the dev build [here](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts).

There has been some changes to the config format so let me know if you need help.
Config that should work based on the above config (works only on dev build):
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController",
          "Config": {
            "Speed": 50,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "PingPongEffect",
          "Config": {
            "Step": 0.01,
            "Height": 0.2,
            "Width": 0.5,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
         }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8501, 1]],
      "Config": {
        "LedRotation": [0, 0, 0]
      }
    },
    {
      "Ports": [[9802, 8501, 2]],
      "Config": {
        "LedRotation": [0, 0, 0]
      }
    },
    {
      "Ports": [[9802, 8501, 3]],
      "Config": {
        "LedRotation": [0, 0, 0]
      }
    },
    {
      "Ports": [[9802, 8501, 4]],
      "Config": {
        "LedRotation": [0, 0, 0]
      }
    },
    {
      "Ports": [[9802, 8501, 5]],
      "Config": {
        "LedRotation": [0, 0, 0]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [ "/amdcpu/0/temperature/0" ],
      "Config": {
        "CriticalValue": 75
      }
    }
  ]
}
```

You now have to edit "PortConfigs" section and update "LedRotation" for each fan. Each "LedRotation" is a list with led offsets for each zone. Riing Trio has 3 zones so there are 3 offsets. The first two offsets are for the outer rings so you want to set them to the same value, the last offset should be the inner ring.

Effect config explanation:
* Step: how much should the light advance each tick 
* Height: Height of the light strip
* Width: Width of the light strip

Here is a small demo but I only have Riing fans, should look better with Riing Trio/Duo:
https://gfycat.com/repulsiverapidafricanpiedkingfisher

jlaineinat0r, 2019-09-22:
Think the zip provided is missing the effect, was curious to try this out. Though getting it to work may make me want to upgrade my fans...

MoshiMoshi0, 2019-09-22:
Download from here:
https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/27420953/artifacts

I doubt corsair allows you to rotate the fans in the software, so you will have to physically rotate them to match, but that will change cable routing.