MoshiMoshi0/TTController #24: PluginLoader causing System.IO.FileLoadException

redaxmedia, 2019-08-16:
### Prerequisites

* [x] The controller is recognized without issues in the official software
* [x] I made sure no official software or any other software that can communicate with the controllers (e.g. Aida64) was running when the issue with the service occurred

**Describe the issue**

The service could not be started as of:

```
2019-08-16 02:11:52.2080|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
2019-08-16 02:11:52.2200|FATAL|TTController.Service.TTService|System.IO.FileLoadException: Die Datei oder Assembly "TTController.Plugin.BlinkEffect, Version=0.7.0.0, Culture=neutral, PublicKeyToken=null" oder eine Abhängigkeit davon wurde nicht gefunden. Der Vorgang wird nicht unterstützt. (Ausnahme von HRESULT: 0x80131515)
Dateiname: "TTController.Plugin.BlinkEffect, Version=0.7.0.0, Culture=neutral, PublicKeyToken=null" ---> System.NotSupportedException: Es wurde versucht, eine Assembly von einer Netzwerkadresse zu laden, was in früheren Versionen von .NET Framework zum Ausführen der Assembly als Sandkastenassembly geführt hätte. In dieser Version von .NET Framework wird die CAS-Richtlinie standardmäßig nicht aktiviert, dieser Ladevorgang kann daher gefährlich sein. Wenn Sie nicht beabsichtigen, durch diesen Ladevorgang eine Sandkastenassembly zu erstellen, aktivieren Sie den loadFromRemoteSources-Schalter. Weitere Informationen finden Sie unter "http://go.microsoft.com/fwlink/?LinkId=155569".
   bei System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   bei System.Reflection.RuntimeAssembly.nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   bei System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   bei System.Reflection.Assembly.Load(AssemblyName assemblyRef)
   bei TTController.Service.Utils.PluginLoader.<>c.<Load>b__3_0(Assembly a)
   bei System.Linq.Enumerable.WhereSelectEnumerableIterator`2.MoveNext()
   bei System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   bei System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   bei TTController.Service.Utils.PluginLoader.Load(String path, Type[] types)
   bei TTController.Service.Utils.PluginLoader.LoadAll(String path)
   bei TTController.Service.TTService.Initialize()
   bei TTController.Service.TTService.OnStart(String[] args)
```

I'm not a .NET developer but I found that `loadFromRemoteSources` is an issue here. Once I unblocked each `dll` file under `Plugins` the service could be started.

[image]

MoshiMoshi0, 2019-08-16:
I'll have to read how to prevent the files from being blocked because iirc one other use had this problem too. I have no idea why windows does it only on some systems.

Workaround for now is like you said to uncheck "Unblock" on each file.
Thanks for reporting.

MoshiMoshi0, 2019-08-16:
Ok, so it seems that when you download files from the internet Windows marks them as being unsafe with a special zone identifier. Since this is a Windows thing I cant do much about it. 

The solution here would be to either:
* Unblock the zip, and then unpack. This will mark all unzipped files as unblocked.
* Use 7zip which seems to strips zone identifer data from the files even if the zip itself is not unblocked.

I assume you used windows zip extraction? Or do you use 7zip/winrar.

redaxmedia, 2019-08-16:
I used Windows internal Unzip solution. You should catch those exceptions and point the user to an solution.

What about providing an executable archive instead of an ZIP file?

MoshiMoshi0, 2019-08-16:
Yes exe installer/archive would work, but I was not able to find any satisfying solution that would work locally and on ci.

So I think just adding `loadFromRemoteSources` is a simpler solution and should fix this.
Try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts

MoshiMoshi0, 2019-08-24:
Confirmed that `loadFromRemoteSources` fixes the problem.
Thanks!

closed 2019-08-24 by MoshiMoshi0