MoshiMoshi0/TTController #25: Failed to start the service

mchnb003, 2019-09-01:
### Prerequisites

> Replace `[ ]` to `[x]` to check

* [x] The controller is recognized without issues in the official software
* [x] I made sure no official software or any other software that can communicate with the controllers (e.g. Aida64) was running when the issue with the service occurred

**Describe the issue**
when i try to start the service, i receive a timeout, se attachements.
Can you please help me, to figure out what i am doing wrong ?
Thx a lot and best regards from munich
Boris

[current.log](https://github.com/MoshiMoshi0/TTController/files/3563445/current.log)
[image]

MoshiMoshi0, 2019-09-01:
If you use the windows unzip functionality you have to first right click the zip archive, click properties, and click "Unblock" button. But I would recommend using winrar or 7zip.

You can also use the latest [dev build](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts) which has a builtin solution for that problem.

mchnb003, 2019-09-01:
Perfect, ti used 7Zip now, reinstalled the service and it's on :-)
Thank you very much !

closed 2019-09-01 by mchnb003