MoshiMoshi0/TTController #26: Automatic profile switching

ray-holmium, 2019-09-04:
@MoshiMoshi0 thanks a bunch for making this program! It is an awesome improvement over the trash stock thermaltake software. 

I wonder if it would be possible to add automatic profile switching based on time of day. Late at night I'd like to have my leds shut off and have my fans adapt a less aggressive curve.

Currently I just have two config files, one with lights on and one with lights off, and I manually stop the service and switch the config files to enable or disable lights.

Cheers!

MoshiMoshi0, 2019-09-05:
Good idea!

I'll add a trigger that will allow you to specify a schedule. So you will have two speed controllers/effects that will switch depending on the schedule you specify.
I'll also think about having triggers for the whole profiles, but it might require a big code change.

ray-holmium, 2019-09-05:
Amazing, thank you!

Having triggers for entire profiles is definitely overkill for my needs, but I can see how that would enable an incredible level of fine-tuning.

MoshiMoshi0, 2019-09-05:
Added ScheduleTrigger to dev build, you can get it [here](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts).

Sample config:
```json
...
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 0, 0],
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                "Scope": "Day",
                "Schedule": [
                  "00:00:00 -> 08:00:00",
                  "20:00:00 -> 23:59:59"
                ],
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 255, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
...
```

This will trigger the red full color on the specified schedule, otherwise it will fallback to green.
The format for the schedule is `"{start} -> {end}"` with `[d.]hh:mm:ss` for time ([ms docs](https://docs.microsoft.com/en-us/dotnet/standard/base-types/standard-timespan-format-strings#the-constant-c-format-specifier)).

Let me know if you need help with configuring this with your current setup.

MoshiMoshi0, 2019-09-09:
@ray-holmium does this work/is this what you wanted?

I would like to close this issue if that is the case.

ray-holmium, 2019-09-25:
@MoshiMoshi0 I'm so sorry for the late reply! I had kind of a crappy time this past month.

Thank you so much for setting up that schedule trigger for me. It works perfectly! 

Though I've run into a new, unrelated bug which I will start a new issue for.

jlaineinat0r, 2019-09-30:
Trying to run this with my existing rotation setup doesn't seem to work for me, but the idea of auto-shutdown during a schedule is awesome - my machine runs 24/7 but I'd love to kill all the lights during the night.

` Newtonsoft.Json.JsonSerializationException: Error converting value 10 to type 'System.Int32[]'. Path 'LedRotation', line 5, position 19. ---> System.ArgumentException: Could not cast or convert from System.Int64 to System.Int32[].
   at Newtonsoft.Json.Utilities.ConvertUtils.EnsureTypeAssignable(Object value, Type initialType, Type targetType)
   at Newtonsoft.Json.Utilities.ConvertUtils.ConvertOrCast(Object initialValue, CultureInfo culture, Type targetType)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.EnsureType(JsonReader reader, Object value, CultureInfo culture, JsonContract contract, Type targetType)
   --- End of inner exception stack trace ---`

MoshiMoshi0, 2019-09-30:
@jlaineinat0r Please make a new issue and post your config there. Current develop builds have configuration changes compared to latest release.