MoshiMoshi0/TTController #27: Razer Effect

Uncreal, 2019-09-18:
### Prerequisites

> Replace `[ ]` to `[x]` to check

* [x] The controller is recognized without issues in the official software
* [x] I made sure no official software or any other software that can communicate with the controllers (e.g. Aida64) was running when the issue with the service occurred

**Describe the issue**
Hey first of all great work so far. Now to my Problem. I got a Floe Riing RGB 360 + 3 Riing Trio 12 and when i use the RazerConnectEffect only the led´s  in the top of each fan are lighting up. On all other Effects everything works fine. I tryed to set the LedCount and LedCountHandling but it makes no difference. Maybe some one can help me :).

MoshiMoshi0, 2019-09-18:
Only few leds are lighting up because razer connect only provides 5 led colors, that is limitation by razer. So you have to specify the correct LedCount and a LedCountHandling that will fill the remaining leds.

Can you show me the config you tried?

Uncreal, 2019-09-18:
Sure this is my Config 

``` json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
		[9802, 8101, 1],
		[9802, 8101, 2],
		[9802, 8101, 3],
		[9802, 8101, 4]
	],
	"Config": {			
			"LedCount": 12,
			"LedCountHandling": "Lerp",			
		  },		
	"Ports": [
		[9802, 8502, 1],			
		[9802, 8502, 2],
		[9802, 8502, 3]
      ],
	  "Config": {			
			"LedCount": 30,
			"LedCountHandling": "Lerp",			
		  },		
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
	  ]
	}
  ]
}
```

MoshiMoshi0, 2019-09-18:
That is not a valid config.
Try this:

```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 75,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4]
      ],
      "Config": {
        "LedCount": 12,
        "LedCountHandling": "Lerp"
      }
    },
    {
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3]
      ],
      "Config": {
        "LedCount": 30,
        "LedCountHandling": "Lerp"
      }
    }
  ]
}
```

Uncreal, 2019-09-18:
Thx it works!

closed 2019-09-18 by Uncreal

MoshiMoshi0, 2019-09-18:
Nice! Let me know if you need help with configuring anything else (boot/shutdown profile, custom fan curve, other effects etc.)