MoshiMoshi0/TTController #28: CPU temperature sensor not working

ray-holmium, 2019-09-25:
### Prerequisites

> Replace `[ ]` with `[x]` to check

* [ x] The controller is recognized without issues in the official software
* [ x] I made sure no official software or any other software that can communicate with the controllers (e.g. Aida64) was running when the issue with the service occurred

**Describe the issue**
OpenHardwareMonitor is no longer reading the temperature sensor for my cpu (intel 9600k). 

I did recently switch to the developer build of TTController in order to test the new ScheduleTrigger, but I also recently had to reinstall Windows 10 on my system. Either of these changes could have caused the issue.

As far as I can tell, my cpu is only supported in the latest alpha build of OpenHardwareMonitor. See here:  https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1156

Unfortunately I lack the technical skills to recompile TTController with the alpha build of OpenHardwareMonitor.

**Current configuration**
```json
If applicable, replace this text (between ``` characters) with contents of your current configuration here. 
```

MoshiMoshi0, 2019-09-25:
We are using [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor) which is a fork of OpenHardwareMonitor. It has support for Coffee Lake processors so that is not the issue.

Can you post output of `Main Menu -> Debug -> Sensors` and your configuration?

Also, which develop version did you download? Latest build on AppVeyor is for netcore3.0 port which should work but can have issues. Latest develop download is [this](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/27420953/artifacts). (I should start linking to specific builds rather than posting a link to the latest build)

ray-holmium, 2019-09-25:
Ah, roger that.

The netcore version didn't run for me at all, so I used develop-45-a3788fe. When I noticed the cpu sensors weren't working, I also tried develop-36-7e5f95b, which is the version where you added the ScheduleTrigger. The cpu sensor issue was also present in that build.

Just now I tested the latest release build and my cpu sensors are reading correctly with that one.

Sensors (develop-45-a3788fe):
 Intel Core i5-9600K:
»        Load:
»                CPU Core #1 (/intelcpu/0/load/1): 8.3 %
»                CPU Core #2 (/intelcpu/0/load/2): 2.8 %
»                CPU Core #3 (/intelcpu/0/load/3): 11.1 %
»                CPU Core #4 (/intelcpu/0/load/4): 23.6 %
»                CPU Core #5 (/intelcpu/0/load/5): 4.2 %
»                CPU Core #6 (/intelcpu/0/load/6): 5.6 %
»                CPU Total (/intelcpu/0/load/0): 9.3 %
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): NaN °C
»                CPU Core #2 (/intelcpu/0/temperature/1): NaN °C
»                CPU Core #3 (/intelcpu/0/temperature/2): NaN °C
»                CPU Core #4 (/intelcpu/0/temperature/3): NaN °C
»                CPU Core #5 (/intelcpu/0/temperature/4): NaN °C
»                CPU Core #6 (/intelcpu/0/temperature/5): NaN °C
»                CPU Package (/intelcpu/0/temperature/6): NaN °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): NaN °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): NaN °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): NaN °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): NaN °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): NaN °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): NaN °C
»                Core Max (/intelcpu/0/temperature/13): NaN °C
»                Core Average (/intelcpu/0/temperature/14): NaN °C
»        Clock:
»                CPU Core #1 (/intelcpu/0/clock/1): NaN MHz
»                CPU Core #2 (/intelcpu/0/clock/2): NaN MHz
»                CPU Core #3 (/intelcpu/0/clock/3): NaN MHz
»                CPU Core #4 (/intelcpu/0/clock/4): NaN MHz
»                CPU Core #5 (/intelcpu/0/clock/5): NaN MHz
»                CPU Core #6 (/intelcpu/0/clock/6): NaN MHz
»
»  NVIDIA GeForce RTX 2080 Ti:
»        Temperature:
»                GPU Core (/nvidiagpu/0/temperature/0): 26.0 °C
»        Clock:
»                GPU Core (/nvidiagpu/0/clock/0): 300 MHz
»                GPU Memory (/nvidiagpu/0/clock/1): 405 MHz
»                GPU Shader (/nvidiagpu/0/clock/2): 0 MHz
»        Load:
»                GPU Core (/nvidiagpu/0/load/0): 0.0 %
»                GPU Memory Controller (/nvidiagpu/0/load/1): 7.0 %
»                GPU Video Engine (/nvidiagpu/0/load/2): 0.0 %
»                GPU Memory (/nvidiagpu/0/load/3): 7.1 %
»        SmallData:
»                GPU Memory Total (/nvidiagpu/0/smalldata/3): 11264.0 MB
»                GPU Memory Used (/nvidiagpu/0/smalldata/2): 802.4 MB
»                GPU Memory Free (/nvidiagpu/0/smalldata/1): 10461.6 MB
»
»  Samsung SSD 860 EVO M.2 500GB:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 30.0 °C
»        Data:
»                Total Bytes Written (/hdd/0/data/0): 5329 GB
»        Load:
»                Used Space (/hdd/0/load/0): 63.3 %
»
»  TOSHIBA MQ01ABC150:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 24.0 °C
»        Load:
»                Used Space (/hdd/1/load/0): 24.7 %
[... 143 lines cut ...]
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                "Scope": "Day",
                "Schedule": [
                  "00:00 -> 10:00",
                  "20:00 -> 23:59"
                ],
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 255, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
	}	
]
}

MoshiMoshi0, 2019-09-25:
Huh, ok somehow I did not notice that. There seems to be a bug in LibreHardwareMonitor.

For now you should be able to just copy the `OpenHardwareMonitorLib.dll` from the latest release and paste it to the develop build.

ray-holmium, 2019-09-25:
Beautiful, that fixed it. Thank you!!

MoshiMoshi0, 2019-09-26:
This bug is now fixed in LibreHardwareMonitor.
You can try [this](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/27691162/artifacts) develop build which has the updated LibreHardwareMonitor.

Thanks for reporting!

ray-holmium, 2019-09-26:
Ok, I tested the develop build you linked. LibreHardwareMonitor is indeed working again. But the build throws an error when I try to run in console:

12:15:43.8092: ================================================================
12:15:43.8292: Initializing...
12:15:44.0639: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.7.0.0]
12:15:44.0639: Loading plugin assembly: TTController.Plugin.FlowEffect [0.7.0.0]
12:15:44.0639: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.7.0.0]
12:15:44.0639: Loading plugin assembly: TTController.Plugin.PerLedColorEffect [0.7.0.0]
12:15:44.0639: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.7.0.0]
12:15:44.0639: Loading plugin assembly: TTController.Plugin.PulseEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.RawEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.RippleEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.SensorEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.SoundEffect [0.7.0.0]
12:15:44.0784: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.WaveEffect [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.7.0.0]
12:15:44.0939: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.DpsgController [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.RiingController [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.7.0.0]
12:15:44.1089: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.7.0.0]
12:15:44.1240: Creating Config Manager...
12:15:44.3038: Loading config...
12:15:44.4638: Creating DataCache...
12:15:44.4688: Creating Sensor Manager...
12:15:44.4688: Initializing Open Hardware Monitor...
12:15:45.5340: Detected 57 sensors
12:15:45.5340: Creating Effect Manager...
12:15:45.5440: Creating Speed Controller Manager...
12:15:45.5440: Creating Device Manager...
12:15:45.6488: Searching for "Dpsg" controllers
12:15:45.9611: Searching for "Riing" controllers
12:15:45.9806: Searching for "Riing Plus" controllers
12:15:45.9989: Initialized "Riing Plus" controller [0x264A, 0x1FA5]
12:15:45.9989: Initialized "Riing Plus" controller [0x264A, 0x1FA7]
12:15:45.9989: Initialized "Riing Plus" controller [0x264A, 0x1FA6]
12:15:45.9989: Searching for "Riing Trio" controllers
12:15:46.0128: Adding effect: FullColorEffect [9365b812-b40d-4ae8-84ba-7d8025e66f88]
12:15:46.0128: Adding effect: FullColorEffect [9365b812-b40d-4ae8-84ba-7d8025e66f88]
12:15:46.0128: Adding speed controller: PwmSpeedController [9365b812-b40d-4ae8-84ba-7d8025e66f88]

Process is terminated due to StackOverflowException.

MoshiMoshi0, 2019-09-26:
Yes, another bug in LibreHardwareMonitor...

Sorry about that, there was a big update there recently and it seems that it was not checked enough.
Just keep using the the previous develop build with release dll.

I'll reopen this and update when hopefully fixed.

ray-holmium, 2019-09-26:
No apology necessary, my friend! I'll stay ready to test the next build.

MoshiMoshi0, 2019-09-28:
Ok, I've fixed the bug and done some LibreHardwareMonitor testing and I think it will be fine now.
Try this build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/27713665/artifacts

MoshiMoshi0, 2019-10-04:
Closing as the latest LibreHardwareMonitor seems to work fine.
Feel free to comment here if anything else breaks.

closed 2019-10-04 by MoshiMoshi0