MoshiMoshi0/TTController #30: Razer Chroma Connect

vrtllg, 2019-12-25:
Hi, Razer chroma doesn't connect. Could you helping me?

`{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "10af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1],
        [9802, 8102, 2],
        [9802, 8102, 3],
        [9802, 8102, 4],
        [9802, 8102, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "StaticSpeedController", 
          "Config": {
            "Speed": 50,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
 "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/8"],
      "Config": {
        "CriticalValue": 70
      }
    }
  ]
}`

MoshiMoshi0, 2019-12-25:
I can't really help much because I dont have anything from razer so I dont use synapse.
But here it the thermaltake guide: https://www.thermaltake.com/pub/media/productattach/db/support/usermanual/_9f2cf3913a424d9aa69df7b471f88ab5.pdf 
You only follow steps in synapse.

I assume you installed synapse 3 with connect module.

vrtllg, 2019-12-25:
I 

> I can't really help much because I dont have anything from razer so I dont use synapse.
> But here it the thermaltake guide: https://www.thermaltake.com/pub/media/productattach/db/support/usermanual/_9f2cf3913a424d9aa69df7b471f88ab5.pdf
> You only follow steps in synapse.
> 
> I assume you installed synapse 3 with connect module.
> 
> edit: And there is no configuration on TTController side, it just receives the colors from synapse.

The Synapse recognize your software but the effects doesn't work.. maybe the synapse version has a little bug with that. I tried the TT Software too and the same thing happens..

The sync works but the in the effects, nothing happens.

But Thanks anyway, it's a nice project.

MoshiMoshi0, 2019-12-25:
What do you mean by the effects don't work? Are the fan leds completely off or they just dont change colors (leds are on but stuck on one color)?

Also, could you change this line 
`<logger name="*" minLevel="Info" writeTo="file" enabled="true" />` 
to this 
`<logger name="*" minLevel="Trace" writeTo="file" enabled="true" />` 
in `TTController.Service.exe.Config` then restart the service, run it for 10 seconds, stop it and post the latest log file from Logs folder?

vrtllg, 2019-12-25:
> What do you mean by the effects don't work? Are the fan leds completely off or they just dont change colors (leds are on but stuck on one color)?
>
> Also, could you change this line
> `<logger name="*" [...]

Are the fan leds completely off.. 
This happens with the original TT Plus Software too.. they show me a sucessfull connecting message but the leds just stay off.

I'm trying to talk with Razer Support if they could helping me and i made this video for expain to them:

https://drive.google.com/file/d/1S5V46GzWun8V-L2xrtDWM6RI3PPImLVA/view?usp=sharing
PS.: Sorry for my english, i'm from Brazil

vrtllg, 2019-12-25:
> What do you mean by the effects don't work? Are the fan leds completely off or they just dont change colors (leds are on but stuck on one color)?
>
> Also, could you change this line
> `<logger name="*" [...]

[current.log](https://github.com/MoshiMoshi0/TTController/files/4000723/current.log)
The log file

MoshiMoshi0, 2019-12-25:
Thanks for the log, you can change that line back to what it was so your log files are not so big.

It looks like you are running the official thermaltake software and TTController at the same time since some of the data is corrupted. Make sure you dont run both at the same time.

The razer effect does not even try to send colors to the fans meaning it did not connect to the razer sdk or the sdk returned the status of the connect module to be off.
I would try to completely uninstall razer synapse and razer sdk, reboot and install again.

I will close this issue for now since this seems to be an issue with synapse.
Feel free to comment here if you get it to work with official thermaltake software but it wont work in TTController and I will reopen.

closed 2019-12-25 by MoshiMoshi0