MoshiMoshi0/TTController #31: Configuration Help

TraceKotae, 2020-01-04:
**Describe your desired configuration**
First of all really great job on the project, had it working earlier, but my old config was lost during a pc reset and I somehow cant get it working again and would appreciate your help :D
I have 1 controller with 5 devices attached (1 Pump, 4 fans).
Desired config would be the following curve: 
35°C -> 20%
45°C -> 100%
For the effect the RazerChroma integration would be nice.
Would really appreciate your help, its a great project

**Hardware info**
```
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8116
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 50, Rpm: 886, Unknown: 255]
»                        Identifier: [9802, 8116, 1]
»                2:
»                        Data: [PortId: 2, Speed: 50, Rpm: 871, Unknown: 255]
»                        Identifier: [9802, 8116, 2]
»                3:
»                        Data: [PortId: 3, Speed: 50, Rpm: 885, Unknown: 255]
»                        Identifier: [9802, 8116, 3]
»                4:
»                        Data: [PortId: 4, Speed: 50, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8116, 4]
»                5:
»                        Data: [PortId: 5, Speed: 100, Rpm: 1114, Unknown: 255]
»                        Identifier: [9802, 8116, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Intel Core i7-4790:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 39,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 39,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 44,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 41,0 °C
»                CPU Package (/intelcpu/0/temperature/4): 44,0 °C
»
»  NVIDIA GeForce GTX 970:
»        Temperature:
»                GPU Core (/nvidiagpu/0/temperature/0): 27,0 °C
»
»  KINGSTON RBU-SMS100S360GD:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 41,0 °C
»
»  WDC WD10EZEX-21M2NA0:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 35,0 °C
```

**Current configuration**
```{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "20af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
	[9802, 8116, 4],
	[9802, 8116, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [35, 0],
              [45, 100]
            ],
            "Sensors": ["/nvidiagpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 139],
            "Trigger": {
              "Type": "OneTimeTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
	[9802, 8116, 4],
	[9802, 8116, 5]
      ],
      "Speed": 35
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/nvidiagpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-01-04:
I can't find anything wrong with the config, I think you were just missing the main {} brackets.

Some notes on the config tho:
* You are using only the gpu sensor to control the fans, most of the time the gpu is almost not used so your cpu can be at 90C but the fans will not respond. To fix: 
   * Switch to cpu sensor
   * Or, add the cpu sensor to the Sensors array and use SensorMixFunction.Max to select the highest sensor value
   * Or, split cpu/gpu fans to two profiles
* I dont think you can control the pump speed, port 4 does not report rpm so I assume that is the pump. If you can feel/hear the pump speed changing with the fans I would move it to a separate profile and set a different curve profile so the pump does not go below 30-50%, or set it to always 100%.
* Setting the fans to 0% can be quite dangerous if the temperature will fluctuate between 35-36C because the fans have a minimum operating speed of 20%. So at 35C they will be set to 0%, but at 36C at 20%, that might add wear to the fans. I'd say do this only if you know what you are doing and your idle temps are around 30C.

This config should work:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Guid": "20af9207-7e67-4581-9d13-506cad5d53c1",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
        [9802, 8116, 4],
        [9802, 8116, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [35, 0],
              [45, 100]
            ],
            "Sensors": ["/nvidiagpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8116, 1],
        [9802, 8116, 2],
        [9802, 8116, 3],
        [9802, 8116, 4],
        [9802, 8116, 5]
      ],
      "Speed": 35
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/nvidiagpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

TraceKotae, 2020-01-04:
wow that was fast :D, the config is working now, awesome stuff!
Yeah, i have two loops, one for the GPU thats used mainly for rendering videos and one seperate for the CPU, so the GPU sits either around 20 in idle or around 50 when in action thats why I´d like the fans completely off when im working.
And yes port 4 is the pump, but its not hooked up to a PWM and always runs the same speed.
Great project, I love it :3

closed 2020-01-04 by TraceKotae