MoshiMoshi0/TTController #32: Configuration Help

jlaineinat0r, 2020-01-18:
**Describe your desired configuration**
I'm stuck on configuring color with PingPongEffect - I've been able to get everything else working properly but I'm missing something. Any guidance would be appreciated. 

**Hardware info**
```
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 50, Rpm: 946, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 50, Rpm: 944, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 50, Rpm: 946, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 50, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6798D:
»        Temperature:
»                Temperature #1 (/lpc/nct6798d/temperature/1): 41.0 °C
»                Temperature #2 (/lpc/nct6798d/temperature/2): 34.0 °C
»                Temperature #4 (/lpc/nct6798d/temperature/4): 14.0 °C
»                Temperature #5 (/lpc/nct6798d/temperature/5): 8.0 °C
»                Temperature #6 (/lpc/nct6798d/temperature/6): 49.0 °C
»
»  Intel Core i9-9900K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 47.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 51.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 46.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 49.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 45.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 47.0 °C
»                CPU Core #7 (/intelcpu/0/temperature/6): 45.0 °C
»                CPU Core #8 (/intelcpu/0/temperature/7): 48.0 °C
»                CPU Package (/intelcpu/0/temperature/8): 51.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/9): 53.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/10): 49.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/11): 54.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/12): 51.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/13): 55.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/14): 53.0 °C
»                CPU Core #7 Distance to TjMax (/intelcpu/0/temperature/15): 55.0 °C
»                CPU Core #8 Distance to TjMax (/intelcpu/0/temperature/16): 52.0 °C
»                Core Max (/intelcpu/0/temperature/17): 51.0 °C
»                Core Average (/intelcpu/0/temperature/18): 47.3 °C
»
»  NVIDIA GeForce GTX 1080 Ti:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 42.0 °C
»
»  Samsung SSD 860 QVO 1TB:
»        Temperature:
»                Temperature (/ssd/4/temperature/0): 26.0 °C
»
»  Samsung SSD 960 EVO 500GB:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 40.0 °C
»                Temperature 1 (/nvme/0/temperature/6): 40.0 °C
»                Temperature 2 (/nvme/0/temperature/7): 48.0 °C
»
»  Samsung SSD 950 PRO 256GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 41.0 °C
»
```

**Current configuration**
```{
  "Profiles": [
    {
      "Name": "Main",
      "Guid": "01c14011-919a-478e-8941-0949115b2887",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [35, 40],
              [40, 45],
              [45, 50],
              [50, 55],
              [55, 60],
              [60, 65],
              [65, 75],
              [70, 85],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/8","/nvidiagpu/0/temperature/0"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "PingPongEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "Pump",
      "Guid": "4522b673-2f58-469c-bbad-6fa2a3487724",
      "Ports": [
        [9802, 8101, 4]
      ],
      "SpeedControllers": [],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/intelcpu/0/temperature/8"],
            "ColorGradient": [
              [40, [0, 0, 255]],
              [41, [0, 42, 255]],
              [42, [0, 85, 255]],
              [43, [0, 127, 255]],
              [44, [0, 170, 255]],
              [45, [0, 212, 255]],
              [46, [0, 255, 255]],
              [47, [0, 255, 212]],
              [48, [0, 255, 170]],
              [49, [0, 255, 127]],
              [50, [0, 255, 85]],
              [51, [0, 255, 42]],
              [52, [0, 255, 0]],
              [53, [42, 255, 0]],
              [54, [85, 255, 0]],
              [55, [127, 255, 0]],
              [56, [170, 255, 0]],
              [57, [212, 255, 0]],
              [58, [255, 255, 0]],
              [59, [255, 212, 0]],
              [60, [255, 170, 0]],
              [61, [255, 127, 0]],
              [62, [255, 85, 0]],
              [63, [255, 42, 0]],
              [64, [255, 0, 0]],
              [65, [255, 0, 42]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 5]],
      "Config": {
      "Name": "CPU Block",
      "LedRotation": [10],
      "LedReverse": [false]
    }
    },
    {
      "Ports": [[9802, 8101, 4]],
      "Config": {
      "Name": "Pump",
      "LedRotation": [0],
      "LedReverse": [false]
    }
    },
    {
      "Ports": [[9802, 8101, 3]],
      "Config": {
      "Name": "Top Fan",
      "LedRotation": [9],
      "LedReverse": [false]
    }
    },
    {
      "Ports": [[9802, 8101, 2]],
      "Config": {
      "Name": "Middle Fan",
      "LedRotation": [11],
      "LedReverse": [false]
    }
    },
    {
      "Ports": [[9802, 8101, 1]],
      "Config": {
      "Name": "Bottom Fan",
      "LedRotation": [11],
      "LedReverse": [false]
    }
    }
  ],
  "TemperatureTimerInterval": 250,
  "DeviceSpeedTimerInterval": 2500,
  "DeviceRgbTimerInterval": 32
}
```

MoshiMoshi0, 2020-01-18:
Do you mean that you have trouble configuring the effect to look correct, or does the effect not change the leds at all?

jlaineinat0r, 2020-01-18:
I've been able to adapt the rotation with the individual ports perfectly, but color so far has eluded me. I can't get anything other than pure white out of the configuration for the ports assigned to PingPongEffect.

MoshiMoshi0, 2020-01-18:
Oh, well thats true. It can only do white for now.

I'll add an option to set a gradient for colors.
Anything else you might want to customize?

jlaineinat0r, 2020-01-18:
Overall speed across the span of the ports that PingPongEffect applies to possibly? I'm stuck with 3 controllers (iCue for DRAM, AURA for the motherboard assembly) so trying to coordinate them. I have 4 devices tied to PingPongEffect, 2 on Asus' AURA, and 2 on iCue - none have nearly the level of control that this provides though.

MoshiMoshi0, 2020-01-18:
These are values you can configure now:
* "Step", default 0.01, how fast is the strip moving
* "Height", default 0.2, how "tall" is the moving strip
* "Width", default 0.5, how "wide" is the moving strip, value 1.0 will mean it lights the whole fan

jlaineinat0r, 2020-01-18:
I tried "Step" - but was met with an exception, thought it was a formatting issue on my end to be honest. 

``Application: TTController.Service.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at TTController.Service.TTService.Finalize(TTController.Service.Config.Data.ComputerStateType)
   at TTController.Service.TTService.OnStop()
   at System.ServiceProcess.ServiceBase.DeferredStop()
   at TTController.Service.TTService.OnStart(System.String[])
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(System.Object)
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase[])
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase)
   at TTController.Service.Program.Main(System.String[])``

MoshiMoshi0, 2020-01-18:
Please post the whole log file.

This was an error when shutting down the service so the config should work just fine.  
First time I've seen one on shut down.

```json
        {
          "Type": "PingPongEffect",
          "Config": {
            "Step": 0.01,
            "Height": 0.2,
            "Width": 0.5,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

jlaineinat0r, 2020-01-18:
That worked fine, and was verbatim from what I had before on my backup file. I must have formatted the json wrong somewhere. :/

I'd consider this a feature request at this point limited to just gradient.

MoshiMoshi0, 2020-01-18:
Try this dev build:
https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/30209120/artifacts

Sample config:
```json
        {
          "Type": "PingPongEffect",
          "Config": {
            "Step": 0.01,
            "Height": 0.2,
            "Width": 1.0,
            "ColorGradient": [
              [0.00, [255,0,0]],
              [0.31, [255,0,0]],
              [0.33, [255,255,255]],
              [0.35, [0,255,0]],
              [0.64, [0,255,0]],
              [0.66, [255,255,255]],
              [0.68, [0,0,255]],
              [1.00, [0,0,255]],
            ],
            "EnableSmoothing": true,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

Gradient goes from 0.0 (bottom of first fan) to 1.0 (top of last fan)

jlaineinat0r, 2020-01-19:
I can get a full cycle on the gradient and get stopped by

`The Thermaltake Controller service is marked as an interactive service.  However, the system is configured to not allow interactive services.  This service may not function properly.`

It works in the entirety up until that first cycle though. 👍

MoshiMoshi0, 2020-01-19:
That message should not matter, if it was working before it should work now. Maybe its just crashing.

Check the latest log, or start the service in console mode to see why it stops.

jlaineinat0r, 2020-01-19:
Console isn't showing it stopping or any error (in fact the pump continues to change color based on CPU temp just fine) - but the PingPong effect cycles once and stops.

MoshiMoshi0, 2020-01-19:
Whoops, yes I did not recompile after one last change.

Fixed build: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/30216277/artifacts

jlaineinat0r, 2020-01-19:
Works perfectly! Thank you for the feature add!

MoshiMoshi0, 2020-01-19:
Nice!

I also just now fixed one issue I noticed. 

If you had for example 2 fans setup with this gradient
```
[0.0, [255,0,0]],
[0.5, [255,0,0]],
[0.5, [0,255,0]],
[1.0, [0,255,0]],
```
you would want to have one fan red and one fan green, but because of how the gradients work that is not what was happening as the last led of the first fan was getting a green color instead of red.

Fixed by shrinking the radius of the fan to a little. Physically the leds are also not perfectly on the outer edge of the fan so that works. 
Build here: https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/30216848/artifacts

closed 2020-01-19 by MoshiMoshi0