MoshiMoshi0/TTController #33: Configuration Help

jmaeshawn, 2020-02-02:
**Describe your desired configuration**
I switched to this from the buggy pile of garbage that is the official TT software, but now I'm confused as to how to get this configured the way I want it to.

I have 2 controllers daisy-chained together. The first has 4 Pure Plus 12 Premium Edition fans and a Riing Plus 20 Premium Edition fan. The second controller has 1 additional Pure Plus 12 Premium Edition fan.

What I would like is to have all 6 fans to have sky blue color (Photoshop tells the RGB code for it is 0, 78, 255 if that helps.) I want the 5 Pure Plus 12 fans to have a static color at 15% brightness, and I want the Riing Plus 20 to be 25% brightness and have something similar to the "Radar" effect that I had with the official software before it started going haywire.

If possible, I'd also like to have all 6 lights automatically turn off at 10:00pm and turn back on at 7:00am.

For fan speeds, I'd like something like 28°C= 20% and 45°C = 100%.

I hope this wasn't more detail than you wanted.

**Hardware info**
```
Controllers
-------------------------------
≫
≫  Name: Riing Plus
≫       VendorId: 9802
≫       ProductId: 8101
≫       Ports:
≫               1:
≫                       Data: [PortId: 1, Speed: 35, Rpm: 647, Unknown: 255]
≫                       Identifier: [9802, 8101, 1]
≫               2:
≫                       Data: [PortId: 2, Speed: 35, Rpm: 317, Unknown: 255]
≫                       Identifier: [9802, 8101, 2]
≫               3:
≫                       Data: [PortId: 3, Speed: 35, Rpm: 655, Unknown: 255]
≫                       Identifier: [9802, 8101, 3]
≫               4:
≫                       Data: [PortId: 4, Speed: 35, Rpm: 617, Unknown: 255]
≫                       Identifier: [9802, 8101, 4]
≫               5:
≫                       Data: [PortId: 5, Speed: 35, Rpm: 648, Unknown: 255]
≫                       Identifier: [9802, 8101, 5]
≫       Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
≫  Name: Riing Plus
≫       VendorId: 9802
≫       ProductId: 8102
≫       Ports:
≫               1:
≫                       Data: [PortId: 1, Speed: 35, Rpm: 644, Unknown: 255]
≫                       Identifier: [9802, 8102, 1]
≫               2:
≫                       Data: [PortId: 2, Speed: 35, Rpm: 0, Unknown: 255]
≫                       Identifier: [9802, 8102, 2]
≫               3:
≫                       Data: [PortId: 3, Speed: 35, Rpm: 0, Unknown: 255]
≫                       Identifier: [9802, 8102, 3]
≫               4:
≫                       Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
≫                       Identifier: [9802, 8102, 4]
≫               5:
≫                       Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
≫                       Identifier: [9802, 8102, 5]
≫       Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
≫
-------------------------------
Sensors
-------------------------------
≫
≫  Nuvoton NCT6791D:
≫       Temperature:
≫               CPU Core (/lpc/nct6791d/temperature/0): 28.0 °C
≫               Temperature #1 (/lpc/nct6791d/temperature/1): 29.5 °C
≫               Temperature #2 (/lpc/nct6791d/temperature/2): 29.0 °C
≫               Temperature #3 (/lpc/nct6791d/temperature/3): 94.0 °C
≫               Temperature #4 (/lpc/nct6791d/temperature/4): 101.0 °C
≫               Temperature #5 (/lpc/nct6791d/temperature/5): 100.0 °C
≫               Temperature #6 (/lpc/nct6791d/temperature/6): 101.0 °C
≫
≫  Intel Core i7-4930K:
≫       Temperature:
≫               CPU Core #1 (/intelcpu/0/temperature/0): 25.0 °C
≫               CPU Core #2 (/intelcpu/0/temperature/1): 29.0 °C
≫               CPU Core #3 (/intelcpu/0/temperature/2): 25.0 °C
≫               CPU Core #4 (/intelcpu/0/temperature/3): 28.0 °C
≫               CPU Core #5 (/intelcpu/0/temperature/4): 19.0 °C
≫               CPU Core #6 (/intelcpu/0/temperature/5): 23.0 °C
≫               CPU Package (/intelcpu/0/temperature/6): 29.0 °C
≫               CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 65.0 °C
≫               CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 61.0 °C
≫               CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 65.0 °C
≫               CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 62.0 °C
≫               CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 71.0 °C
≫               CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 67.0 °C
≫               Core Max (/intelcpu/0/temperature/13): 29.0 °C
≫               Core Average (/intelcpu/0/temperature/14): 24.8 °C
≫
≫  NVIDIA GeForce GTX 780 Ti:
≫       Temperature:
≫               GPU Core (/gpu/0/temperature/0): 34.0 °C
≫
≫  Hitachi HDS5C4040ALE630:
≫       Temperature:
≫               Temperature (/hdd/1/temperature/0): 33.0 °C
≫
≫  WDC WD40EZRX-00SPEB0:
≫       Temperature:
≫               Temperature (/hdd/4/temperature/0): 30.0 °C
≫
≫  WDC WD60EZRZ-00GZ5B1:
≫       Temperature:
≫               Temperature (/hdd/3/temperature/0): 29.0 °C
≫
≫  WDC WD40EZRX-00SPEB0:
≫       Temperature:
≫               Temperature (/hdd/6/temperature/0): 33.0 °C
≫
≫  Samsung SSD 840 PRO Series:
≫       Temperature:
≫               Temperature (/ssd/2/temperature/0): 26.0 °C
≫
≫  Hitachi HDS5C4040ALE630:
≫       Temperature:
≫               Temperature (/hdd/0/temperature/0): 32.0 °C
≫
≫  WDC WD40EZRX-00SPEB0:
≫       Temperature:
≫               Temperature (/hdd/5/temperature/0): 29.0 °C
≫
-------------------------------
Press any key to continue...
        
```

**Current configuration**
```
I don't have one yet, the fans are just doing their default rainbow effect that they do out of the box.
```

MoshiMoshi0, 2020-02-02:
Do you have a video how the radar effect looks? I would have to add it I think.

jmaeshawn, 2020-02-03:
Luckily, I managed to take a short video of the effect when the official software was still working semi-properly. You can see it here:
https://drive.google.com/open?id=1byatbF7e1NvjqJcBkIaEDHoHWNewX78d

It had an option to do that effect with either a single color or a gradient, but the one in my video is with a single color.

I really hope you can add this effect :)

MoshiMoshi0, 2020-02-03:
Oh, that is called ripple effect. Its already implemented.

Below should be your config.

The color you posted did not look like sky blue for me so I modified it.

I also added a "Shutdown" profile, when the pc or the service shuts down it will change the colors to red and set the fans at 35% for safety. You can remove or tweak it if you dont like it. 
You could also add a "Boot" profile that saves settings to the controller box so you wont see the default color gradient effect.

Obviously you wont see anything right now if your time does not match your desired schedule.
You can invert the schedule to test how it will look by changing `"Value": true,` to `"Value": false,` in both ScheduleTrigger configs.

```json
{
  "Profiles": [
    {
      "Name": "PurePlus12",
      "Guid": "01c14011-919a-478e-8941-0949115b2887",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [28, 20],
              [45, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/6"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [22, 45, 55],
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                "Scope": "Day",
                "Value": true,
                "Schedule": [
                  "07:00 -> 22:00"
                ]
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "RiingPlus20",
      "Guid": "01c14011-919a-478e-8941-0949115b2888",
      "Ports": [
        [9802, 8101, 2]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [28, 20],
              [45, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/6"],
            "MinimumChange": 4,
            "MaximumChange": 8,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RippleEffect",
          "Config": {
            "Color": [37, 76, 91],
            "Length": 5,
            "TickInterval": 3,
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                "Scope": "Day",
                "Value": true,
                "Schedule": [
                  "07:00 -> 22:00"
                ]
              }
            }
          }
        },
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [0, 0, 0],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1]
      ],
      "Speed": 35,
      "EffectType": "Full",
      "EffectColors": [
        [50, 0, 0]
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/6"],
      "Config": {
        "CriticalValue": 50
      }
    }
  ]
}
```

jmaeshawn, 2020-02-03:
I guess they changed the name of the Ripple effect to the "Radar" effect in the recent version of the official software I had.. But wow, you sure are quick at making all of that work! It even boots/shuts down with the color I wanted now. ^_^

I did notice something though. Does TTController have the ability to adjust the brightness of the lights? Right now the effects are as I wanted them, but the LEDs all seem still to be at 100% brightness.

[image]

As you can see in the screenshot I took before I uninstalled TT RGB Plus, there was a slider that allowed the user to adjust the brightness of the LEDs, which I had set to display at about 20-25%. Does TTController have something similar? The reason I ask is because the computer sits right in front of me, and 100% brightness is a little too much for staring at it directly for long periods of time.

Also, and this doesn't have anything to do with the config I'm after, but I'm just curious as to what effects are possible during the Boot and Shutdown states. Do these states only accept static colors?

MoshiMoshi0, 2020-02-03:
There is no brightness even in TT RGB Plus, less bright colors are just simply colors that are closer to black.
So `[0, 0, 255]` = 100% bright blue, `[0, 0, 128]` = 50% bright blue, `[0, 0, 64]` = 25% bright blue, etc.

You can use a HSV color picker like [this](https://alloyui.com/examples/color-picker/hsv.html) to generate RGB colors you want. If you put [0, 0, 255] in the RGB column, the brighness is the V value/slider.

Boot/Shutdown/Suspend profiles can only use built in effects that are on the controller box since the service cant control them anymore. Those are the `Available effect types` you see in the debug report. RiingPlus supports a lot of them but newer controllers like RiingTrio have only implemented "PerLed" effect (not tested since I only have RiingPlus), so just static colors are possible.

jmaeshawn, 2020-02-03:
Thank you for your explanation on how the "brightness" actually works. I got my lights to the brightness I wanted thanks to that.

I did have one other question that stemmed from what you said about the Boot/Shutdown profiles. The debug log for my fan said that "Pulse_Fast" was an available effect, so I decided I wanted them to pulse as the computer turns on and off.

I set it up with the following:
`{
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5],
        [9802, 8102, 1]
      ],
      "Speed": 35,
      "EffectType": "Pulse_Fast",
      "EffectColors": [
        [0, 23, 77]
      ]
    }`
However, when the computer begins shutting down,the effect turns out how you see in [this video](https://drive.google.com/open?id=1MyDyJH5PVW0AeqB4Ca9QqqAhGvPOzWY4) I just took.

Am I doing something wrong that makes only a small portion of the ring (and all the other fans) light up while pulsing? 

In trying to fix the pulsing issue, I thought that maybe I would need to either add a second "StateType": "Shutdown", to run a "Full" effect in conjunction with the pulse effect, but that didn't change anything. I also considered that a OneTimeTrigger or PulseTrigger or something might be needed, but since the documentation for the triggers hasn't been completed yet, I have no idea how they work, or if those even do what I'm trying to accomplish.

Hopefully you can help me with this too. :)

MoshiMoshi0, 2020-02-03:
Its way more simple than you think! You just have to copy the color 12 times for each led.
I guess I should add a helper that will automatically clone the colors for each port to match the led count.

So:
```json
      "EffectColors": [
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77],
        [0, 23, 77]
      ]
```

jmaeshawn, 2020-02-03:
Oh, I see. I suppose listing each color separately allows someone to have each LED be a different color if they wanted to. But I agree that it would be very helpful if you added some sort of function/helper/whatever that makes all the LEDs be the same color if only 1 color is specified.

What would also be very helpful is when the documentation and examples for how to use all the features this great tool you've written is complete. At the moment, most of the pages [here](https://moshimoshi0.github.io/TTController/) come up as 404 errors or are blank.

Anyway, thank you so much for all the help in configuring this, and for making a tool that is a million times better than the official software! Keep up the good work on it! ^_^

closed 2020-02-03 by jmaeshawn

MoshiMoshi0, 2020-02-03:
Yea I'll add that helper soon.

I just started with the docs so only basic stuff is in for now, but I plan to add at least one example for each speed controller, effect and trigger with explanation of what it does. Also maybe a few full configs as samples.

Anyways, I'm glad you find this useful!