MoshiMoshi0/TTController #35: Configuration Help

heyolli, 2020-02-05:
Hey,
thank you for you software its a godsent compared to the original software (only setting up is kind of hard..). I messed around with it a good portion but sadly i can only get the front side ring of my trio fans illuminated, the fan and the back side ring are not illuminated at all. since i dont have much clue of programming in general it would be really kind of you if you can help me out setting it up properly. I attached my try of setting it up

Thanks in advance

**Describe your desired configuration**

i want controller [9802, 8501, 3], [9802, 8501, 4], [9802, 8501, 5], [9802, 8502, 1] to be controlled by my GPU Temp and [9802, 8502, 2], [9802, 8502, 3], [9802, 8502, 4] controlled by my CPU Temp. I Got two controller in my system and all my connected devices are Riing Trio 120 fans. For now one solid color will do, the rest i can figure out myself.

**Hardware info**

```
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8502
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 56, Rpm: 971, Unknown: 255]
»                        Identifier: [9802, 8502, 1]
»                2:
»                        Data: [PortId: 2, Speed: 40, Rpm: 568, Unknown: 255]
»                        Identifier: [9802, 8502, 2]
»                3:
»                        Data: [PortId: 3, Speed: 40, Rpm: 594, Unknown: 255]
»                        Identifier: [9802, 8502, 3]
»                4:
»                        Data: [PortId: 4, Speed: 40, Rpm: 599, Unknown: 255]
»                        Identifier: [9802, 8502, 4]
»                5:
»                        Data: [PortId: 5, Speed: 40, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 5]
»        Available effect types: PerLed
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 42, Rpm: 609, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 42, Rpm: 618, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 56, Rpm: 914, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 56, Rpm: 966, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 56, Rpm: 858, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6793D:
»        Temperature:
»                CPU Core (/lpc/nct6793d/temperature/0): 51,0 °C
»                Temperature #2 (/lpc/nct6793d/temperature/2): 33,0 °C
»                Temperature #3 (/lpc/nct6793d/temperature/3): 12,0 °C
»                Temperature #4 (/lpc/nct6793d/temperature/4): 11,0 °C
»                Temperature #6 (/lpc/nct6793d/temperature/6): -1,0 °C
»
»  Intel Core i7-6700K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 32,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 36,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 27,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 29,0 °C
»                CPU Package (/intelcpu/0/temperature/4): 39,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 68,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 64,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 73,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 71,0 °C
»                Core Max (/intelcpu/0/temperature/9): 36,0 °C
»                Core Average (/intelcpu/0/temperature/10): 31,0 °C
»
»  NVIDIA GeForce RTX 2080:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 42,0 °C
»
»  SanDisk Ultra II 960GB:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 31,0 °C
»
»  :
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 0,0 °C
»
-------------------------------
```

**Current configuration**
```{
  
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8501, 3],
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1]
      ],
            "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 100],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "PortConfig": [
    {
      "Ports": [
        [9802, 8501, 3], 
        [9802, 8501, 4], 
        [9802, 8501, 5]
      ],
      "Config": {
        "LedCount": 30
      }
    }
  ],
      "Effects": [
        {
          
            "Type": "FullColorEffect",
          "Config": {
 	    "Name": "GPU FANS",
	    "DeviceType": "RiingTrio",
            "LedCountHandling": "copy",
            "Color": [255, 255, 255],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ]
}
```

MoshiMoshi0, 2020-02-05:
You did not specify what to do with ports `[9802, 8501, 1]` and `[9802, 8501, 2]` but you should be able to tweak it from here.

```json
{
   "Profiles": [
    {
      "Name": "GPU",
      "Ports": [
        [9802, 8501, 3], 
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 255, 255],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "CPU",
      "Ports": [
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4] 
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [255, 255, 255],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1], 
        [9802, 8501, 2], 
        [9802, 8501, 3], 
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4] 
      ],
      "Config": {
        "DeviceType": "RiingTrio"
      }
    }
  ]
}
```

heyolli, 2020-02-06:
Thanks for your help! works now as intended.

Sadly i cant get some of the effects to work... here is my current config file:

"{
   "Profiles": [
    {
      "Name": "GPU",
      "Ports": [
        [9802, 8501, 3], 
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 0],
              [55, 25],
              [65, 35],
              [70, 50],
              [75, 60]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
             "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "CPU",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4] 
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 15],
              [45, 35],
              [55, 50],
              [65, 65],
              [75, 75]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
[... 214 lines cut ...]
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
2020-02-06 01:31:22.0968|FATAL|TTController.Service.Managers.ConfigManager|Failed to load the config!
2020-02-06 01:31:34.1145|INFO|TTController.Service.TTService|================================================================
2020-02-06 01:31:34.1145|INFO|TTController.Service.TTService|Finalizing...
2020-02-06 01:31:34.1264|INFO|TTController.Service.Managers.ConfigManager|Disposing Config Manager...
"

Thanks in advance

MoshiMoshi0, 2020-02-06:
The config of each effect is different and some have required variables that have to be set for it to work. All effects will be on the docs with a gif of how they look.

Which effect would you like, I'll help you get it set up. Most of the effect names match the names from TT RGB PLUS software.

Also there is no zone support for RiingDuo/RiingDuo so you cant put 2/3 different effects on different zones/ring but that is planned.

heyolli, 2020-02-06:
thats what i thought.

Id like a slow wave effect, but if docs are coming up in a few days i can try doing it myself then.

Have a good one

MoshiMoshi0, 2020-02-06:
You can tweak the colors how you like but there needs to be 30 of them since riing trio has 30 leds.
I will add an option to generate the colors from a gradient so it will be shorter in a future version.

You can tweak the speed by changing the "TickInterval" variable.

```json
        {
          "Type": "WaveEffect",
          "Config": {
            "Colors": [
                [255, 0, 0],
                [255, 0, 51],
                [255, 0, 102],
                [255, 0, 153],
                [255, 0, 204],
                [255, 0, 255],
                [204, 0, 255],
                [154, 0, 255],
                [104, 0, 255],
                [54, 0, 255],
                [0, 0, 255],
                [0, 46, 255],
                [0, 93, 255],
                [0, 140, 255],
                [0, 187, 255],
                [0, 234, 255],
                [0, 238, 212],
                [0, 242, 170],
                [0, 246, 127],
                [0, 250, 85],
                [0, 255, 43],
                [51, 251, 34],
                [102, 248, 25],
                [153, 244, 17],
                [204, 241, 8],
                [255, 238, 0],
                [255, 199, 0],
                [255, 161, 0],
                [255, 123, 0],
                [255, 85, 0]
            ],
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

heyolli, 2020-02-06:
Thank you very much, works like a treat :-)

closed 2020-02-06 by heyolli