MoshiMoshi0/TTController #36: Linux support

redaxmedia, 2020-03-11:
Hello,

is this library flexible enough to be splitted into a core library that supports Windows and Linux? And a library bases in Python that allows to control the devices? The Python API could be the fundament of a CLI too.

MoshiMoshi0, 2020-03-14:
Sorry for late reply, I didnt see your issue.

I was planning to make this cross platform with .netcore but I could not find a cross platform library for reading sensors. At least temperature sensors of CPU/GPU. Everything else was ported and "should" be cross platform, but I did not test.

I've also had a separate branch with the TTController.Service project split into .Hardware, .Config, .Service, .Console projects so I think the code is pretty flexible (I hope) and it took little effort. But I figured that there is no point if I cant make it cross platform.

I don't think I understand what you want to do with python. Like a python wrapper of the core library?

redaxmedia, 2020-03-15:
I am the author of [chroma-feedback](https://github.com/redaxmedia/chroma-feedback) and would love to support Thermaltake devices. It's a software to visualize the build status using various RGB devices such as Razer Chroma and Philips Hue.

> Turn your RGB powered hardware into an build indicator for continuous integration.

Therefore a Python library would be needed to programmatically set colors across devices.

MoshiMoshi0, 2020-03-15:
Well I dont think a python library is in scope of this project. I might write one someday but I do not plan to for now. 
I update [this](https://moshimoshi0.github.io/ttrgbplusapi/index.html) api docs with what I manage to disassemble and figure out from their official software, so maybe you could use that to integrate Thermaltake.

Also "pulsate" would only be supported on Riing Plus fans as first gen Riing fans only support static colors and everything newer than Riing Plus has to have the effects generated manually (eg. you update at 30fps).
The other issue is that you cant really have more then one application writing to the fan controllers as one application can read responses from the others write command. There is no sdk/service that handles this like with many other manufacturers.

Alternative way would be to create a custom effect for TTController that is signaled from chroma-feedback, this would fix the multi app problem.

redaxmedia, 2020-03-21:
In case you support linux. there should be a way to change colors via bash... therfore I could use this raw approach instead of a Python library... not the best but okay.

diffficult, 2020-12-24:
right now there's https://github.com/chestm007/linux_thermaltake_riing project that works fine for fan speeds but fails to push changes to the leds to change colors. 

I believe it's taking data from `lm_sensors` for the fan curves.

redaxmedia, 2020-12-27:
@diffficult It is there for a while but unfortunately the maintainer left GitHub since years. I asked him for a public API using Python in the summer of 2019: https://github.com/chestm007/linux_thermaltake_riing/issues/29

closed 2025-01-22 by henryruhs