MoshiMoshi0/TTController #37: Configuration Help

fmatiasii, 2020-04-16:
Hi, i'm a newbie with all of these.  I need help on this TTcontroller.  I just downloaded the files but i don't see the TTController.Service.Exe
**Hardware info**
```
Replace this text (between ``` characters) with a debug report.
You can generate it by going to "Main Menu -> Debug -> Report" in the command line menu.

Make sure no official software is not running at the same time.
```

**Current configuration**
```json
If applicable, replace this text (between ``` characters) with contents of your current configuration here. 
```

MoshiMoshi0, 2020-04-16:
I think you downloaded the source code, not a release.
All releases are here: https://github.com/MoshiMoshi0/TTController/releases

fmatiasii, 2020-04-16:
>   Thank you so much for your help, I really appreciate it.

one last question, do i have to uninstall the official tt rgb software before installing this one?

fmatiasii, 2020-04-16:
after i have edited the config.json and started it, this is what is says:

Starting the service...
Failed to start the service...
System.ServiceProcess.TimeoutException: Time out has expired and the operation has not been completed.
   at System.ServiceProcess.ServiceController.WaitForStatus(ServiceControllerStatus desiredStatus, TimeSpan timeout)
   at TTController.Service.Program.<ManageServiceMenu>g__StartService|6_1()
Press any key to return to the menu...

MoshiMoshi0, 2020-04-16:
You dont have to uninstall, but you cant run them at the same time.

MoshiMoshi0, 2020-04-16:
> 
> 
> after i have edited the config.json and started it, this is what is says:
> 
> Starting the service...
> Failed to start the service...
> System.ServiceProcess.TimeoutException: Time out has expired and the operation has not been completed.
> at System.ServiceProcess.ServiceController.WaitForStatus(ServiceControllerStatus desiredStatus, TimeSpan timeout)
> at TTController.Service.Program.g__StartService|6_1()
> Press any key to return to the menu...

Try starting in console mode by selecting "Run in console" from main menu.
If it works there, the above error might just mean that the service took a long time to start, but started anyway.

fmatiasii, 2020-04-17:
I tried it but still no luck.  I don't know what i'm doing wrong.

fmatiasii, 2020-04-17:
I also installed git on my desktop and tried to clone a repository but it says "user already exist."

MoshiMoshi0, 2020-04-17:
You dont need to clone the repository, just download the latest release form the link.

What happens when you launch it in console mode?

fmatiasii, 2020-04-17:
nothing happens.  do i need to edit the config.json?

MoshiMoshi0, 2020-04-17:
What do you mean nothing happens? Nothing is shown in the console?
Yes you need to edit the config.json.

fmatiasii, 2020-04-17:
is this the right content?

{
  "Profiles":  [<ProfileConfig>],
  "ComputerStateProfiles":  [<ComputerStateProfileConfig>],
  "PortConfigs":  [<PortConfig>],
  "SensorConfigs":   [<SensorConfig>],

  "CpuSensorsEnabled": <bool>,
  "GpuSensorsEnabled": <bool>,
  "StorageSensorsEnabled": <bool>,
  "MotherboardSensorsEnabled": <bool>,
  "MemorySensorsEnabled": <bool>,
  "NetworkSensorsEnabled": <bool>,
  "ControllerSensorsEnabled": <bool>,

  "SensorTimerInterval ": <int>,
  "DeviceSpeedTimerInterval": <int>,
  "DeviceRgbTimerInterval": <int>,
  "LoggingTimerInterval": <int>
}

MoshiMoshi0, 2020-04-17:
No, this seems like a default generated config, but this
```json
"CpuSensorsEnabled": ,
"GpuSensorsEnabled": ,
"StorageSensorsEnabled": ,
"MotherboardSensorsEnabled": ,
"MemorySensorsEnabled": ,
"NetworkSensorsEnabled": ,
"ControllerSensorsEnabled": ,

"SensorTimerInterval ": ,
"DeviceSpeedTimerInterval": ,
"DeviceRgbTimerInterval": ,
"LoggingTimerInterval": 
```
is not correct as it has no values.
Can you post output from "Debug -> Report" menu?

fmatiasii, 2020-04-17:
can you help me with what must i input?

MoshiMoshi0, 2020-04-17:
Post output from "Debug -> Report" menu

fmatiasii, 2020-04-17:
ITE IT8688E:
»        Temperature:
»                Temperature #1 (/lpc/it8688e/temperature/0): 42.0 °C
»                Temperature #2 (/lpc/it8688e/temperature/1): 49.0 °C
»                Temperature #3 (/lpc/it8688e/temperature/2): 37.0 °C
»                Temperature #4 (/lpc/it8688e/temperature/3): 45.0 °C
»                Temperature #5 (/lpc/it8688e/temperature/4): 42.0 °C
»
»  ITE IT8792E:
»        Temperature:
»                Temperature #1 (/lpc/it8792e/temperature/0): 47.0 °C
»                Temperature #3 (/lpc/it8792e/temperature/2): 43.0 °C
»
»  AMD Ryzen Threadripper 3970X 32-Core Processor:
»        Temperature:
»                Core (Tctl) (/amdcpu/0/temperature/0): 37.4 °C
»                Core (Tdie) (/amdcpu/0/temperature/1): 37.4 °C
»
»  NVIDIA GeForce RTX 2080 Ti:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 30.0 °C
»
»  Samsung SSD 960 EVO 1TB:
»        Temperature:
»                Temperature (/nvme/2/temperature/0): 36.0 °C
»                Temperature 1 (/nvme/2/temperature/6): 36.0 °C
»                Temperature 2 (/nvme/2/temperature/7): 43.0 °C
»
»  GIGABYTE GP-ASACNE2100TTTDR:
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 42.0 °C
»
»  GIGABYTE GP-ASM2NE6500GTTD:
»        Temperature:
»                Temperature (/nvme/3/temperature/0): 37.0 °C
»
»  Samsung SSD 960 PRO 512GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 35.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 35.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 39.0 °C
»

MoshiMoshi0, 2020-04-17:
That is not everything, there should be controllers listed at the top.

fmatiasii, 2020-04-17:
Controllers
-------------------------------
»
»
-------------------------------
Sensors
-------------------------------

MoshiMoshi0, 2020-04-17:
Well then your controller is not detected, what thermaltake fans do you have?

fmatiasii, 2020-04-17:
the triple riing

fmatiasii, 2020-04-17:
maybe i'm not entering the right code?

MoshiMoshi0, 2020-04-17:
Do this:
* Exit the console
* Edit `TTController.Service.exe.Config` file and change 
`<logger name="*" minLevel="Debug" writeTo="console" enabled="true" />`
to
`<logger name="*" minLevel="Trace" writeTo="console" enabled="true" />`
* Again post the output of the "Debug -> Report"

fmatiasii, 2020-04-17:
Controllers
-------------------------------
»
19:01:31.2649: Found plugin assembly for type IControllerDefinition: TTController.Plugin.DpsgController [0.9.0.0]
19:01:31.2649: Loading plugin assembly: TTController.Plugin.DpsgController [0.9.0.0]
19:01:31.2649: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingController [0.9.0.0]
19:01:31.2649: Loading plugin assembly: TTController.Plugin.RiingController [0.9.0.0]
19:01:31.2649: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingPlusController [0.9.0.0]
19:01:31.2649: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.9.0.0]
19:01:31.2649: Found plugin assembly for type IControllerDefinition: TTController.Plugin.RiingTrioController [0.9.0.0]
19:01:31.2649: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.9.0.0]
19:01:31.2649: Creating Device Manager...
19:01:31.2889: Searching for "Dpsg" controllers
19:01:31.3069: Searching for "Riing" controllers
19:01:31.3239: Searching for "Riing Plus" controllers
19:01:31.3419: Searching for "Riing Trio" controllers
19:01:31.3590: Disposing Device Manager...
19:01:31.3590: Disposed devices: 0
»
-------------------------------
Sensors
-------------------------------
»
19:01:31.3590: Initializing Libre Hardware Monitor...
19:01:32.6226: Updating hardware: Gigabyte TRX40 AORUS XTREME
19:01:32.6226: Updating hardware: ITE IT8688E
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/control/0
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/control/1
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/control/2
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/0
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/1
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/2
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/3
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/4
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/5
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/6
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/7
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/voltage/8
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/temperature/0
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/temperature/1
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/temperature/2
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/temperature/3
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/temperature/4
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/fan/1
19:01:32.6226: Valid sensor identifier: /lpc/it8688e/fan/3
19:01:32.6226: Updating hardware: ITE IT8792E
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/control/0
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/control/1
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/control/2
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/0
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/1
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/2
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/3
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/4
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/5
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/6
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/voltage/7
19:01:32.6226: Valid sensor identifier: /lpc/it8792e/temperature/0
19:01:32.6376: Valid sensor identifier: /lpc/it8792e/temperature/2
19:01:32.6376: Updating hardware: AMD Ryzen Threadripper 3970X 32-Core Processor
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/1
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/2
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/3
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/4
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/5
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/6
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/7
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/8
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/9
19:01:32.6376: Valid sensor identifier: /amdcpu/0/load/10
[... 174 lines cut ...]
19:01:33.3712: Updating hardware: Ethernet 2
19:01:33.3712: Valid sensor identifier: /nic/{6D06F4B3-25C0-4CCB-9766-E1A1DEEFD599}/data/2
19:01:33.3712: Valid sensor identifier: /nic/{6D06F4B3-25C0-4CCB-9766-E1A1DEEFD599}/data/3
19:01:33.3712: Valid sensor identifier: /nic/{6D06F4B3-25C0-4CCB-9766-E1A1DEEFD599}/throughput/7
19:01:33.3712: Valid sensor identifier: /nic/{6D06F4B3-25C0-4CCB-9766-E1A1DEEFD599}/throughput/8
19:01:33.3712: Valid sensor identifier: /nic/{6D06F4B3-25C0-4CCB-9766-E1A1DEEFD599}/load/1
19:01:33.3712: Updating hardware: Local Area Connection* 10
19:01:33.3712: Valid sensor identifier: /nic/{48F052F1-D8BF-4968-BBFC-66C9C4BFF6AF}/data/2
19:01:33.3712: Valid sensor identifier: /nic/{48F052F1-D8BF-4968-BBFC-66C9C4BFF6AF}/data/3
19:01:33.3712: Valid sensor identifier: /nic/{48F052F1-D8BF-4968-BBFC-66C9C4BFF6AF}/throughput/7
19:01:33.3712: Valid sensor identifier: /nic/{48F052F1-D8BF-4968-BBFC-66C9C4BFF6AF}/throughput/8
19:01:33.3712: Valid sensor identifier: /nic/{48F052F1-D8BF-4968-BBFC-66C9C4BFF6AF}/load/1
19:01:33.3712: Updating hardware: Local Area Connection* 9
19:01:33.3712: Valid sensor identifier: /nic/{BFBA46BC-F37B-4529-A807-A4E2FB21D123}/data/2
19:01:33.3712: Valid sensor identifier: /nic/{BFBA46BC-F37B-4529-A807-A4E2FB21D123}/data/3
[... cut ...]
19:01:33.3712: Valid sensor identifier: /nic/{69BA52FD-6AC8-4B04-B713-AC9F7C315559}/data/2
19:01:33.3712: Valid sensor identifier: /nic/{69BA52FD-6AC8-4B04-B713-AC9F7C315559}/data/3
19:01:33.3712: Valid sensor identifier: /nic/{69BA52FD-6AC8-4B04-B713-AC9F7C315559}/throughput/7
19:01:33.3873: Valid sensor identifier: /nic/{69BA52FD-6AC8-4B04-B713-AC9F7C315559}/throughput/8
19:01:33.3873: Valid sensor identifier: /nic/{69BA52FD-6AC8-4B04-B713-AC9F7C315559}/load/1
19:01:33.3873: Detected 228 sensors

fmatiasii, 2020-04-17:
i have not yet edited the json

MoshiMoshi0, 2020-04-17:
No controller is detected on your pc. Does the official software work for you?

fmatiasii, 2020-04-17:
no it doesn't work.

fmatiasii, 2020-04-17:
this is what appears when i run console:  19:15:59.6655: ================================================================
19:15:59.6655: Initializing...
19:15:59.6820: Found plugin assembly for type IPlugin: TTController.Plugin.DpsgController [0.9.0.0]
19:15:59.6820: Loading plugin assembly: TTController.Plugin.DpsgController [0.9.0.0]
19:15:59.6820: Found plugin assembly for type IPlugin: TTController.Plugin.RiingController [0.9.0.0]
19:15:59.6820: Loading plugin assembly: TTController.Plugin.RiingController [0.9.0.0]
19:15:59.6820: Found plugin assembly for type IPlugin: TTController.Plugin.RiingPlusController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.RiingTrioController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.BlinkEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.FlowEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.FlowEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.FullColorEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.FullColorEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.PerLedColorEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.PerLedColorEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.PingPongEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.PulseEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.PulseEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.RawEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.RawEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.RazerConnectEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.RippleEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.RippleEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.SensorEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.SensorEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.SoundEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.SoundEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.SpectrumEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.WaveEffect [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.WaveEffect [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.CopySpeedController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.DpsgSpeedController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.PwmSpeedController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.StaticSpeedController [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.AlwaysTrigger [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.9.0.0]
19:15:59.6840: Found plugin assembly for type IPlugin: TTController.Plugin.LogicTrigger [0.9.0.0]
19:15:59.6840: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.9.0.0]
19:15:59.7000: Found plugin assembly for type IPlugin: TTController.Plugin.OneTimeTrigger [0.9.0.0]
19:15:59.7000: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.9.0.0]
19:15:59.7000: Found plugin assembly for type IPlugin: TTController.Plugin.ProcessTrigger [0.9.0.0]
19:15:59.7000: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.9.0.0]
19:15:59.7000: Found plugin assembly for type IPlugin: TTController.Plugin.PulseTrigger [0.9.0.0]
19:15:59.7000: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.9.0.0]
19:15:59.7000: Found plugin assembly for type IPlugin: TTController.Plugin.ScheduleTrigger [0.9.0.0]
19:15:59.7000: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.9.0.0]
19:15:59.7000: Found plugin assembly for type IPlugin: TTController.Plugin.SensorTrigger [0.9.0.0]
19:15:59.7000: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.9.0.0]
19:15:59.7000: Creating Config Manager...
19:15:59.7000: Loading config...
19:15:59.7220: Creating DataCache...
19:15:59.7220: Creating Plugin Store...
19:15:59.7220: Creating Sensor Manager...

MoshiMoshi0, 2020-04-17:
The config does not matter, the report does not use it.

If the official software does not work then either your controller is bad, or you connected it wrong, as it is also not detected by TTController.

Cant help you with that unfortunately. 
If you get the official software working, then reply here and I will create a config for you that you want.

fmatiasii, 2020-04-17:
whenever i run the official software it says "could not find fan controller"  error codeH_0X001

MoshiMoshi0, 2020-04-17:
Yes, because it also cant detect the controller. You should ask thermaltake support and check that you installed it correctly.

closed 2020-04-17 by MoshiMoshi0

fmatiasii, 2020-04-17:
Its not only me whos having this kind of issue, ive read on the forums that it has something to do with amd chipset.  I am able to make it work by disconnecting the controller cable then re-inserting again. But its quite a hassle since i need to do it with 4 controllers.