MoshiMoshi0/TTController #38: Configuration Help

ghosty033, 2020-04-21:
**Describe your desired configuration**
Hey man, its ghosty from the TT forum. I need some help writing the code for the new software. I am a complete dumb ass and as a result i have been unsuccessful.
I know exactly what i want to achieve fan settings wise, it is basically what i have now but it might play around the color profiles and effects later on, hopefully with some successs.
If you link a json code generator that will work with your software id imagine i could handle it myself from there pretty easily lol. Everthing listed below is from the old GUI 1.7 version.

**Hardware info**
I dont have this debug option with the older version. Let me know what hardware information you need.  Currently running the 5 pack Thermaltake Riing Plus 12 with the included Thermaltake controller.

**Current configuration**
My current json file from the old GUI 1.7 verion. probably the only thing i would change about this config is deleting the startup profile as it is basically useless. I only "use" it to see if all my fans are working as most of the time 3 of the fans turn off completely unless required.

"{
  "ProfileMap": [
    {
      "Key": {
        "Guid": "63aa295c-0f0c-4939-9642-511e8dff1db8",
        "Name": "Boot",
        "ProfileType": "Boot",
        "Points": [
          {
            "Temperature": 50.0,
            "Speed": 20
          },
          {
            "Temperature": 60.0,
            "Speed": 25
          },
          {
            "Temperature": 70.0,
            "Speed": 40
          },
          {
            "Temperature": 80.0,
            "Speed": 75
          },
          {
            "Temperature": 85.0,
            "Speed": 100
          }
        ],
        "SpeedMode": "Manual",
        "RgbMode": "Full",
        "RgbSpeed": "Normal",
        "RgbColors": [
          "#333333"
        ],
        "ManualFanSpeed": 50,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureMixFunction": "Max"
      },
      "Value": [
        689174312,
        689174313,
        689174314,
        689174315,
        689174316
      ]
    },
    {
      "Key": {
        "Guid": "1612838e-3f27-429b-96af-5b948a1cbb56",
        "Name": "Intake",
        "ProfileType": "Normal",
        "Points": [
          {
            "Temperature": 60.0,
            "Speed": 30
          },
          {
            "Temperature": 70.0,
            "Speed": 50
          },
          {
            "Temperature": 100.0,
            "Speed": 70
          }
        ],
        "SpeedMode": "Pwm",
        "RgbMode": "ByLed",
        "RgbSpeed": "Slow",
        "RgbColors": [
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919"
        ],
        "ManualFanSpeed": 50,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureSensorIdentifiers": [
          "/amdcpu/0/temperature/1",
          "/amdcpu/0/temperature/0"
        ],
        "TemperatureMixFunction": "Average"
      },
      "Value": [
        689174312,
        689174313
      ]
    },
    {
      "Key": {
        "Guid": "a2af8a87-04dd-46b9-afa0-d6c7a40584e4",
        "Name": "Exhaust",
        "ProfileType": "Normal",
        "Points": [
          {
            "Temperature": 50.0,
            "Speed": 0
          },
          {
            "Temperature": 50.0,
            "Speed": 29
          },
          {
            "Temperature": 80.0,
            "Speed": 50
          }
        ],
        "SpeedMode": "Pwm",
        "RgbMode": "ByLed",
        "RgbSpeed": "Slow",
        "RgbColors": [
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919",
          "#191919"
        ],
        "ManualFanSpeed": 50,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureSensorIdentifiers": [
          "/nvidiagpu/0/temperature/0"
        ],
        "TemperatureMixFunction": "Max"
      },
      "Value": [
        689174314,
        689174315,
        689174316
      ]
    },
    {
      "Key": {
        "Guid": "abcd0349-d117-4e86-8b73-97abddb1cd1a",
        "Name": "Quiet Time",
        "ProfileType": "Normal",
        "Points": [
          {
            "Temperature": 60.0,
            "Speed": 30
          },
          {
            "Temperature": 100.0,
            "Speed": 60
          }
        ],
        "SpeedMode": "Pwm",
        "RgbMode": "Full",
        "RgbSpeed": "Normal",
        "RgbColors": [
          "#000000"
        ],
        "ManualFanSpeed": 50,
        "CriticalTemperature": 90,
        "MinimumFanSpeedChange": 4,
        "MaximumFanSpeedChange": 8,
        "TemperatureSensorIdentifiers": [
          "/amdcpu/0/temperature/1",
          "/amdcpu/0/temperature/0"
        ],
        "TemperatureMixFunction": "Average"
      },
      "Value": []
    }
  ],
  "TemperatureTimerInterval": 250.0,
  "FanTimerInterval": 2500.0
}"

MoshiMoshi0, 2020-04-21:
Do you want to enable "Quiet Time" values at a specific time?

MoshiMoshi0, 2020-04-21:
This should be the ported config but without the "Quiet Time". As you can't manually change profiles you would for example need to specify the desired time you want to switch on/off the quiet time values.

Added shutdown and boot profile but you can easily change them.
 
```json
{
   "Profiles": [
    {
      "Guid": "1612838e-3f27-429b-96af-5b948a1cbb56",
      "Name": "Intake",
      "Ports": [
        [9802, 8101, 1], 
        [9802, 8101, 2],
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [25, 25, 25],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [60, 30],
              [70, 50],
              [100, 70]
            ],
            "Sensors": [
              "/amdcpu/0/temperature/1",
              "/amdcpu/0/temperature/0"
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Guid": "a2af8a87-04dd-46b9-afa0-d6c7a40584e4",
      "Name": "Exhaust",
      "Ports": [
        [9802, 8101, 3], 
        [9802, 8101, 4],
        [9802, 8101, 5],
      ],
      "Effects": [
        {
          "Type": "FullColorEffect",
          "Config": {
            "Color": [25, 25, 25],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [50, 0],
              [50, 29],
              [80, 50]
            ],
            "Sensors": [ "/nvidiagpu/0/temperature/0" ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "Speed": 35,
      "EffectType": "Full",
      "EffectColors": [
        [25, 0, 0]
      ]
    },
    {
      "StateType": "Boot",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "Speed": 35,
      "EffectType": "Full",
      "EffectColors": [
        [0, 25, 0]
      ]
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [              
        "/amdcpu/0/temperature/1",
        "/amdcpu/0/temperature/0",
        "/nvidiagpu/0/temperature/0"
      ],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

ghosty033, 2020-04-22:
Whoops i forgot about the quiet time setting, dw about it dont use it anymore, same with the startup and shutdown profiles. Thankyou so much man i really appreciate it!

That issue in 1.7 where upon switching to my normal profiles from the startup startup profile was really getting annoying. The fan speed would be wrong on the intake and one of exhausts fans would just hang stuck on spinning up and down to no end. It was really getting annoying.

And woohoo the config you made for me works! I had a minor glitch initially as the GPU sensor name was incorrect which caused the exhaust fans to run at 100%. changed to the correct name and it works. Interesting as it takes an amount of time for the fans to ramp down to 0% where as with the old software it would immediately turn them off the moment i made a change.

Now to experiment with different colour modes and effects.

What kind of program do you use to help you with your json files? or are you old school and just use notepad haha. I find it really hard to keep track of having the correct number of closing brackets.

MoshiMoshi0, 2020-04-22:
Yea I use sublime text, it highlights if you have errors in the json.
Or sometimes i use this to visualize the structure better: https://jsoneditoronline.org

Anyways, if you need more help reply here or make a new issue.

closed 2020-04-22 by MoshiMoshi0