MoshiMoshi0/TTController #39: Razer plugin does not map lights properly

MrBradford, 2020-04-21:
**Describe the issue**

The Razer effect plugin takes the five LED colors and maps them directly to the LEDs. However the Razer SDK uses the lights differently. The first light is used for devices where all LEDs should be the same color. The next four are used for devices that have more than one color or for animations, and those default back to the first light if there is no effect/animation defined.

For TT products (I think they all have more than one LED), the easiest solution is to ignore the first light color and just use the four after it. This is what native Razer devices do and how it works in the official TT implementation.

https://developer.razer.com/works-with-chroma/razer-chroma-led-profiles/

I can fix this and send a pull request when I'm not at work :)

MrBradford, 2020-04-21:
I'm getting a permission error when I try to commit. That said, all I did was change RazerConnectEffect.GenerateColors to

return ports.ToDictionary(p => p, _ => _colors.Skip(1).ToList());

Technically this is not fully compliant with the SDK since you're supposed to read the first color for devices that have only  one LED/color, but I don't think there are TT devices that fit that description. Functionally it's correct for all my TT devices and doesn't have the error of an incorrect fifth light.

MoshiMoshi0, 2020-04-21:
Well this is not compliant with the SDK anyway since at least when I was writing this, the only way to receive colors was to make razer provide you with a GUID that is used at initialization. So it was only for brands that have a agreement with razer. TTController is using thermaltake GUID to act as if its thermaltake application.

If you want to do a PR please do it to develop, not master.

MrBradford, 2020-04-21:
Sorry - switched to develop.

MoshiMoshi0, 2020-04-21:
Thanks for the pr 😄 

I should add contribution guidelines to make it clear to pr to develop next time.

closed 2020-04-21 by MoshiMoshi0

MrBradford, 2020-04-21:
No worries - happy to help.