MoshiMoshi0/TTController #41: Configuration Help

Luishi5000, 2020-04-24:
Hi, I have 3 Riing Plus fans hooked up to a standard controller. I would like to replicate the "Aurora" effect found in the original software (which keeps freezing the animation and fan sensors in the stock program). From what I understand I would have to fade between different preset RGB color sets to achieve this, correct? (The effect smoothly flows between colors like the Aurora Borealis).

If so, it seems as though it has to be done via a schedule trigger?

I cannot seem to figure out a way to program fades between color sets based on time intervals (if this is the best approach).

Additionally, it would be nice to fade the brightness of the whole set between two preset system times. I don't see any options for layering modifiers, correct? Can you please provide some direction for this. Thanks in advance!

MoshiMoshi0, 2020-04-24:
I think the default effect is just a [WaveEffect](https://moshimoshi0.github.io/TTController/effects/waveeffect) with rainbow colors. So you would have to set the Colors list with rainbow colors for 12 leds. Unless you are talking about something different.

The develop build also has a nicer way of specifying colors with a gradient instead of one color per led but that is not documented yet.

There is no way to smoothly transition between different effects, its just instant for now.

If you add one effect that has dim colors with a ScheduleTrigger, and then add another with normal colors with AlwaysTrigger it will work like you want. If the ScheduleTrigger is not enabled it will fall back to the next effect.
So:
```json
"Effects": [
	{
	    "Type": "WaveEffect",
	    "Config": {
	        ...
	        "Trigger": {
	        	"Type": "ScheduleTrigger",
	        	"Config": {
	        		...
	        	}
	        }
	    }
	},
	{
	    "Type": "WaveEffect",
	    "Config": {
	        ...
	        "Trigger": {
	        	"Type": "AlwaysTrigger"
	        }
	    }
	}
]
```

Luishi5000, 2020-04-24:
Thank you for the advice. The effect I was hoping to re-create is shown here:
https://youtu.be/TDPuM56G7to?t=716
Which smoothly transitions between different colors in wide gradients within the individual fans. Even if there was a way to combine the LEDColorGradiant effect with multiple colors like PerLedColorEffect? Then use a 30,60,120s loop timer as the sensor to change between them?

The other effect I was hoping to re-create was the "party" mode, but I haven't found a good way to do that:
https://youtu.be/TDPuM56G7to?t=555

Is this the proper way to configure the schedule trigger to work from 8pm to 7am? (Can't test on my computer at the moment.)
{
  "Type": "ScheduleTrigger",
  "Config": {
      "Scope": "Hour",
      "Schedule": "20:00" -> "07:00"
  }
}

Thanks again for your work on this, it's a life saver.

MoshiMoshi0, 2020-04-24:
Like I said there is no way to smoothly transition between effects, only instant.

You can kinda cheat that effect by specifying more colors that you have leds in WaveEffect and creating a custom device (in Plugins\Devices) and finally setting all ports to use that new device type. More colors will create a smooth gradient and WaveEffect will rotate them (and the fan will only show first 12).

Nice that its doable without a separate effect but I will just add AuroraEffect.

There is no way to create party mode other than writing a new effect from scratch. There is SoundEffect but that works as a simple audio spectrum visualizer (each led responds to a specific audio frequency).

The time schedule cant cross days, but you can specify time from 7am to 8pm and invert it:
```json
{
    "Type": "ScheduleTrigger",
    "Config": {
        "Scope": "Day",
        "Schedule": [
            "07:00 -> 20:00"
        ],
        "Value": false
    }
}
```

So it will return false (and the next effect will be enabled) when time is between 07:00 and 20:00, otherwise returns true (and enable current effect).

MoshiMoshi0, 2020-04-25:
Pushed a build with AuroraEffect, you can download it [here](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/32434367/artifacts).

Defaults:
```json
        {
          "Type": "AuroraEffect",
          "Config": {
            "TickInterval": 1,
            "Length": 64,
            "ColorGenerationMethod": "PerPort",
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

`TickInterval` changes a frame delay between advancing the gradient, values from 0 and up.
`Length` length/precision of the gradient, higher values means smoother colors

You can also try changing `ColorGenerationMethod` to `"SpanPorts"`, it will extend the effect to all of the fans. You might need to reorder them in the profile to get correct transitions from one fan to another.

On the develop branch all effects were updated to `LedColorProvider` to streamline color configurations so your old config might not work.
Using `StaticColorEffect` as example:
```json
{
    "Type": "StaticColorEffect",
    "Config": {
        "Color": {
            "Full": [255, 0, 0],
            "PerLed": [
                [255, 0, 0],
                [0, 255, 0],
                [0, 0, 255],
                ...
            ],
            "Gradient": [
                [0.0, [255, 0, 0]],
                [0.5, [0, 255, 0]],
                [1.0, [0, 0, 255]]
            ]
        },
        "Trigger": {
            "Type": "AlwaysTrigger"
        }
    }
},
```
Only one of `"Full"`, `"PerLed"`, `"Gradient"` can be used. 
This allows to only have one color config property that can be used as Full, PerLed or Gradient type where before you had to had each as a separate property.

Luishi5000, 2020-04-27:
Thank you for the quick reply and development of the Aurora effect! I have tried it and it works very well. The SpanPorts feature is fantastic. Your support is much appreciated on this.

I have a couple thoughts on it to make it more accurate to the default effect.

For reference, I took a recording of the above effect:
https://youtu.be/wHy3KloLvPQ

Here is the default TT RGB Plus software Aurora:
https://youtu.be/MaXGlprFkWg

(Hopefully it is visible... it's crazy how easily LEDs blow out the sensor to capture color well)

The main difference with the default effect is that it wraps in both directions instead of 1, creating a smoother gradient. There particularly jarring transition is from the purple to red, then yellow. Wrapping in both directions would help alleviate that. That specific color transition is actually jarring in the default software's Aurora effect too, the red and yellow is extremely short lived. (It would be wonderful to pass in custom colors)

Also I tried a TickInterval of less than "1" and it crashed the app. Ideally it would be nice to have it a good deal slower, perhaps even the ability to make it 10 times slower.

- Semi unrelated, I feel like this software is very helpful to lots of people, especially those who cannot work with the default software, yet people seem to get hung up on the setup. Is it okay with you if I make a basic setup video guide for YouTube?

Thanks again!

MoshiMoshi0, 2020-04-27:
> The main difference with the default effect is that it wraps in both directions instead of 1, creating a smoother gradient. There particularly jarring transition is from the purple to red, then yellow. Wrapping in both directions would help alleviate that.

Sure I'll add that.

> That specific color transition is actually jarring in the default software's Aurora effect too, the red and yellow is extremely short lived. (It would be wonderful to pass in custom colors).

You can. 
The config for aurora effect accepts a custom [led color gradient](https://moshimoshi0.github.io/TTController/common/led-color-gradient). If its not configured it generates the default one.
Location is from 0-1.

```json
        {
          "Type": "AuroraEffect",
          "Config": {
            "Gradient": [
                [0.0, [255, 0, 0]],
                [0.5, [0, 255, 0]],
                [1.0, [0, 0, 255]]
            ],
            ...
          }
        }
```

I think its jarring because the leds have a problem with red colors, the default generated rainbow is a perfectly smooth loop.

> Also I tried a TickInterval of less than "1" and it crashed the app.

How did it crash? Do you have logs? It should be fine to set it to 0.

>  Ideally it would be nice to have it a good deal slower, perhaps even the ability to make it 10 times slower.

You can make it slower by setting `TickInterval` higher but that will also make it choppy, or by setting `Length` to a higher number like 100-200.

MoshiMoshi0, 2020-04-27:
Pushed some updates. Develop build [here](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/32469973/artifacts).

Sample config:
```json
        {
          "Type": "AuroraEffect",
          "Config": {
            "Step": 0.003,
            "Length": 36,
            "Mirror": true,
            "Brightness": 0.5,
            "Saturation": 0.75,
            "ColorGenerationMethod": "SpanPorts",
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

`Step` now allows for fine tuning of the speed. 
Added global `Brightness` and `Saturation` for colors. 
`Mirror` will mirror the colors on both sides. 
Tweaked color generation so now red should be a little less jarring.

Luishi5000, 2020-04-27:
Awesome! I've tested it and it works perfectly - absolutely love it. Thanks so much again!

This way I'll be able to pass in a huge custom gradient that will let me control brightness and saturation in the animation as well. I'm trying to make one as lively and full of variation as possible.

`How did it crash? Do you have logs? It should be fine to set it to 0.`
I set tick to 0.5 originally but the step interval will take care of that.

I've run into an issue where the animation suddenly freezes randomly, sometimes it takes 2 min, sometimes it takes an hour to freeze. I believe this could be an issue with my device connection? Since I was experiencing the same issue with the default program.

Then I am unable to stop the service, as it just hangs. Then attempting to uninstall it at this point gives me this error:

[image]

I then have to wait a good amount of time (several minutes) until the connection seems to get re-established. Then I am able to start the service once again.

Does this indicate physical connectivity/ controller issues to you? Should I RMA my controller?

MoshiMoshi0, 2020-04-27:
> I set tick to 0.5 originally but the step interval will take care of that.

Oh, well it can only be set to a whole number so I guess the JSON parser did not like that

> I've run into an issue where the animation suddenly freezes randomly, sometimes it takes 2 min, sometimes it takes an hour to freeze. I believe this could be an issue with my device connection? Since I was experiencing the same issue with the default program.

Can you post log file when that happens? It should be in the Logs folder.
That is a first time this has been reported so most likely its something with your pc or controller.

Luishi5000, 2020-04-27:
`2020-04-27 09:05:18.5903|INFO|TTController.Service.TTService|================================================================
2020-04-27 09:05:18.6088|INFO|TTController.Service.TTService|Initializing...
2020-04-27 09:05:18.8760|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.9.0.0]
2020-04-27 09:05:18.8830|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.9.0.0]
2020-04-27 09:05:18.8940|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LogicTrigger [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.9.0.0]
2020-04-27 09:05:18.9115|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.9.0.0]
2020-04-27 09:05:18.9250|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseTrigger [0.9.0.0]
2020-04-27 09:05:18.9250|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.9.0.0]
2020-04-27 09:05:18.9250|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorTrigger [0.9.0.0]
2020-04-27 09:05:18.9250|INFO|TTController.Service.Managers.ConfigManager|Creating Config Manager...
2020-04-27 09:05:19.1112|INFO|TTController.Service.Managers.ConfigManager|Loading config...
2020-04-27 09:05:19.2783|INFO|TTController.Service.Utils.DataCache|Creating DataCache...
2020-04-27 09:05:19.2783|INFO|TTController.Service.Utils.PluginStore|Creating Plugin Store...
2020-04-27 09:05:19.2783|INFO|TTController.Service.Managers.SensorManager|Creating Sensor Manager...
2020-04-27 09:05:19.2783|INFO|TTController.Service.Hardware.LibreHardwareMonitorFacade|Initializing Libre Hardware Monitor...
2020-04-27 09:05:19.3934|INFO|TTController.Service.Managers.DeviceManager|Creating Device Manager...
2020-04-27 09:05:19.6126|INFO|TTController.Service.Managers.DeviceManager|Initialized "Riing Plus" controller [0x264A, 0x1FA5]
2020-04-27 09:05:19.6126|INFO|TTController.Service.Managers.SensorManager|Enabling sensor: /gpu/0/temperature/0
2020-04-27 09:05:19.6282|INFO|TTController.Service.Utils.PluginStore|Adding plugin "AuroraEffect" [Default]
2020-04-27 09:05:19.6282|INFO|TTController.Service.Utils.PluginStore|Adding plugin "PwmSpeedController" [Default]
[... cut ...]
2020-04-27 09:39:55.8116|INFO|TTController.Service.TTService|================================================================
2020-04-27 09:39:55.8116|INFO|TTController.Service.TTService|Finalizing...
2020-04-27 09:39:55.8116|INFO|TTController.Service.Managers.TimerManager|Disposing Timer Manager...
2020-04-27 09:39:55.8116|INFO|TTController.Service.Managers.TimerManager|Stopping 4 timers...
`

Seems like it's the same as any other log file I have. Considering that it happened in both apps, I suspect it's something with my controller. I'll RMA it and see how a new one does.

MoshiMoshi0, 2020-04-27:
Try editing `TTController.Service.exe.Config` file and changing 
`<logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to
`<logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`

But dont keep it running for TOO long because there will be a lot of stuff in the logs.
Then post/upload when it freezes.

Luishi5000, 2020-04-27:
I was able to capture it when it froze:

`2020-04-27 11:32:08.9698|INFO|TTController.Service.TTService|================================================================
2020-04-27 11:32:08.9888|INFO|TTController.Service.TTService|Initializing...
2020-04-27 11:32:09.2625|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.DpsgController [0.9.0.0]
2020-04-27 11:32:09.2625|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RiingController [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RiingPlusController [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RiingTrioController [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.AuroraEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.BlinkEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.CopyColorEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.FlowEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.PingPongEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.PulseEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RawEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RazerConnectEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.9.0.0]
2020-04-27 11:32:09.2695|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.RippleEffect [0.9.0.0]
2020-04-27 11:32:09.2695|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.9.0.0]
2020-04-27 11:32:09.2845|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.SensorEffect [0.9.0.0]
2020-04-27 11:32:09.2845|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.9.0.0]
2020-04-27 11:32:09.2845|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.SoundEffect [0.9.0.0]
2020-04-27 11:32:09.2845|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.9.0.0]
2020-04-27 11:32:09.2845|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.SpectrumEffect [0.9.0.0]
2020-04-27 11:32:09.2845|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.9.0.0]
2020-04-27 11:32:09.2845|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.StaticColorEffect [0.9.0.0]
2020-04-27 11:32:09.2845|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.9.0.0]
2020-04-27 11:32:09.2996|TRACE|TTController.Service.Utils.PluginLoader|Found plugin assembly for type IPlugin: TTController.Plugin.WaveEffect [0.9.0.0]
[... cut ...]
2020-04-27 11:37:05.1070|DEBUG|TTController.Service.TTService|Sensor "/gpu/0/temperature/0" value: 40.34948
2020-04-27 11:37:10.1060|DEBUG|TTController.Service.TTService|Port [9802, 8101, 1] data: [PortId: 1, Speed: 47, Rpm: 853, Unknown: 255]
2020-04-27 11:37:10.1060|DEBUG|TTController.Service.TTService|Sensor "/gpu/0/temperature/0" value: 40.0473
2020-04-27 11:37:15.1060|DEBUG|TTController.Service.TTService|Port [9802, 8101, 1] data: [PortId: 1, Speed: 47, Rpm: 853, Unknown: 255]
2020-04-27 11:37:15.1060|DEBUG|TTController.Service.TTService|Sensor "/gpu/0/temperature/0" value: 40.41006
`

MoshiMoshi0, 2020-04-27:
Yeah, for some reason it hangs when trying to write or read data. There is a 1000ms timeout so this does not happen but maybe there is a problem with the hidlibrary.

I actually just switched to the newer and more performant HidSharp library so you can try [this ](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/build/artifacts) build. 
HidSharp might also provide better errors so its worth a try. If it also hangs or crashes there might be something wrong with the controller or the connection to the motherboard.

Luishi5000, 2020-04-28:
So far so good on the new library, haven't run into any freezing issues since I got it and have been running all day. Great timing. Thank you!

I'm setting up scheduling, and perhaps I misunderstood the way to configure multiple effects with a schedule?

`{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Gradient": [
                [0.0, [220, 0, 0]],
                [0.1, [215, 5, 0]],
                [0.2, [200, 10, 0]],
                [0.3, [140, 80, 0]],
                [0.4, [20, 255, 30]],
                [0.5, [0, 140, 170]],
                [0.6, [0, 80, 200]],
                [0.7, [55, 60, 160]],
                [0.8, [180, 30, 60]],
                [0.9, [200, 2, 4]],
                [1.00, [220, 0, 1]]

            ],
            "Step": 0.002,
            "Length": 36,
            "Mirror": true,
            "Brightness": 1,
            "Saturation": 1,
            "ColorGenerationMethod": "SpanPorts",
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                  "Scope": "Day",
                  "Schedule": [
                      "07:00 -> 20:00"
                  ],
                  "Value": true
              }
            }
          }
        },
        {
            "Type": "SensorEffect",
            "Config": {
              "ColorGradient": [
                  [47, [255, 0, 0]],
                  [65, [255, 20, 0]],
                  [75, [256, 0, 255]]
              ]
                "Sensors": ["/gpu/0/temperature/0"],
                "Trigger": {
                    "Type": "AlwaysTrigger"
                }
            }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 65,
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 1]],
      "Config": {
        "Name": "Top Fan"
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/gpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
`
I would like the sensor effect to be active when not in the schedule window, yet it produces an error.

Unrelated - is there a way to bind hotkeys to an effect? It would be handy to have a way to turn the lights off with a keystroke.

MoshiMoshi0, 2020-04-28:
I assume it produces an error because you json is not valid. You are missing a comma in SensorEffect config.

You can use https://jsoneditoronline.org to check if your json is valid.

> Unrelated - is there a way to bind hotkeys to an effect? It would be handy to have a way to turn the lights off with a keystroke.

No there is no way to do that for now. It was designed to be configure once and forget type of service.

Luishi5000, 2020-04-28:
Thanks for the site, I'll be sure to use that. I got the scheduling working for my use case. Two things I noticed.

https://moshimoshi0.github.io/TTController/triggers/scheduletrigger

On this page, it has this in the example:

"Schedule": [
    "00:10" -> "00:20",
    "00:30" -> "00:40"
]

Yet above it, it has a conflicting syntax of:

List of schedule entries in "<start time> -> <end time>" format.

The other thing I noticed, setting a start time had no effect in my config. I would assume that my always trigger SensorEffect is on unless it is within the Aurora scheduled window. However, in my testing, the effect only disables after the end time window, but is on before that (regardless of the start time occurring or not). It doesn't matter for my use case but I wanted to point it out in case it is a bug. My now working config:

`{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Gradient": [
                [0.0, [220, 0, 0]],
                [0.1, [215, 5, 0]],
                [0.2, [200, 10, 0]],
                [0.3, [140, 80, 0]],
                [0.4, [20, 255, 30]],
                [0.5, [0, 140, 170]],
                [0.6, [0, 80, 200]],
                [0.7, [55, 60, 160]],
                [0.8, [180, 30, 60]],
                [0.9, [200, 2, 4]],
                [1.00, [220, 0, 1]]

            ],
            "Step": 0.002,
            "Length": 36,
            "Mirror": true,
            "Brightness": 1,
            "Saturation": 1,
            "ColorGenerationMethod": "SpanPorts",
            "Trigger": {
              "Type": "ScheduleTrigger",
              "Config": {
                  "Scope": "Day",
                  "Schedule": [
                      "00:00 -> 20:30"
                  ]
              }
            }
          }
        },
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/gpu/0/temperature/0"],
            "ColorGradient": [
              [47, [20, 0, 0]],
              [60, [125, 0, 0]],
              [75, [255, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Speed": 65,
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 1]],
      "Config": {
        "Name": "Top Fan"
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/gpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
`

MoshiMoshi0, 2020-04-28:
>On this page, it has this in the example:
>
>"Schedule": [
>"00:10" -> "00:20",
>"00:30" -> "00:40"
>]
>
>Yet above it, it has a conflicting syntax of:>
>
>List of schedule entries in " -> " format.

Thanks fixed.

> The other thing I noticed, setting a start time had no effect in my config. I would assume that my always trigger SensorEffect is on unless it is within the Aurora scheduled window. However, in my [...]

When a trigger returns a true value, the effect that it is configured on will be enabled. ScheduleTrigger returns true value when the current time is within the specified range/ranges.

If you want to invert it (return false when withing range), you have to set the "Value" parameter to false. It is described on the docs. I also described this in one of the first replies here.

Luishi5000, 2020-04-28:
Apologies, I mistook a configuration error in my JSON as an issue with the scheduling setup. I have it all sorted now. Thanks again for your help on everything, and thank you for making this software to begin with.

closed 2020-04-28 by Luishi5000