MoshiMoshi0/TTController #43: Help Installing

admawby, 2020-05-05:
**Describe the issue**
First of all, thank you all for working on this! Sorry if this is a dumb question, but i don't see any EXE files to install the service as described in the docs. Am I to understand that i should build it myself? I tried building with visual studio but it failed with a bunch of errors, most of which were citing failure to find files in the Third Party folder. Any guidance would be appreciated!

**Debug report**
```
Replace this line with output of the debug report (Main Menu -> Debug -> Report).
```

**Current configuration**
```json
Replace this line with contents of your current configuration (If configured). 
```

MoshiMoshi0, 2020-05-05:
You downloaded the source code not a released build.
All releases are on the "[releases](https://github.com/MoshiMoshi0/TTController/releases)" tab.

Github makes it pretty confusing.

closed 2020-05-05 by MoshiMoshi0

admawby, 2020-05-05:
*facepalm*

thank you