MoshiMoshi0/TTController #44: Occasional Service Launch Failure on Boot

Luishi5000, 2020-05-06:
About 1 out of 3 times the service does not auto start on boot. It then starts manually no problem. Using 0.9.0-alpha.40.

Here is the log:

```
2020-05-05 20:45:36.5492|INFO|TTController.Service.TTService|================================================================
2020-05-05 20:45:36.5718|INFO|TTController.Service.TTService|Initializing...
2020-05-05 20:45:36.8645|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgController [0.9.0.0]
2020-05-05 20:45:36.8875|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingController [0.9.0.0]
2020-05-05 20:45:36.9055|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingPlusController [0.9.0.0]
2020-05-05 20:45:36.9226|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RiingTrioController [0.9.0.0]
2020-05-05 20:45:36.9396|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AuroraEffect [0.9.0.0]
2020-05-05 20:45:36.9561|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.BlinkEffect [0.9.0.0]
2020-05-05 20:45:36.9721|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.9.0.0]
2020-05-05 20:45:36.9886|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.FlowEffect [0.9.0.0]
2020-05-05 20:45:37.0051|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PingPongEffect [0.9.0.0]
2020-05-05 20:45:37.0216|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseEffect [0.9.0.0]
2020-05-05 20:45:37.0377|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RawEffect [0.9.0.0]
2020-05-05 20:45:37.0567|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.9.0.0]
2020-05-05 20:45:37.0732|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.RippleEffect [0.9.0.0]
2020-05-05 20:45:37.0902|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorEffect [0.9.0.0]
2020-05-05 20:45:37.1422|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SoundEffect [0.9.0.0]
2020-05-05 20:45:37.1588|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.9.0.0]
2020-05-05 20:45:37.1758|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.9.0.0]
2020-05-05 20:45:37.1923|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.WaveEffect [0.9.0.0]
2020-05-05 20:45:37.2098|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.CopySpeedController [0.9.0.0]
2020-05-05 20:45:37.2268|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.9.0.0]
2020-05-05 20:45:37.2699|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.9.0.0]
2020-05-05 20:45:37.2874|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.9.0.0]
2020-05-05 20:45:37.3039|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.9.0.0]
2020-05-05 20:45:37.3209|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.LogicTrigger [0.9.0.0]
2020-05-05 20:45:37.3369|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.9.0.0]
2020-05-05 20:45:37.3539|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.9.0.0]
2020-05-05 20:45:37.3704|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.PulseTrigger [0.9.0.0]
2020-05-05 20:45:37.3880|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.9.0.0]
2020-05-05 20:45:37.4050|INFO|TTController.Service.Utils.PluginLoader|Loading plugin assembly: TTController.Plugin.SensorTrigger [0.9.0.0]
2020-05-05 20:45:37.4120|INFO|TTController.Service.Managers.ConfigManager|Creating Config Manager...
2020-05-05 20:45:37.4720|INFO|TTController.Service.Managers.ConfigManager|Loading config...
2020-05-05 20:45:37.7337|INFO|TTController.Service.Utils.DataCache|Creating DataCache...
2020-05-05 20:45:37.7398|INFO|TTController.Service.Utils.PluginStore|Creating Plugin Store...
2020-05-05 20:45:37.7398|INFO|TTController.Service.Managers.SensorManager|Creating Sensor Manager...
2020-05-05 20:45:37.7398|INFO|TTController.Service.Hardware.LibreHardwareMonitorFacade|Initializing Libre Hardware Monitor...
2020-05-05 20:45:37.8652|INFO|TTController.Service.Managers.DeviceManager|Creating Device Manager...
2020-05-05 20:45:37.9322|INFO|TTController.Service.Managers.DeviceManager|Initialized "Riing Plus" controller [9802, 8101]
2020-05-05 20:45:37.9322|INFO|TTController.Service.Managers.SensorManager|Enabling sensor: /gpu/0/temperature/0
2020-05-05 20:45:37.9422|INFO|TTController.Service.Utils.PluginStore|Adding plugin "SensorEffect" [Default]
2020-05-05 20:45:37.9422|INFO|TTController.Service.Utils.PluginStore|Adding plugin "SoundEffect" [Default]
2020-05-05 20:45:37.9422|INFO|TTController.Service.Utils.PluginStore|Adding plugin "SensorEffect" [Default]
2020-05-05 20:45:37.9422|INFO|TTController.Service.Utils.PluginStore|Adding plugin "AuroraEffect" [Default]
2020-05-05 20:45:37.9422|INFO|TTController.Service.Utils.PluginStore|Adding plugin "SensorEffect" [Default]
2020-05-05 20:45:37.9567|INFO|TTController.Service.Utils.PluginStore|Adding plugin "PwmSpeedController" [Default]
2020-05-05 20:45:37.9723|INFO|TTController.Service.Managers.TimerManager|Creating Timer Manager...
2020-05-05 20:45:37.9723|INFO|TTController.Service.Managers.TimerManager|Starting 4 timers...
2020-05-05 20:45:37.9723|INFO|TTController.Service.TTService|Initializing done!
2020-05-05 20:45:37.9723|INFO|TTController.Service.TTService|================================================================
2020-05-05 21:16:09.4569|INFO|TTController.Service.TTService|================================================================
2020-05-05 21:16:09.4569|INFO|TTController.Service.TTService|Finalizing...
2020-05-05 21:16:09.4569|INFO|TTController.Service.Managers.TimerManager|Disposing Timer Manager...
2020-05-05 21:16:09.4569|INFO|TTController.Service.Managers.TimerManager|Stopping 4 timers...
2020-05-05 21:16:09.4804|INFO|TTController.Service.TTService|Applying computer state profile: Suspend
2020-05-05 21:16:09.4804|INFO|TTController.Service.Managers.SensorManager|Disposing Sensor Manager...
2020-05-05 21:16:09.4874|INFO|TTController.Service.Hardware.LibreHardwareMonitorFacade|Finalizing Libre Hardware Monitor...
2020-05-05 21:16:09.5299|INFO|TTController.Service.Managers.DeviceManager|Disposing Device Manager...
2020-05-05 21:16:09.5344|INFO|TTController.Service.Managers.ConfigManager|Disposing Config Manager...
2020-05-05 21:16:09.5344|INFO|TTController.Service.Utils.PluginStore|Disposing Plugin Store...
2020-05-05 21:16:09.5344|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "SensorEffect" [Default]
2020-05-05 21:16:09.6308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "SoundEffect" [Default]
2020-05-05 21:16:09.6308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "SensorEffect" [Default]
2020-05-05 21:16:09.6308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "AuroraEffect" [Default]
2020-05-05 21:16:09.6308|INFO|TTController.Service.Utils.PluginStore|Disposing plugin "SensorEffect" [Default]
[... 313 lines cut ...]
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 1]],
      "Config": {
        "Name": "Top Fan"
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/gpu/0/temperature/0"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}

```

MoshiMoshi0, 2020-05-06:
I guess the service loads faster than it can access audio capture.
I'll see if I can delay initialization.

Thanks for reporting.

MoshiMoshi0, 2020-05-07:
Try [this](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/32722754/artifacts) build.

It will constantly try to initialize if it fails it will just log the error. Not the best method but I could not find anything in cscore library that would be nicer.

MoshiMoshi0, 2020-05-12:
Did you manage to test the above build? I would like to know if it helps.

Luishi5000, 2020-05-13:
Tried it over the last few restarts - so far so good! Thanks for the new build, so nice not to have to start the service manually.

closed 2020-05-13 by Luishi5000