MoshiMoshi0/TTController #45: RazerConnectEffect only half of the ring is illuminated

raiven60, 2020-05-09:
**Describe the issue**
When RazerConnectEffect is active only half of the ring is illuminated.

**Debug report**
```
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 935, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 931, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 964, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 929, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 938, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 65, Rpm: 947, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6776F:
»        Temperature:
»                CPU Core (/lpc/nct6776f/temperature/0): 35,0 °C
»                Temperature #2 (/lpc/nct6776f/temperature/2): 0,5 °C
»                Temperature #3 (/lpc/nct6776f/temperature/3): 44,0 °C
»
»  Intel Core i7-4930K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 33,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 33,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 32,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 31,0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 33,0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 33,0 °C
»                CPU Package (/intelcpu/0/temperature/6): 33,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 57,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 57,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 58,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 59,0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 57,0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 57,0 °C
»                Core Max (/intelcpu/0/temperature/13): 33,0 °C
»                Core Average (/intelcpu/0/temperature/14): 32,5 °C
»
»  NVIDIA GeForce RTX 2070 SUPER:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 55,0 °C
»
»  ST2000DM001-1ER164:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 33,0 °C
»
»  Samsung SSD 850 EVO 500GB:
»        Temperature:
»                Temperature (/ssd/1/temperature/0): 38,0 °C
»
»  Samsung SSD 860 EVO 1TB:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 35,0 °C
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/13"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [                              
        {
          "Type": "RazerConnectEffect",
          "Config": {            
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3]
      ],
      "Speed": 65,
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8101, 1],
				[9802, 8101, 2],
				[9802, 8101, 3],
				[9802, 8101, 4]],
      "Config": {
        "Name": "Case Fan"
      }
    },
	{
      "Ports": [[9802, 8102, 1],
				[9802, 8102, 2],
				[9802, 8102, 3]],
      "Config": {
        "Name": "CPU Fan"
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/intelcpu/0/temperature/13"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
![IMG_20200509_135936-min](https://user-images.githubusercontent.com/34771698/81473373-47c47700-91fe-11ea-8d88-35f6057c17ec.jpg)
![IMG_20200509_135946-min](https://user-images.githubusercontent.com/34771698/81473375-48f5a400-91fe-11ea-878f-d298cd546d94.jpg)
![IMG_20200509_135921-min](https://user-images.githubusercontent.com/34771698/81473377-4abf6780-91fe-11ea-8848-ddf31d2a7fb3.jpg)

```

MoshiMoshi0, 2020-05-09:
You need to use [LedCountHandling](https://moshimoshi0.github.io/TTController/config/port#ledcounthandling) to stretch the colors because razer effect only outputs 4/5. 
So for example add `"LedCountHandling": "Lerp",` to both of your port configs. You can also use `"Nearest"` or `"Copy"`.

raiven60, 2020-05-09:
Wow! Thank you for your quick answer. It worked like a charm! 
Your work is amazing, thanks for getting rid of the bad TT software.

closed 2020-05-09 by raiven60