MoshiMoshi0/TTController #46: API or other control without static configuration file

DimOriN-dev, 2020-05-20:
Hello.
Nice tool! Good job. 
I would like to control fans from the PC -game - S.T.A.L.K.E.R. . For example, to display the level of health or radiation infection through the color of a certain fan. In the presence of a LUA-script with the ability to call any CMD through the CLI. Or I'll try to make any other handler that you can provide.
P.S. Google translate =)

MoshiMoshi0, 2020-05-20:
So you want the ability to set colors/speed from an external program?
Or do you want me to add a stalker specific color effect?

DimOriN-dev, 2020-05-20:
Stalker have LUA based script language. I can run via LUA (ingame) external program like "TTCtrl.exe -fan_id 1 -set_color 255,255,255". :
function on_health_change(actor.health) 
...
if actor.health<=0.1 then luarun("%appdata%\TTCtrl.exe -fan_id 1 -set_color 255,0,0")
...
end

MoshiMoshi0, 2020-05-20:
I think this is out of scope of this project.
I could add some type of IPC plugin that would allow you to send colors or speeds from external program, but if you want a cli executable then im affraid you have to create your own. You can use [this](https://moshimoshi0.github.io/ttrgbplusapi/index.html) api for reference.

DimOriN-dev, 2020-05-20:
Ok. Thank you. Tool is really powerfull )) GPU-sensor is awesome!

MoshiMoshi0, 2020-09-20:
Took a while but I merged initial implementation of IPC communication using websockets ([build](https://ci.appveyor.com/project/MoshiMoshi0/ttcontroller/builds/35303393/artifacts)).
Included basic plugins: IpcTrigger, IpcEffect, IpcSpeedController and ability to receive data from the service.

This should be documented on next release.

closed 2020-09-20 by MoshiMoshi0