MoshiMoshi0/TTController #48: Configuration Help

hpfuhl88, 2020-05-22:
I am trying to setup my configuration file to run using the flow normal LED function with a fan curve based off of /gpu/0/temperature/1.  The config file is using the sensor method only because I was attempting to get the config to work correctly based off of the existing samples you have.  My goal is to have the fans under this config marked as Bottom Fans.

I did have a question though I am going to be installing 2 200mm Ring Trios in a few days once they arrive.  Am I able to have separate configs coming from one controller using the different Port IDs or do I need to install at least 1 other controller to separate functionality.  For instance could I use them in Port 4 and 5 below instead of installing a whole new controller.

The primary issue I am having is a timeout error that is occurring when trying to have TT Controller initialize.  I do have a copy of the log that I have attached below my Configuration file

**Hardware info**
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 30, Rpm: 578, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 30, Rpm: 566, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 30, Rpm: 567, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6795D:
»        Temperature:
»                Temperature #1 (/lpc/nct6795d/temperature/1): 28.0 °C
»                Temperature #2 (/lpc/nct6795d/temperature/2): 29.0 °C
»                Temperature #3 (/lpc/nct6795d/temperature/3): 33.5 °C
»                Temperature #5 (/lpc/nct6795d/temperature/5): 37.0 °C
»                Temperature #6 (/lpc/nct6795d/temperature/6): -2.0 °C
»
»  AMD Ryzen 7 3800X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 33.4 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 35.5 °C
»
»  AMD Radeon RX 5700 XT:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 32.0 °C
»                GPU Hot Spot (/gpu/0/temperature/6): 32.0 °C
»                GPU VR VDDC (/gpu/0/temperature/2): 31.0 °C
»                GPU Memory (/gpu/0/temperature/1): 36.0 °C
»
»  Samsung SSD 970 EVO Plus 250GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 41.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 41.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 55.0 °C
»
»  ST2000LX001-1RG174:
»        Temperature:
[... 120 lines cut ...]
   at Newtonsoft.Json.Utilities.ConvertUtils.ConvertOrCast(Object initialValue, CultureInfo culture, Type targetType)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.EnsureType(JsonReader reader, Object value, CultureInfo culture, JsonContract contract, Type targetType)
   --- End of inner exception stack trace ---
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.EnsureType(JsonReader reader, Object value, CultureInfo culture, JsonContract contract, Type targetType)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList list, JsonReader reader, JsonArrayContract contract, JsonProperty containerProperty, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, Object existingValue, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty property, JsonConverter propertyConverter, JsonContainerContract containerContract, JsonProperty containerProperty, JsonReader reader, Object target)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
2020-05-22 11:56:35.9805|FATAL|TTController.Service.Managers.ConfigManager|Failed to load the config!
2020-05-22 11:56:35.9805|FATAL|TTController.Service.TTService|System.Exception: Service failed to start!
   at TTController.Service.TTService.OnStart(String[] args)
2020-05-22 11:56:35.9970|INFO|TTController.Service.TTService|================================================================
2020-05-22 11:56:35.9970|INFO|TTController.Service.TTService|Finalizing...
2020-05-22 11:56:35.9970|INFO|TTController.Service.Managers.ConfigManager|Disposing Config Manager...
2020-05-22 11:56:35.9970|INFO|TTController.Service.TTService|Finalizing done!
2020-05-22 11:56:36.0115|INFO|TTController.Service.TTService|================================================================

MoshiMoshi0, 2020-05-22:
> Am I able to have separate configs coming from one controller using the different Port IDs or do I need to install at least 1 other controller to separate functionality. 

You need to use a separate controller because Riing Trio only works on Riing Trio controller. You cant connect it to older controllers.

This should be the fixed config:

```json
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/gpu/0/temperature/1"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "FlowEffect",
          "Config": {
            "FillStep": 0.05,
            "HueStep": 30,
            "Saturation": 1.0,
            "Brightness": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
      	[9802, 8101, 1],
      	[9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 85
      }
    }
  ]
}
```

hpfuhl88, 2020-05-22:
Ok so I need to hookup the second controller and what alterations to the
config will I have to make to get it to takeover the controls for the ring
trio fans?

MoshiMoshi0, 2020-05-23:
You will have to add correct port identifiers to the profile and also add a new [port config](https://moshimoshi0.github.io/TTController/config/port) that specifies `"RiingTrio"` as the [device type](https://moshimoshi0.github.io/TTController/config/port#devicetype) for those port identifiers.

hpfuhl88, 2020-05-26:
Ok I tried to do what you described above and I just want to make sure that
I am inputting it correctly into the config file.   I have included a
pasted copy of what I have done to add the Riing Trios so far.  Thanks for
your help.

{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 40],
              [50, 50],
              [70, 60],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [50, 45],
              [60, 50],
              [80, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1.0,
            "Brightness": 1.0,
            "HueStep": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
[... 126 lines cut ...]
»                Temperature #6 (/lpc/nct6795d/temperature/6): -2.0 °C
»
»  AMD Ryzen 7 3800X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 36.0 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 40.8 °C
»
»  AMD Radeon RX 5700 XT:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 29.0 °C
»                GPU Hot Spot (/gpu/0/temperature/6): 29.0 °C
»                GPU VR VDDC (/gpu/0/temperature/2): 27.0 °C
»                GPU Memory (/gpu/0/temperature/1): 32.0 °C
»
»  Samsung SSD 970 EVO Plus 250GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 43.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 43.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 45.0 °C
»
»  ST2000LX001-1RG174:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 24.0 °C

Again thanks for your help.

MoshiMoshi0, 2020-05-26:
The config was missing some commas but it seems ok.

```json
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 40],
              [50, 50],
              [70, 60],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [50, 45],
              [60, 50],
              [80, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SpectrumEffect",
          "Config": {
            "Saturation": 1.0,
            "Brightness": 1.0,
            "HueStep": 1.0,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [true]
      }
    },
    {
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedCountHandling": "Lerp",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 90
      }
    }
  ]
}
```

hpfuhl88, 2020-05-26:
So when I tried to run it it only ran the control over the RiingTrios and
not the Plus's.  I thought maybe the Plus's were missing their devicetype
identifier was the issue, but then it failed to run either function.  I
adjusted what my fans are monitoring for speed control on the RiingTrios
and believe I got that all right and set, but I still can't get both fans
to correspond correctly to one another.  I'm not sure what else to do.
I've tried adjusting different things and I either get a timeout error or
only one set of fans to follow the commands.  I've attached the program
line below.

{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 40],
              [50, 50],
              [70, 60],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "CopySpeedController",
          "Config": {
            "Target": [9802, 8101, 1]
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "TickInterval": 2,
            "PerLed": [
                [255, 0, 0],
                [0, 255, 0],
                [0, 0, 255]
             ]
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [true]
      }
    },
    {
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedCountHandling": "Lerp",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 90
      }
    }
  ]
}

MoshiMoshi0, 2020-05-28:
Your configuration has small errors. You can use [this](https://jsoneditoronline.org) to check if the JSON is valid or not.
Also each effect and speed controller HAS to have a trigger in its config.

```json
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 40],
              [50, 50],
              [70, 60],
              [80, 100]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "CopySpeedController",
          "Config": {
            "Target": [9802, 8101, 1],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "TickInterval": 2,
            "PerLed": [
                [255, 0, 0],
                [0, 255, 0],
                [0, 0, 255]
             ],
             "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [true]
      }
    },
    {
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedCountHandling": "Lerp",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 90
      }
    }
  ]
}
```

MoshiMoshi0, 2020-06-01:
Is everything working? I would like to close this issue.

hpfuhl88, 2020-06-01:
Sorry yep got it all worked out thanks for that JSON file site.  It
actually helps me with some other stuff I was working on too.

closed 2020-06-01 by MoshiMoshi0