MoshiMoshi0/TTController #49: "Debug" always greyed out?

Rookie0ne, 2020-05-24:
I'm using a Thermaltake Battlestation 20 desk which has a LED controller built in that's AFAIK identical to the Riing Plus controller.
I installed TTController 0.10.0 alpha and I'm able to manage the service (install successfully, start/stop/restart) but I can't select "Debug" to get information about my connected controller.
Using TT iTake software I can control the LEDs without a problem.

Any idea what I'm doing wrong? I understand the default config file is empty but I thought I can get the necessary controller information from the debug mode...

MoshiMoshi0, 2020-05-24:
It is greyed out because the service must be stopped for the debug menu to work.

Rookie0ne, 2020-05-24:
Thanks for the quick response and yes, I should have realized that myself x-)

But it looks like TTController can't find my controller although it is visible as "USB Input Device" in Windows Device Manager and TT iTAKE software can find it and use it.

Any idea what could be the cause of this?

MoshiMoshi0, 2020-05-24:
You cant be running both TTController and any other thermaltake software, so make sure only TTController is running.

Rookie0ne, 2020-05-24:
Yes, I closed the Thermaltake software before but TTController still can't find the LED controller (see screenshot attached).
[image]

MoshiMoshi0, 2020-05-24:
Oh, I didnt know they have a rgb desk. It will probably use a different controller because it has a lot of leds. So for now I would say that is unsupported.

Can you post output of `Device Manager -> Properties (on the desk usb device) -> Details -> Device Instance Path`

Rookie0ne, 2020-05-24:
Thanks for the super quick answers!

Here is the device manager output:
USB\VID_264A&PID_07D1\8&6DC261&0&1

MoshiMoshi0, 2020-05-24:
Dont think this is using Riing controller since it has a lot of leds.

This will not be supported for now but if you want you can help by doing this:

* Install Wireshark with usbpcap (you will be asked if you want to install usbpcap in the installer)
* Start Wireshark
* Double click on USBPCap1 capture on the main screen
* Start the official software (without the TTController service running)
* After 10s-30s stop the capture in wireshark using the red square button
* Save and upload the dump using File -> Save As

I need to see the data the official software sends to the desk, preferably rainbow colors, to check if this is possible to add to TTController.

Rookie0ne, 2020-05-24:
See attached the Wireshark capture.
[battlestation20desk_capture.zip](https://github.com/MoshiMoshi0/TTController/files/4674580/battlestation20desk_capture.zip)

MoshiMoshi0, 2020-05-25:
Try this build: [TTController_0.10.0.0-alpha.3.a03efa6cfb53fdf2d9b260be07d24f5fbad9fc81.zip](https://github.com/MoshiMoshi0/TTController/files/4676599/TTController_0.10.0.0-alpha.3.a03efa6cfb53fdf2d9b260be07d24f5fbad9fc81.zip)

Sample config, all leds should be red:
```json
{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "StaticColorEffect",
          "Config": {
            "Color": {
              "Full": [255, 0, 0]
            },
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk"
      }
    }
  ]
}
```

Rookie0ne, 2020-05-25:
Thanks for the great support!

It seems to mostly work now:
1) I installed the build you linked.
2) I made sure no Thermaltake software/service is running.
3) Installed the TTController service.
4) Checked in debug that it finds the controller now, see screenshot attached. [image]
5) When I start the service the table indeed turns red (yay!), see photo attached.
[image]
6) But after a few seconds it falls back to the default rainbow animation (boo!), see photo attached.
[image]

Is the issue in 6) just a config issue or something else?

MoshiMoshi0, 2020-05-25:
Thats weird, maybe it does not like to be updated that often.
Stop the service and from main menu select "Run in console". Run it until it goes into rainbow and stop it by pressing any key. Post the output or the log file.

Rookie0ne, 2020-05-25:
hmm it doesn't output anything when it switches from solid back to default animation:
`12:55:39.5882: ================================================================
12:55:39.6242: Initializing...
12:55:39.9283: Loading plugin assembly: TTController.Plugin.DpsgController [0.10.0.0]
12:55:39.9283: Loading plugin assembly: TTController.Plugin.Level20DeskController [0.10.0.0]
12:55:39.9283: Loading plugin assembly: TTController.Plugin.RiingController [0.10.0.0]
12:55:39.9283: Loading plugin assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.FlowEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.PulseEffect [0.10.0.0]
12:55:39.9463: Loading plugin assembly: TTController.Plugin.RawEffect [0.10.0.0]
12:55:39.9582: Loading plugin assembly: TTController.Plugin.RazerConnectEffect [0.10.0.0]
12:55:39.9582: Loading plugin assembly: TTController.Plugin.RippleEffect [0.10.0.0]
12:55:39.9582: Loading plugin assembly: TTController.Plugin.SensorEffect [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.SoundEffect [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.SpectrumEffect [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.StaticColorEffect [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.WaveEffect [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.CopySpeedController [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.DpsgSpeedController [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.PwmSpeedController [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.StaticSpeedController [0.10.0.0]
12:55:39.9752: Loading plugin assembly: TTController.Plugin.AlwaysTrigger [0.10.0.0]
12:55:39.9912: Loading plugin assembly: TTController.Plugin.LogicTrigger [0.10.0.0]
12:55:39.9912: Loading plugin assembly: TTController.Plugin.OneTimeTrigger [0.10.0.0]
12:55:39.9912: Loading plugin assembly: TTController.Plugin.ProcessTrigger [0.10.0.0]
12:55:39.9912: Loading plugin assembly: TTController.Plugin.PulseTrigger [0.10.0.0]
12:55:39.9912: Loading plugin assembly: TTController.Plugin.ScheduleTrigger [0.10.0.0]
12:55:40.0051: Loading plugin assembly: TTController.Plugin.SensorTrigger [0.10.0.0]
12:55:40.0051: Creating Config Manager...
12:55:40.0790: Loading config...
12:55:40.2276: Creating DataCache...
12:55:40.2276: Creating Plugin Store...
12:55:40.2276: Creating Sensor Manager...
12:55:40.2276: Initializing Libre Hardware Monitor...
12:55:41.1361: Detected 45 sensors
12:55:41.1361: Creating Device Manager...
12:55:41.2149: Searching for "Dpsg" controllers
12:55:41.2757: Searching for "Level 20 Desk" controllers
12:55:41.3246: Initialized "Level 20 Desk" controller [9802, 2001]
12:55:41.3246: Searching for "Riing" controllers
12:55:41.3246: Searching for "Riing Plus" controllers
12:55:41.3246: Searching for "Riing Trio" controllers
12:55:41.3246: Adding plugin "StaticColorEffect" [Desk]
12:55:41.3455: Creating Timer Manager...
12:55:41.3455: Starting 4 timers...
12:55:41.3455: Initializing done!
12:55:41.3495: ================================================================
12:55:50.6787: ================================================================
12:55:50.6787: Finalizing...
12:55:50.6787: Disposing Timer Manager...
12:55:50.6787: Stopping 4 timers...
12:55:50.6928: Applying computer state profile: Shutdown
12:55:50.6928: Disposing Sensor Manager...
12:55:50.6928: Finalizing Libre Hardware Monitor...
12:55:50.7335: Disposing Device Manager...
12:55:50.7335: Disposed devices: 1
12:55:50.7335: Disposing Config Manager...
12:55:50.7335: Disposing Plugin Store...
12:55:50.7335: Disposing plugin "StaticColorEffect" [Desk]
12:55:50.7335: Finalizing done!
12:55:50.7396: ================================================================
Press any key to return to the menu...`

MoshiMoshi0, 2020-05-25:
Ok, open `TTController.Service.exe.Config` file and change:
`      <logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to
`      <logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`

Then attach the log file.

Rookie0ne, 2020-05-25:
> Ok, open `TTController.Service.exe.Config` file and change:
> ` <logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
> to
> ` <logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`
> 
> Then do the same thing and attach the log file.

See attached
[current.log](https://github.com/MoshiMoshi0/TTController/files/4676915/current.log)

MoshiMoshi0, 2020-05-25:
Oh, right. There is an optimization in TTController where if the colors does not change, it wont try to write it again as this is costly.

For now you just need to use other effects that are not static:

```json
{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Length": 1024,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk"
      }
    }
  ]
}
```

Rookie0ne, 2020-05-25:
Ah too bad ... I'm not a big fan of the animation but at least that explains it, thanks :-)

I ran into another issue when I tried to work around the limitation by trying to use the Razer Chroma Link (which works fine with the Thermaltake sofware):
When I'm using the RazerConnect feature TTController crashes with an unhandled exception:

> 14:10:18.9367: Searching for "Level 20 Desk" controllers
> 14:10:18.9808: Initialized "Level 20 Desk" controller [9802, 2001]
> 14:10:18.9808: Searching for "Riing" controllers
> 14:10:18.9808: Searching for "Riing Plus" controllers
> 14:10:18.9808: Searching for "Riing Trio" controllers
> 14:10:18.9808: Adding plugin "RazerConnectEffect" [Desk]
> 14:10:19.0016: Creating Timer Manager...
> 14:10:19.0016: Starting 4 timers...
> 14:10:19.0096: Initializing done!
> 14:10:19.0096: ================================================================
> 
> Unhandled Exception: System.NullReferenceException: Object reference not set to an instance of an object.
>    at TTController.Common.Plugin.EffectBase`1.IsEnabled(ICacheProvider cache) in H:\.projects\csharp\TTController\Source\TTController.Common\Plugin\Effect\EffectBase.cs:line 32
>    at TTController.Plugin.RazerConnectEffect.RazerConnectEffect.IsEnabled(ICacheProvider cache) in H:\.projects\csharp\TTController\Plugins\Effects\TTController.Plugin.RazerConnectEffect\RazerConnectEffect.cs:line 20
>    at TTController.Service.TTService.<DeviceRgbTimerCallback>b__20_1(IEffectBase e) in H:\.projects\csharp\TTController\Source\TTController.Service\TTService.cs:line 441
>    at System.Linq.Enumerable.FirstOrDefault[TSource](IEnumerable`1 source, Func`2 predicate)
>    at TTController.Service.TTService.DeviceRgbTimerCallback() in H:\.projects\csharp\TTController\Source\TTController.Service\TTService.cs:line 439
>    at TTController.Service.Managers.TimerManager.<>c__DisplayClass3_0.<RegisterTimer>b__0(Object sender, TimerElapsedEventArgs args) in H:\.projects\csharp\TTController\Source\TTController.Service\Managers\TimerManager.cs:line 37
>    at TTController.Service.Managers.TimerManager.Timer.ExecuteTimer() in H:\.projects\csharp\TTController\Source\TTController.Service\Managers\TimerManager.cs:line 243
>    at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>    at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>    at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
>    at System.Threading.ThreadHelper.ThreadStart()
> 
> C:\Program Files\TTController>                                                             

Here is my config:
{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk"
      }
    }
  ]
}

MoshiMoshi0, 2020-05-25:
You always need to add some type of trigger:

```json
      {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
      }
```

Rookie0ne, 2020-05-25:
Ah ok, I'm just stupid :-D
Thanks for all the help. I have another issue with the RazerConnect feature (only seems to work for the first LED instead of the whole table) but i'll open a separate issue for that to not confuse things :-)

Would be lovely to see static colors work at some point!
Thanks for the great support!

MoshiMoshi0, 2020-05-25:
Try this build: [TTController_0.10.0.0-alpha.6.78de8322289c75658a5445f07b31602b8698234e.zip](https://github.com/MoshiMoshi0/TTController/files/4677148/TTController_0.10.0.0-alpha.6.78de8322289c75658a5445f07b31602b8698234e.zip)

Added a way to disable cache optimization:
```json
"PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk",
        "IgnoreColorCache": true
      }
    }
  ]
```

So now the static color should work fine, but it will update it 30 times per second.

Only the first 4 leds light up because razer effect only provides 4 colors. You need to stretch them to match. See [this](https://github.com/MoshiMoshi0/TTController/issues/45#issuecomment-626166703)

Rookie0ne, 2020-05-25:
> Try this build: [...]
Nope, still doesnt' work... falls back to default rainbow animation after a few seconds again.

Rookie0ne, 2020-05-25:
> Only the first 4 leds light up because razer effect only provides 4 colors. You need to stretch them to match. See this

Thank you! This worked.
Same issue with static colors (falling back to Rainbow animation) if I configure them in Razer Chroma/Synapse by the way but I guess this is expected :-)

Rookie0ne, 2020-05-25:
AAAAH it works!
I'm an idiot and didn't save the config with '"IgnoreColorCache": true' in... x-)

Thanks so much, this works perfect now!

MoshiMoshi0, 2020-05-25:
Nice 👍 

Thanks for help with implementing this.

closed 2020-05-25 by MoshiMoshi0