MoshiMoshi0/TTController #50: Battlestation 20 RGB desk: RazerConnectEffect only works for first LED segment

Rookie0ne, 2020-05-25:
**Describe the issue**
When I use the RazerConnectEffect to sync up the LEDs in my Battlestation 20 RGB desk it generally works and shows the colors/effects as configured in the Razer Synapse software but only for the first LED segment, the rest is disabled/blank:
[image]

**Debug report**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha.3+Branch.develop.Sha.a03efa6cfb53fdf2d9b260be07d24f5fbad9fc81
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Level 20 Desk
»        VendorId: 9802
»        ProductId: 2001
»        Ports:
»                0:
»                        Data:
»                        Identifier: [9802, 2001]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Intel Core i7-8750H:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 53.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 51.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 61.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 53.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 51.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 54.0 °C
»                CPU Package (/intelcpu/0/temperature/6): 61.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 47.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 49.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 39.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 47.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 49.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 46.0 °C
»                Core Max (/intelcpu/0/temperature/13): 61.0 °C
»                Core Average (/intelcpu/0/temperature/14): 53.8 °C
»
»  NVIDIA:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 0.0 °C
»
»  :
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 0.0 °C
»
-------------------------------
Press any key to continue...
```

**Current configuration**
```json
{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
			}
		  }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk"
      }
    }
  ]
} 
```

Rookie0ne, 2020-05-25:
Note: This works just fine if I let the Thermaltake iTake software synchronize with Razer Chroma.

MoshiMoshi0, 2020-05-25:
You need to use LedCountHandling to stretch the colors because razer effect only outputs 4.
Add `"LedCountHandling": "Lerp"` to your port config. You can also use `"Nearest"` or `"Copy"`.

closed 2020-05-25 by MoshiMoshi0