MoshiMoshi0/TTController #51: Configuration Help: ComputerStateProfile

Rookie0ne, 2020-05-25:
**Describe your desired configuration**
One thing that annoys me with the Battlestation 20 RGB desk LEDs is that when the PC the LED controller is plugged in to (via USB) is powered down or in sleep mode the LEDs always play their default rainbow animation. The moment the PC is back running and TTController (or TT iTAKE software) starts the desired config is loaded. 
I'm wondering if the ComputerStateProfile config in TTController can help with that?
Can I set "StateType": "Shutdown" or "Suspend" and configure the LEDs off for when the PC/service is show down/suspended? 

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha.6+Branch.develop.Sha.78de8322289c75658a5445f07b31602b8698234e
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Level 20 Desk
»        VendorId: 9802
»        ProductId: 2001
»        Ports:
»                0:
»                        Data:
»                        Identifier: [9802, 2001]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Intel Core i7-8750H:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 59.0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 58.0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 60.0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 56.0 °C
»                CPU Core #5 (/intelcpu/0/temperature/4): 52.0 °C
»                CPU Core #6 (/intelcpu/0/temperature/5): 53.0 °C
»                CPU Package (/intelcpu/0/temperature/6): 60.0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/7): 41.0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/8): 42.0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/9): 40.0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/10): 44.0 °C
»                CPU Core #5 Distance to TjMax (/intelcpu/0/temperature/11): 48.0 °C
»                CPU Core #6 Distance to TjMax (/intelcpu/0/temperature/12): 47.0 °C
»                Core Max (/intelcpu/0/temperature/13): 60.0 °C
»                Core Average (/intelcpu/0/temperature/14): 56.3 °C
»
»  NVIDIA:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 0.0 °C
»
»  :
»        Temperature:
»                Temperature (/nvme/0/temperature/0): 0.0 °C
»
-------------------------------
Press any key to continue...
```

**Current configuration**
```{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
			}
		  }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk",
		"LedCountHandling": "Lerp",
		"IgnoreColorCache": true
      }
    }
  ]
} 
```

MoshiMoshi0, 2020-05-25:
Since it seems that the desk controller automatically goes to the rainbow mode when it does not receive new colors, Shutdown profile will only work for a couple of seconds.

I also dont know if there is a way to make Boot profile work. It works by overriding the default effect on the controller. I will try to check if there is support for saving profiles on the desk controller.

If the desk is usb powered then you can try configuring your motherboard bios to completely disable usb power when pc is off or in sleep.

Rookie0ne, 2020-05-25:
Yeah, disabling USB completely unfortunately isn't easily possible because the desk is plugged in at an always powered docking station where USB port is unfortunately always powered.

If I can somehow help with the boot profile validation let me know.