MoshiMoshi0/TTController #52: Level 20 desk: If desk LEDs turned on after service has started it needs a service restart

Rookie0ne, 2020-06-20:
Hardware:
Thermaltake Level 20 Battlestation desk

Issue Description:
I have TTcontroller set up with a RazerConnectEffect
The desk has a switch that turns the LEDs on or off. If its being switched on after TTcontroller service has already started  TTcontroller won't detect it and the desk LEDs are just in their default animation.
Just restarting the TTcontroller service fixes it, the desk LEDs immediately switch to the color-scheme configured in Razer Chroma.

Config:
```
{
  "Profiles": [
    {
      "Name": "Desk",
      "Ports": [
        [9802, 2001]
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
			}
		  }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 2001]],
      "Config": {
        "DeviceType": "Level20Desk",
		"LedCountHandling": "Lerp",
		"IgnoreColorCache": true
      }
    }
  ]
} 
```

MoshiMoshi0, 2020-06-20:
The controllers are only detected when the service starts because TTController was designed mainly for fans and I did not think hot plug support would be needed.
I also dont know what will happen if you turn off the controller while the service is running.
Technically this could be solved with #51 by overriding the default colors with black/off and having the leds on all the time.

So i'll change this issue to a feature request.

MoshiMoshi0, 2020-07-03:
Please try this build: 
[TTController_0.10.0.0-alpha.6.78de8322289c75658a5445f07b31602b8698234e.zip](https://github.com/MoshiMoshi0/TTController/files/4870422/TTController_0.10.0.0-alpha.6.78de8322289c75658a5445f07b31602b8698234e.zip)

I don't know how it will behave since I cant really test plugging/unplugging my controller, but you should see that each time you turn off/on your desk it tries to scan for new controllers.
Try running in console mode first.

Rookie0ne, 2020-07-04:
Hey MoshiMoshi0 , thanks this works!
Just tried two times and the desk immediately switches to the right theme/color despite when turned on while the service is already running.

One minor feedback:
There seems to be a problem with starting the service via TTControllers main menu in this build.
I had stopped the TTController service via Windows Settings > Services.
When I start TTContoller.Service.exe on the command prompt (with Admin rights) and select "[1] Manage Service" -> "Start Service" it sits there forever and then times out with this error message:

`Starting the service...
Failed to start the service...
System.ServiceProcess.TimeoutException: Time out has expired and the operation has not been completed.
   at System.ServiceProcess.ServiceController.WaitForStatus(ServiceControllerStatus desiredStatus, TimeSpan timeout)
   at TTController.Service.Program.<ManageServiceMenu>g__StartService|6_1() in H:\.projects\csharp\TTController\Source\TTController.Service\Program.cs:line 79
Press any key to return to the menu...`

Starting the service via Windows settings > Services works fine, so not a big deal.

MoshiMoshi0, 2020-07-04:
Thats weird since there were no other changes in that build.
Could you attach the latest log file when long startup happens? They are created in 'Logs' folder.

Rookie0ne, 2020-07-04:
Weird indeed because no i can't reproduce it anymore :)
Might have been a temporary hickup in my system/configuration/Windows... tried several times now to reproduce the error from earlier but everything works fine now and services starts/stops as expected.

Thanks for the great work!

closed 2020-07-04 by Rookie0ne