MoshiMoshi0/TTController #53: Configuration Help

gitarian, 2020-06-27:
**Describe your desired configuration**
I have 3 RiingTrio fans and want to configurate them individiually, that is separate colors, speed fan curves etc.

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 20, Rpm: 492, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 20, Rpm: 509, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 20, Rpm: 538, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  ITE IT8686E:
»        Temperature:
»                System #1 (/lpc/it8686e/temperature/0): 46,0 °C
»                Chipset (/lpc/it8686e/temperature/1): 48,0 °C
»                CPU (/lpc/it8686e/temperature/2): 52,0 °C
»                PCIe x16 (/lpc/it8686e/temperature/3): 44,0 °C
»                VRM (/lpc/it8686e/temperature/4): 53,0 °C
»
»  ITE IT8792E/IT8795E:
»        Temperature:
»                PCIe x8 (/lpc/it879xe/temperature/0): 43,0 °C
»                System #2 (/lpc/it879xe/temperature/2): 41,0 °C
»
»  AMD Ryzen 5 2600X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 51,9 °C
»
»  Radeon RX Vega:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 41,0 °C
»                GPU Memory (/gpu/0/temperature/1): 42,0 °C
»                GPU VR VDDC (/gpu/0/temperature/2): 42,0 °C
»                GPU VR MVDD (/gpu/0/temperature/3): 45,0 °C
»                GPU Hot Spot (/gpu/0/temperature/6): 43,0 °C
»
»  Samsung SSD 970 EVO 1TB:
»        Temperature:
»                Temperature (/nvme/4/temperature/0): 49,0 °C
»                Temperature 1 (/nvme/4/temperature/6): 49,0 °C
»                Temperature 2 (/nvme/4/temperature/7): 54,0 °C
»
»  Samsung SSD 860 EVO 1TB:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 35,0 °C
»
»  WDC WD30EZRX-00D8PB0:
»        Temperature:
»                Temperature (/hdd/2/temperature/0): 33,0 °C
»
-------------------------------`
```

**Current configuration**
```{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [0, 0],
              [10, 0],
              [20, 0],
              [30, 0],
              [45, 20],
              [50, 25],
              [55, 30],
              [60, 40],
              [60, 45],
              [65, 50],
              [70, 55],
              [75, 60],
              [80, 70],
              [85, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "SensorEffect",
          "Config": {
            "Sensors": ["/amdcpu/0/temperature/2"],
            "ColorGradient": [
              [40, [0, 0, 255]],
              [55, [00, 255, 0]],
              [75, [255, 0, 0]]
            ],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Suspend",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Speed": 35,
      "EffectType": "Full",
      "EffectColors": [
        [255, 0, 0]
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [[9802, 8501, 1]],
      "Config": {
        "Name": "Front Bottom [1]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      },
      "Ports": [[9802, 8501, 2]],
      "Config": {
        "Name": "Front Top [2]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      },
      "Ports": [[9802, 8501, 3]],
      "Config": {
        "Name": "Back [3]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": ["/amdcpu/0/temperature/2"],
      "Config": {
        "CriticalValue": 90
      }
    }
  ]
}
```

Additionally I've found that the following port configurations are not equivalent though they should as far as I understand the configuration rules

Works
```
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Config": {
        "Name": "Front Bottom [1]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    }
  ]
```

Works, but 2 fans have one outer led ring off and the brigtness is reduced as compared to the 3rd one.
```
  "PortConfigs": [
    {
      "Ports": [[9802, 8501, 1]],
      "Config": {
        "Name": "Front Bottom [1]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      },
      "Ports": [[9802, 8501, 2]],
      "Config": {
        "Name": "Front Top [2]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      },
      "Ports": [[9802, 8501, 3]],
      "Config": {
        "Name": "Back [3]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    }
  ]
```
Shouldn't the second variation not just have the same outcome as the first compact one (but offering possible individual port setups)?

MoshiMoshi0, 2020-06-27:
> I have 3 RiingTrio fans and want to configurate them individiually, that is separate colors, speed fan curves etc.

Just add multiple profiles, one for each fan:
```json
"Profiles": [
    {
      "Name": "Port 1",
      "Ports": [
        [9802, 8501, 1]
      ],
      ...
    },
    {
      "Name": "Port 2",
      "Ports": [
        [9802, 8501, 2]
      ],
      ...
    },
    {
      "Name": "Port 3",
      "Ports": [
        [9802, 8501, 3]
      ],
      ...
    }
],
```

> Additionally I've found that the following port configurations are not equivalent though they should as far as I understand the configuration rules

Yes they are the same but you made a small error and did not enclose each PortConfig in `{}`:
```json
  "PortConfigs": [
    {
      "Ports": [[9802, 8501, 1]],
      "Config": {
        "Name": "Front Bottom [1]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    },
    {
      "Ports": [[9802, 8501, 2]],
      "Config": {
        "Name": "Front Top [2]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    },
    {
      "Ports": [[9802, 8501, 3]],
      "Config": {
        "Name": "Back [3]",
        "DeviceType": "RiingTrio",
        "LedRotation": [11],
        "LedReverse": [false]
      }
    }
  ]
```

Btw I noticed that your suspend profile is wrong, I think it might be because I forgot to update the docs after last update, sorry for that, I'll update asap.
You should now be using [color providers](https://moshimoshi0.github.io/TTController/common/led-color-provider):
```json
  "ComputerStateProfiles": [
    {
      "StateType": "Suspend",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3]
      ],
      "Speed": 35,
      "Color": {
        "Full": [255, 0, 0]
      }
    }
  ],
```

gitarian, 2020-06-27:
Thanks for your quick reply and support - that "urges" me to make an immediate donation. :-)

Regarding the missing curly brackets: since the json parser didn't throw any error, I would have never find that (this is what I don't like about json. One has to be very careful with the various entity separators).

So far, it looks good. Now I have to play with the fan curves. :-)

May I ask another thing. The controller report outputs a _Unknown: 255_ as the last entry of the each controller data set. What is the expected value here?

MoshiMoshi0, 2020-06-27:
> Thanks for your quick reply and support - that "urges" me to make an immediate donation. :-)

Thanks! And no problem :)

> Regarding the missing curly brackets: since the json parser didn't throw any error, I would have never find that (this is what I don't like about json. One has to be very careful with the various entity separators).

True, the plan was to create a separate configurator app that creates the json configs so you wont have to edit them manually.

> May I ask another thing. The controller report outputs a Unknown: 255 as the last entry of the each controller data set. What is the expected value here?

Everything I've seen outputs 255, but I think its temperature value for thermaltake pumps. But not confirmed so I just mark it as Unknown.

gitarian, 2020-06-27:
Thanks.

I'm still totally baffled by the original controller software, what a crap. 👿
I'm happy to have an alternative and support it. 😃 

Have a nice weekend.

closed 2020-06-27 by MoshiMoshi0