MoshiMoshi0/TTController #54: Configuration Help

heyolli, 2020-06-30:
**Describe your desired configuration**
Hey,
i switched from alpha 9 to alpha 10, but sadly i cant use my old config... All the fans are not lighting, did you change something so i have to adopt my config?

Also i wanted to try something like this https://youtu.be/0jXurJMYQYk?t=120 but i cant get the aurora effect to work like the one in the video i've linked

Thanks in advance

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8502
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 0, Rpm: 553, Unknown: 255]
»                        Identifier: [9802, 8502, 1]
»                2:
»                        Data: [PortId: 2, Speed: 0, Rpm: 557, Unknown: 255]
»                        Identifier: [9802, 8502, 2]
»                3:
»                        Data: [PortId: 3, Speed: 0, Rpm: 549, Unknown: 255]
»                        Identifier: [9802, 8502, 3]
»                4:
»                        Data: [PortId: 4, Speed: 35, Rpm: 552, Unknown: 255]
»                        Identifier: [9802, 8502, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 5]
»        Available effect types: PerLed
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8501
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 35, Rpm: 396, Unknown: 255]
»                        Identifier: [9802, 8501, 1]
»                2:
»                        Data: [PortId: 2, Speed: 35, Rpm: 379, Unknown: 255]
»                        Identifier: [9802, 8501, 2]
»                3:
»                        Data: [PortId: 3, Speed: 35, Rpm: 456, Unknown: 255]
»                        Identifier: [9802, 8501, 3]
»                4:
»                        Data: [PortId: 4, Speed: 0, Rpm: 534, Unknown: 255]
»                        Identifier: [9802, 8501, 4]
»                5:
»                        Data: [PortId: 5, Speed: 0, Rpm: 454, Unknown: 255]
»                        Identifier: [9802, 8501, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6793D:
»        Temperature:
»                CPU Core (/lpc/nct6793d/temperature/0): 51,5 °C
»                Temperature #2 (/lpc/nct6793d/temperature/2): 39,0 °C
»                Temperature #3 (/lpc/nct6793d/temperature/3): 12,0 °C
»                Temperature #4 (/lpc/nct6793d/temperature/4): 11,0 °C
»                Temperature #6 (/lpc/nct6793d/temperature/6): -1,0 °C
»
»  Intel Core i7-6700K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 41,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 46,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 37,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 45,0 °C
»                CPU Package (/intelcpu/0/temperature/4): 48,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 59,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 54,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 63,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 55,0 °C
»                Core Max (/intelcpu/0/temperature/9): 46,0 °C
»                Core Average (/intelcpu/0/temperature/10): 42,3 °C
»
»  NVIDIA GeForce RTX 2080:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 47,0 °C
»
»  Samsung SSD 970 EVO 500GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 56,0 °C
»                Temperature 1 (/nvme/1/temperature/6): 56,0 °C
»                Temperature 2 (/nvme/1/temperature/7): 72,0 °C
»
```

**Current configuration**
```{
   "Profiles": [
    {
      "Name": "GPU",
      "Ports": [
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
      
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 0],
              [55, 25],
              [65, 45],
              [70, 60],
              [75, 65]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "Colors": [
                [255, 0, 0],
                [255, 0, 51],
                [255, 0, 102],
                [255, 0, 153],
                [255, 0, 204],
                [255, 0, 255],
                [204, 0, 255],
                [154, 0, 255],
                [104, 0, 255],
                [54, 0, 255],
                [0, 0, 255],
                [0, 46, 255],
                [0, 93, 255],
                [0, 140, 255],
                [0, 187, 255],
                [0, 234, 255],
                [0, 238, 212],
                [0, 242, 170],
                [0, 246, 127],
                [0, 250, 85],
                [0, 255, 43],
                [51, 251, 34],
                [102, 248, 25],
                [153, 244, 17],
                [204, 241, 8],
                [255, 238, 0],
                [255, 199, 0],
                [255, 161, 0],
                [255, 123, 0],
                [255, 85, 0]
            ],
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "CPU",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8502, 4] 
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 15],
              [45, 35],
              [55, 50],
              [65, 55],
              [75, 60]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "WaveEffect",
          "Config": {
            "Colors": [
                [255, 0, 0],
                [255, 0, 51],
                [255, 0, 102],
                [255, 0, 153],
                [255, 0, 204],
                [255, 0, 255],
                [204, 0, 255],
                [154, 0, 255],
                [104, 0, 255],
                [54, 0, 255],
                [0, 0, 255],
                [0, 46, 255],
                [0, 93, 255],
                [0, 140, 255],
                [0, 187, 255],
                [0, 234, 255],
                [0, 238, 212],
                [0, 242, 170],
                [0, 246, 127],
                [0, 250, 85],
                [0, 255, 43],
                [51, 251, 34],
                [102, 248, 25],
                [153, 244, 17],
                [204, 241, 8],
                [255, 238, 0],
                [255, 199, 0],
                [255, 161, 0],
                [255, 123, 0],
                [255, 85, 0]
            ],
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1], 
        [9802, 8501, 2], 
        [9802, 8501, 3], 
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4] 
      ],
      "Config": {
        "DeviceType": "RiingTrio"
      }
    }
  ]
}
```

MoshiMoshi0, 2020-06-30:
Yes all effects now use [color providers](https://moshimoshi0.github.io/TTController/common/led-color-provider)

So:
```json
        {
          "Type": "WaveEffect",
          "Config": {
            "Color": {
                "PerLed": [
                    [255, 0, 0],
                    [255, 0, 51],
                    [255, 0, 102],
                    [255, 0, 153],
                    [255, 0, 204],
                    [255, 0, 255],
                    [204, 0, 255],
                    [154, 0, 255],
                    [104, 0, 255],
                    [54, 0, 255],
                    [0, 0, 255],
                    [0, 46, 255],
                    [0, 93, 255],
                    [0, 140, 255],
                    [0, 187, 255],
                    [0, 234, 255],
                    [0, 238, 212],
                    [0, 242, 170],
                    [0, 246, 127],
                    [0, 250, 85],
                    [0, 255, 43],
                    [51, 251, 34],
                    [102, 248, 25],
                    [153, 244, 17],
                    [204, 241, 8],
                    [255, 238, 0],
                    [255, 199, 0],
                    [255, 161, 0],
                    [255, 123, 0],
                    [255, 85, 0]
                ]
            },
            "TickInterval": 3,
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
```

I dont see anything special in that video so you should be able to tweak the effect to almost match it.
I think what is happening is that the aurora effect does not handle multiple zones at the moment.
I have proper multi zone support planned but I have not figured out a nice way to do that.

heyolli, 2020-06-30:
Thanks for your fast answer!
Made my WaveEffekt work again, thanks!

About the Aurora:
i cant get a grip on how i manage to go "full rainbow" and color change per Fan. (if that makes any sense to you)
In the video you can see the color changes per fan and not the fan-stack. 
Currently only the color of all fans change at the same time, i reckon i messed up somewhere in the config. Took the effekt from an other issue that i found:

{
   "Profiles": [
    {
      "Name": "GPU",
      "Ports": [
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
      
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [50, 0],
              [55, 25],
              [65, 45],
              [70, 60],
              [75, 65]
            ],
            "Sensors": ["/gpu/0/temperature/0"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
       "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Gradient": [
                [0.0, [220, 0, 0]],
                [0.1, [215, 5, 0]],
                [0.2, [200, 10, 0]],
                [0.3, [140, 80, 0]],
                [0.4, [20, 255, 30]],
                [0.5, [0, 140, 170]],
                [0.6, [0, 80, 200]],
                [0.7, [55, 60, 160]],
                [0.8, [180, 30, 60]],
                [0.9, [200, 2, 4]],
                [1.00, [220, 0, 1]]

            ],
            "Step": 0.009,
            "Length": 256,
            "Mirror": false,
            "Brightness": 1,
            "Saturation": 1,

            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "CPU",
      "Ports": [
        [9802, 8501, 1],
        [9802, 8501, 2],
        [9802, 8501, 3],
        [9802, 8502, 4] 
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [30, 15],
              [45, 35],
              [55, 50],
              [65, 55],
              [75, 60]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
       "Effects": [
        {
          "Type": "AuroraEffect",
          "Config": {
            "Gradient": [
                [0.0, [220, 0, 0]],
                [0.1, [215, 5, 0]],
                [0.2, [200, 10, 0]],
                [0.3, [140, 80, 0]],
                [0.4, [20, 255, 30]],
                [0.5, [0, 140, 170]],
                [0.6, [0, 80, 200]],
                [0.7, [55, 60, 160]],
                [0.8, [180, 30, 60]],
                [0.9, [200, 2, 4]],
                [1.00, [220, 0, 1]]

            ],
            "Step": 0.009,
            "Length": 256,
            "Mirror": false,
            "Brightness": 1,
            "Saturation": 1,

            "Trigger": {
              "Type": "AlwaysTrigger"

            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8501, 1], 
        [9802, 8501, 2], 
        [9802, 8501, 3], 
        [9802, 8501, 4],
        [9802, 8501, 5],
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4] 
      ],
      "Config": {
        "DeviceType": "RiingTrio"
      }
    }
  ]
}

if tried leaving the "gradient" part away, as it says in the docs it will go rainbow, but saldy i'm far off the color fidelity of the video...

MoshiMoshi0, 2020-06-30:
Try this:
```
    {
      "Type": "AuroraEffect",
      "Config": {
        "ColorGenerationMethod": "SpanPorts",
        "Step": 0.009,
        "Length": 60,
        "Mirror": true,
        "Brightness": 1,
        "Saturation": 1,
        "Trigger": {
          "Type": "AlwaysTrigger"
        }
      }
    }
```

By tweaking the `"Length"` you will get different behaviors.
If its equals to total led count in profile then it will create a perfectly looping gradient.
If its larger then you will only see a portion of the gradient at a time, this smoothes the gradient.
If its smaller then you will see the gradient loop multiple times.

Note that when using `"Mirror": true` the `"Length"` should be halved (because the other half is mirrored).
So for perfectly looping gradient: `Length = 4 ports in profile * 30 leds (Riing Trio) = 120`, if mirror is enabled then you have to halve it to 60.

But like I said AuroraEffect does not know how to handle multiple zones so it might look weird.

heyolli, 2020-06-30:
thank you so much!
I'll play around with it for some time.

Thanks again, love your work

closed 2020-06-30 by heyolli