MoshiMoshi0/TTController #56: Configuration Help

hpfuhl88, 2020-07-02:
**Describe your desired configuration**
I am trying to setup Razer Connect Effect for both sets of fans and LEDS in my config file.  However, I have been unable to get it to work I am getting consistent timeout error when running the configuration.

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Home 1909 [18362.1.amd64fre.19h1_release.190318-1202]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 35, Rpm: 663, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 35, Rpm: 684, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 35, Rpm: 678, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 35, Rpm: 648, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 35, Rpm: 657, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Trio
»        VendorId: 9802
»        ProductId: 8502
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 35, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 1]
»                2:
»                        Data: [PortId: 2, Speed: 35, Rpm: 528, Unknown: 255]
»                        Identifier: [9802, 8502, 2]
»                3:
»                        Data: [PortId: 3, Speed: 35, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 3]
»                4:
»                        Data: [PortId: 4, Speed: 35, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8502, 4]
»                5:
»                        Data: [PortId: 5, Speed: 35, Rpm: 530, Unknown: 255]
»                        Identifier: [9802, 8502, 5]
»        Available effect types: PerLed
»
-------------------------------
Sensors
-------------------------------
»
»  Nuvoton NCT6795D:
»        Temperature:
»                Temperature #1 (/lpc/nct6795d/temperature/1): 32.0 °C
»                Temperature #2 (/lpc/nct6795d/temperature/2): 31.0 °C
»                Temperature #3 (/lpc/nct6795d/temperature/3): 37.0 °C
»                Temperature #5 (/lpc/nct6795d/temperature/5): 44.0 °C
»                Temperature #6 (/lpc/nct6795d/temperature/6): -2.0 °C
»
»  AMD Ryzen 7 3800X:
»        Temperature:
»                Core (Tctl/Tdie) (/amdcpu/0/temperature/2): 36.6 °C
»                CCD1 (Tdie) (/amdcpu/0/temperature/3): 38.3 °C
»
»  AMD Radeon RX 5700 XT:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 33.0 °C
»                GPU Hot Spot (/gpu/0/temperature/6): 33.0 °C
»                GPU VR VDDC (/gpu/0/temperature/2): 32.0 °C
»                GPU Memory (/gpu/0/temperature/1): 34.0 °C
»
»  Samsung SSD 970 EVO Plus 250GB:
»        Temperature:
»                Temperature (/nvme/1/temperature/0): 49.0 °C
»                Temperature 1 (/nvme/1/temperature/6): 49.0 °C
»                Temperature 2 (/nvme/1/temperature/7): 53.0 °C
»
»  ST2000LX001-1RG174:
»        Temperature:
»                Temperature (/hdd/0/temperature/0): 26.0 °C

Make sure no official software is not running at the same time.
```

**Current configuration**
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
       "Effects": [
         {
          "Type": "RazerConnectEffect",
          "Trigger": {
              "Type": "AlwaysTrigger",
          }
        }
      ]
    },
    {  
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController", 
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Trigger": {
              "Type": "AlwaysTrigger",
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
      	[9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "DeviceType": "RiingPlus",
        "LedRotation": [3],
        "LedReverse": [false]
      }
    },
    {
      "Ports": [
      	[9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 85
      }
    }
  ]
}

MoshiMoshi0, 2020-07-03:
Run that configuration in console mode and post the output.

hpfuhl88, 2020-07-03:
This is what I am getting out of the log file.  However, it is missing the
exception error that appears at ttcontroller closes completely when it
fails in the console.

2020-07-02
21:34:06.8293|INFO|TTController.Service.TTService|================================================================
2020-07-02 21:34:06.8578|INFO|TTController.Service.TTService|Initializing...
2020-07-02
21:34:06.8578|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.DpsgController [0.10.0.0]
2020-07-02
21:34:06.8739|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingController [0.10.0.0]
2020-07-02
21:34:06.8739|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
2020-07-02
21:34:06.8739|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
2020-07-02
21:34:06.8739|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-07-02
21:34:06.8739|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-07-02
21:34:06.8904|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-07-02
21:34:06.8904|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.FlowEffect [0.10.0.0]
2020-07-02
21:34:06.8904|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-07-02
21:34:06.8904|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PulseEffect [0.10.0.0]
2020-07-02
21:34:06.9069|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RawEffect [0.10.0.0]
2020-07-02
21:34:06.9069|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RazerConnectEffect [0.10.0.0]
2020-07-02
21:34:06.9069|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RippleEffect [0.10.0.0]
2020-07-02
21:34:06.9069|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SensorEffect [0.10.0.0]
2020-07-02
21:34:06.9219|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SoundEffect [0.10.0.0]
2020-07-02
21:34:06.9219|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SpectrumEffect [0.10.0.0]
2020-07-02
21:34:06.9219|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.StaticColorEffect [0.10.0.0]
2020-07-02
21:34:06.9219|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.WaveEffect [0.10.0.0]
2020-07-02
21:34:06.9369|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.CopySpeedController [0.10.0.0]
2020-07-02
21:34:06.9369|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.DpsgSpeedController [0.10.0.0]
2020-07-02
21:34:06.9369|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PwmSpeedController [0.10.0.0]
[... 81 lines cut ...]
find device with name "RiingPlus"!
2020-07-02
21:34:07.6891|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-02
21:34:07.6891|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-02
21:34:07.6891|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-02 21:34:07.7196|INFO|TTController.Service.TTService|Applying
computer state profile: Boot
2020-07-02
21:34:07.7196|INFO|TTController.Service.Managers.TimerManager|Creating
Timer Manager...
2020-07-02
21:34:07.7196|INFO|TTController.Service.Managers.TimerManager|Starting 4
timers...
2020-07-02 21:34:07.7341|INFO|TTController.Service.TTService|Initializing
done!
2020-07-02
21:34:07.7341|INFO|TTController.Service.TTService|================================================================

Like I said I missing the exception text when as ttcontroller closes as
soon as it completes the console run or should I say fails the console run.

MoshiMoshi0, 2020-07-03:
Ok, edit `TTController.Service.exe.Config` file and change:
`      <logger name="*" minLevel="Info" writeTo="file" enabled="true" />`
to
`      <logger name="*" minLevel="Trace" writeTo="file" enabled="true" />`

Then rerun in console mode and post the latest log file it produces.

hpfuhl88, 2020-07-03:
Here is the updated log file...TTController still closes when it complete
the console run, not sure if that is supposed to be a thing or not.

2020-07-03
09:08:38.5618|INFO|TTController.Service.TTService|================================================================
2020-07-03 09:08:38.5903|INFO|TTController.Service.TTService|Initializing...
2020-07-03
09:08:39.3600|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.DpsgController [0.10.0.0]
2020-07-03
09:08:39.3600|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.DpsgController [0.10.0.0]
2020-07-03
09:08:40.0546|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.RiingController [0.10.0.0]
2020-07-03
09:08:40.0546|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingController [0.10.0.0]
2020-07-03
09:08:40.6481|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.RiingPlusController
[0.10.0.0]
2020-07-03
09:08:40.6481|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
2020-07-03
09:08:41.3402|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.RiingTrioController
[0.10.0.0]
2020-07-03
09:08:41.3437|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
2020-07-03
09:08:41.9412|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-07-03
09:08:41.9412|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-07-03
09:08:42.6153|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-07-03
09:08:42.6153|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-07-03
09:08:42.6243|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-07-03
09:08:42.6243|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-07-03
09:08:43.2153|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.FlowEffect [0.10.0.0]
2020-07-03
09:08:43.2193|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.FlowEffect [0.10.0.0]
2020-07-03
09:08:44.1981|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-07-03
09:08:44.2021|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-07-03
09:08:44.8322|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
assembly for type IPlugin: TTController.Plugin.PulseEffect [0.10.0.0]
2020-07-03
09:08:44.8322|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PulseEffect [0.10.0.0]
2020-07-03
09:08:45.4452|TRACE|TTController.Service.Utils.PluginLoader|Found plugin
[... 427 lines cut ...]
find device with name "RiingPlus"!
2020-07-03
09:08:59.6142|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-03
09:08:59.6142|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-03
09:08:59.6267|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-03
09:08:59.6267|WARN|TTController.Service.Managers.ConfigManager|Unable to
find device with name "RiingPlus"!
2020-07-03 09:08:59.6547|INFO|TTController.Service.TTService|Applying
computer state profile: Boot
2020-07-03
09:08:59.6547|INFO|TTController.Service.Managers.TimerManager|Creating
Timer Manager...
2020-07-03
09:08:59.6547|INFO|TTController.Service.Managers.TimerManager|Starting 4
timers...
2020-07-03 09:08:59.6713|INFO|TTController.Service.TTService|Initializing
done!
2020-07-03
09:08:59.6713|INFO|TTController.Service.TTService|================================================================

MoshiMoshi0, 2020-07-03:
I didnt even notice but you set `"DeviceType"` to `"RiingPlus"` on one of your port configs, there is no such device type and the docs do not mention that.
Riing plus is included in the `"Default"` type.
Remove `"DeviceType": "RiingPlus",` line and try again.

MoshiMoshi0, 2020-07-03:
But that will not fix the issue, I need to see the exception that is thrown to know why its crashing.
Try opening windows event viewer and going to Custom Views -> Administrative Events, there should be a couple of events after each crash, please post them as one will contain the exception.

MoshiMoshi0, 2020-07-03:
Also it seems like you only have 2 Riing Trio fans.
So you should probably remove the unused port identifiers:
```
        [9802, 8502, 1]
        [9802, 8502, 3]
        [9802, 8502, 4]
```

hpfuhl88, 2020-07-03:
*-* *System*
*-* *Provider*
[ *Name*] Windows Error Reporting
*-* *EventID* 1001
[ *Qualifiers*] 0
*Level* 4
*Task* 0
*Keywords* 0x80000000000000
*-* *TimeCreated*
[ *SystemTime*] 2020-07-03T14:50:23.614281800Z
*EventRecordID* 5810
*Channel* Application
*Computer* DESKTOP-L73F9GC
*Security*
*-* *EventData*
1368988738075992839
5
CLR20r3
Not available
0
TTController.Service.exe
0.10.0.0
f9903d05
TTController.Common
0.10.0.0
f4cc7eb3
96
0
System.NullReferenceException
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER802F.tmp.mdmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER812A.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER813B.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8149.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8169.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_TTController.Ser_bbb3fafe35904a1f50c74bdb8c729e4f1b676ebb_c550245c_0f8544ec-e569-4351-a0b3-72cd64d5a077
0
860da49d-b1b1-4dcf-b886-813956f7587b
268435456
fdfcf9924bebeced82ffa0010a87a307
0

*+* *System*
*-* *Provider*
[ *Name*] Application Error
*-* *EventID* 1000
[ *Qualifiers*] 0
*Level* 2
*Task* 100
*Keywords* 0x80000000000000
*-* *TimeCreated*
[ *SystemTime*] 2020-07-03T14:50:22.909683200Z
*EventRecordID* 5809
*Channel* Application
*Computer* DESKTOP-L73F9GC
*Security*
*-* *EventData*
TTController.Service.exe
0.10.0.0
f9903d05
unknown
0.0.0.0
00000000
c0000005
07f8ddf6
2090
01d651493748a915
C:\Users\hpfuh\Downloads\TTController_0.10.0.0-alpha.a826afd7df70a522c78f4785f357baa124b6a937\TTController.Service.exe
unknown
860da49d-b1b1-4dcf-b886-813956f7587b
*+* *System*
*-* *Provider*
[ *Name*] .NET Runtime
*-* *EventID* 1026
[ *Qualifiers*] 0
*Level* 2
*Task* 0
*Keywords* 0x80000000000000
*-* *TimeCreated*
[ *SystemTime*] 2020-07-03T14:50:22.815101700Z
*EventRecordID* 5808
*Channel* Application
*Computer* DESKTOP-L73F9GC
*Security*
*-* *EventData*
Application: TTController.Service.exe Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException at
TTController.Common.Plugin.EffectBase`1[[System.__Canon, mscorlib,
Version=4.0.0.0, Culture=neutral,
PublicKeyToken=b77a5c561934e089]].IsEnabled(TTController.Common.ICacheProvider)
at
TTController.Plugin.RazerConnectEffect.RazerConnectEffect.IsEnabled(TTController.Common.ICacheProvider)
at
TTController.Service.TTService.<DeviceRgbTimerCallback>b__21_1(TTController.Common.Plugin.IEffectBase)
at System.Linq.Enumerable.FirstOrDefault[[System.__Canon, mscorlib,
Version=4.0.0.0, Culture=neutral,
PublicKeyToken=b77a5c561934e089]](System.Collections.Generic.IEnumerable`1<System.__Canon>,
System.Func`2<System.__Canon,Boolean>) at
TTController.Service.TTService.DeviceRgbTimerCallback() at
TTController.Service.Managers.TimerManager+<>c__DisplayClass3_0.<RegisterTimer>b__0(System.Object,
TimerElapsedEventArgs) at
TTController.Service.Managers.TimerManager+Timer.ExecuteTimer() at
System.Threading.ThreadHelper.ThreadStart_Context(System.Object) at
System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext,
System.Threading.ContextCallback, System.Object, Boolean) at
System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext,
System.Threading.ContextCallback, System.Object, Boolean) at
System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext,
System.Threading.ContextCallback, System.Object) at
System.Threading.ThreadHelper.ThreadStart()

Those are the three error logs from top to bottom in event viewer after
running in console mode.

MoshiMoshi0, 2020-07-03:
Oh god, sorry, I should have read the config more carefully.
Your razer effects dont have a `"Config"` object:
```json
"Effects": [
  {
    "Type": "RazerConnectEffect",
    "Trigger": {
      "Type": "AlwaysTrigger",
    }
  }
]
```

Should be:
```json
"Effects": [
  {
    "Type": "RazerConnectEffect",
    "Config": {
      "Trigger": {
        "Type": "AlwaysTrigger",
      }
    }
  }
]
```

Weird that it hard crashes like that without showing the error.

hpfuhl88, 2020-07-03:
so I tried running it against with that change and got the following error
in console that thankfully did not force close the program...

13:15:29.9607:
================================================================
13:15:29.9887: Initializing...
13:15:30.0027: Loading plugin assembly: TTController.Plugin.DpsgController
[0.10.0.0]
13:15:30.0147: Loading plugin assembly: TTController.Plugin.RiingController
[0.10.0.0]
13:15:30.0147: Loading plugin assembly:
TTController.Plugin.RiingPlusController [0.10.0.0]
13:15:30.0328: Loading plugin assembly:
TTController.Plugin.RiingTrioController [0.10.0.0]
13:15:30.0328: Loading plugin assembly: TTController.Plugin.AuroraEffect
[0.10.0.0]
13:15:30.0483: Loading plugin assembly: TTController.Plugin.BlinkEffect
[0.10.0.0]
13:15:30.0483: Loading plugin assembly: TTController.Plugin.CopyColorEffect
[0.10.0.0]
13:15:30.0633: Loading plugin assembly: TTController.Plugin.FlowEffect
[0.10.0.0]
13:15:30.0633: Loading plugin assembly: TTController.Plugin.PingPongEffect
[0.10.0.0]
13:15:30.0798: Loading plugin assembly: TTController.Plugin.PulseEffect
[0.10.0.0]
13:15:30.0798: Loading plugin assembly: TTController.Plugin.RawEffect
[0.10.0.0]
13:15:30.0923: Loading plugin assembly:
TTController.Plugin.RazerConnectEffect [0.10.0.0]
13:15:30.0923: Loading plugin assembly: TTController.Plugin.RippleEffect
[0.10.0.0]
13:15:30.1083: Loading plugin assembly: TTController.Plugin.SensorEffect
[0.10.0.0]
13:15:30.1273: Loading plugin assembly: TTController.Plugin.SoundEffect
[0.10.0.0]
13:15:30.1399: Loading plugin assembly: TTController.Plugin.SpectrumEffect
[0.10.0.0]
13:15:30.1399: Loading plugin assembly:
TTController.Plugin.StaticColorEffect [0.10.0.0]
13:15:30.1569: Loading plugin assembly: TTController.Plugin.WaveEffect
[0.10.0.0]
13:15:30.1569: Loading plugin assembly:
TTController.Plugin.CopySpeedController [0.10.0.0]
13:15:30.1734: Loading plugin assembly:
TTController.Plugin.DpsgSpeedController [0.10.0.0]
13:15:30.1734: Loading plugin assembly:
TTController.Plugin.PwmSpeedController [0.10.0.0]
13:15:30.1879: Loading plugin assembly:
TTController.Plugin.StaticSpeedController [0.10.0.0]
13:15:30.1879: Loading plugin assembly: TTController.Plugin.AlwaysTrigger
[0.10.0.0]
13:15:30.2049: Loading plugin assembly: TTController.Plugin.LogicTrigger
[0.10.0.0]
13:15:30.2049: Loading plugin assembly: TTController.Plugin.OneTimeTrigger
[0.10.0.0]
13:15:30.2214: Loading plugin assembly: TTController.Plugin.ProcessTrigger
[0.10.0.0]
13:15:30.2334: Loading plugin assembly: TTController.Plugin.PulseTrigger
[0.10.0.0]
13:15:30.2334: Loading plugin assembly: TTController.Plugin.ScheduleTrigger
[0.10.0.0]
13:15:30.2510: Loading plugin assembly: TTController.Plugin.SensorTrigger
[0.10.0.0]
13:15:30.2510: Creating Config Manager...
13:15:30.3250: Loading config...
13:15:30.5027: Newtonsoft.Json.JsonSerializationException: Unexpected end
when deserializing array. Path 'Profiles[1]', line 129, position 1.
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.ThrowUnexpectedEndException(JsonReader
reader, JsonContract contract, Object currentObject, String message)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateList(IList
list, JsonReader reader, JsonArrayContract contract, JsonProperty
containerProperty, String id)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateList(JsonReader
reader, Type objectType, JsonContract contract, JsonProperty member, Object
existingValue, String id)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader
reader, Type objectType, JsonContract contract, JsonProperty member,
JsonContainerContract containerContract, JsonProperty containerMember,
Object existingValue)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.SetPropertyValue(JsonProperty
property, JsonConverter propertyConverter, JsonContainerContract
containerContract, JsonProperty containerProperty, JsonReader reader,
Object target)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object
newObject, JsonReader reader, JsonObjectContract contract, JsonProperty
member, String id)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader
reader, Type objectType, JsonContract contract, JsonProperty member,
JsonContainerContract containerContract, JsonProperty containerMember,
Object existingValue)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader
reader, Type objectType, JsonContract contract, JsonProperty member,
JsonContainerContract containerContract, JsonProperty containerMember,
Object existingValue)
   at
Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader
reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader,
Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type
type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value,
JsonSerializerSettings settings)
   at TTController.Service.Managers.ConfigManager.LoadOrCreateConfig()
13:15:30.5027: Failed to load the config!

MoshiMoshi0, 2020-07-03:
You must be editing the json wrong. Post the whole config.

You can verify json here: https://jsoneditoronline.org
It will show you any errors.

hpfuhl88, 2020-07-03:
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
"Effects": [
  {
    "Type": "RazerConnectEffect",
    "Config": {
      "Trigger": {
        "Type": "AlwaysTrigger"
      }
    }
  }
]
    },
    {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
"Effects": [
  {
    "Type": "RazerConnectEffect",
    "Config": {
      "Trigger": {
        "Type": "AlwaysTrigger"
      }
    }
   }
  ]
 },
  "PortConfigs": [
    {
      "Ports": [
       [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [false]
      }
    },
    {
      "Ports": [
       [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 85
      }
    }
  ]
}

whole config pasted above. The JSON editor said a bad string on line 36,
which I got it to autocorrect.  On Line 79 it claims an expectation error
where it is looking for a "," instead of ":".  If I remove the colon and
put in a comma it clears the error, but I don't think that will work in the
program.

MoshiMoshi0, 2020-07-03:
```json
{
  "Profiles": [
    {
      "Name": "1",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    },
    {
      "Name": "2",
      "Ports": [
        [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [20, 30],
              [40, 37],
              [50, 40],
              [65, 75],
              [75, 100]
            ],
            "Sensors": ["/amdcpu/0/temperature/2"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [
        {
          "Type": "RazerConnectEffect",
          "Config": {
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
       [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
        [9802, 8101, 4],
        [9802, 8101, 5]
      ],
      "Config": {
        "Name": "Bottom Fans",
        "LedRotation": [3],
        "LedReverse": [false]
      }
    },
    {
      "Ports": [
       [9802, 8502, 1],
        [9802, 8502, 2],
        [9802, 8502, 3],
        [9802, 8502, 4],
        [9802, 8502, 5]
      ],
      "Config": {
        "Name": "Top Fans",
        "DeviceType": "RiingTrio",
        "LedRotation": [3, 3],
        "LedReverse": [true, false]
      }
    }
  ],
  "SensorConfigs": [
    {
      "Sensors": [
        "/gpu/0/temperature/1",
        "/lpc/nct6795d/temperature/1",
        "/lpc/nct6795d/temperature/2",
        "/lpc/nct6795d/temperature/3",
        "/lpc/nct6795d/temperature/5",
        "/amdcpu/0/temperature/2",
        "/amdcpu/0/temperature/3",
        "/gpu/0/temperature/0",
        "/gpu/0/temperature/6",
        "/gpu/0/temperature/2"
      ],
      "Config": {
        "Offset": "-3",
        "CriticalValue": 85
      }
    }
  ]
}
```

hpfuhl88, 2020-07-03:
Well that work for the most part bottom fans are responding to the Chroma
software, but the Riing Trios are not.  They are following the pwm
protocol, but no the color protocol.

MoshiMoshi0, 2020-07-03:
Do the same change in `TTController.Service.exe.Config` I mentioned earlier, run the service in console mode for a few seconds then stop it and post latest log.

hpfuhl88, 2020-07-03:
2020-07-03
15:02:55.5794|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8101] 00, 32, 52, FC, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5794|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8101] 00, 32, 52, 02, 18, 00, 34, 00, 00, 34, 00, 00, 34, 00, 00, 34, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5794|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8101] 00, 32, 52, FC, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5945|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8101] 00, 32, 52, 03, 18, 00, 34, 00, 00, 34, 00, 00, 34, 00, 00, 34, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5945|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8101] 00, 32, 52, FC, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5945|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8101] 00, 32, 52, 04, 18, 00, 34, 00, 00, 34, 00, 00, 34, 00, 00, 34, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.5945|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8101] 00, 32, 52, FC, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.6095|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8101] 00, 32, 52, 05, 18, 00, 34, 00, 00, 34, 00, 00, 34, 00, 00, 34, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.6095|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8101] 00, 32, 52, FC, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.6095|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8502] 00, 32, 52, 01, 24, 03, 01, 00, 00, 34, 00, 00, 34, 00, 00, 34, 00,
00, 34, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.6095|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8502] 00, 32, 52, FC, 01, 24, 03, 01, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:55.6095|TRACE|TTController.Common.Plugin.EffectBase`1|Razer broadcast
colors updated
2020-07-03
[... 1065 lines cut ...]
15:02:56.4387|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8502] 00, 32, 52, FC, 05, 24, 03, 01, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:56.4532|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8502] 00, 32, 52, 05, 24, 03, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:56.4532|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8502] 00, 32, 52, FC, 05, 24, 03, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:56.4532|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8101] 00, 32, 52, 01, 18, 00, 03, 00, 00, 03, 00, 00, 03, 00, 00, 03, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:02:56.4677|TRACE|TTController.Service.Hardwar

MoshiMoshi0, 2020-07-03:
Post the whole log file, they are created in `Logs` folder

hpfuhl88, 2020-07-03:
2020-07-03
15:37:43.6942|INFO|TTController.Service.TTService|================================================================
2020-07-03 15:37:43.7262|INFO|TTController.Service.TTService|Initializing...
2020-07-03
15:37:43.7413|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.DpsgController [0.10.0.0]
2020-07-03
15:37:43.7503|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingController [0.10.0.0]
2020-07-03
15:37:43.7503|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingPlusController [0.10.0.0]
2020-07-03
15:37:43.7503|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RiingTrioController [0.10.0.0]
2020-07-03
15:37:43.7658|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.AuroraEffect [0.10.0.0]
2020-07-03
15:37:43.7658|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.BlinkEffect [0.10.0.0]
2020-07-03
15:37:43.7658|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.CopyColorEffect [0.10.0.0]
2020-07-03
15:37:43.7818|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.FlowEffect [0.10.0.0]
2020-07-03
15:37:43.7818|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PingPongEffect [0.10.0.0]
2020-07-03
15:37:43.7818|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PulseEffect [0.10.0.0]
2020-07-03
15:37:43.7973|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RawEffect [0.10.0.0]
2020-07-03
15:37:43.7973|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RazerConnectEffect [0.10.0.0]
2020-07-03
15:37:43.7973|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.RippleEffect [0.10.0.0]
2020-07-03
15:37:43.8133|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SensorEffect [0.10.0.0]
2020-07-03
15:37:43.8308|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SoundEffect [0.10.0.0]
2020-07-03
15:37:43.8308|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.SpectrumEffect [0.10.0.0]
2020-07-03
15:37:43.8308|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.StaticColorEffect [0.10.0.0]
2020-07-03
15:37:43.8473|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.WaveEffect [0.10.0.0]
2020-07-03
15:37:43.8473|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.CopySpeedController [0.10.0.0]
2020-07-03
15:37:43.8579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.DpsgSpeedController [0.10.0.0]
2020-07-03
15:37:43.8579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.PwmSpeedController [0.10.0.0]
2020-07-03
15:37:43.8579|INFO|TTController.Service.Utils.PluginLoader|Loading plugin
assembly: TTController.Plugin.StaticSpeedController [0.10.0.0]
2020-07-03
[... 1152 lines cut ...]
15:32:44.4993|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8502] 00, 33, 51, 04, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:32:44.4993|TRACE|TTController.Service.Hardware.HidDeviceProxy|R[9802,
8502] 00, 33, 51, 04, FF, 24, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:32:44.5189|TRACE|TTController.Common.Plugin.EffectBase`1|Razer broadcast
colors updated
2020-07-03
15:32:44.5189|TRACE|TTController.Common.Plugin.EffectBase`1|Razer broadcast
colors updated
2020-07-03
15:32:44.5189|TRACE|TTController.Service.Hardware.HidDeviceProxy|W[9802,
8502] 00, 33, 51, 05, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00,
00, 00, 00, 00, 00, 00, 00, 00, 00, 00
2020-07-03
15:32:44.5314|

hpfuhl88, 2020-07-03:
Nvm got it fixed the fans needed the Lerp command put in for the LED County
Handling.

MoshiMoshi0, 2020-07-03:
Yes, was about to write that.

Next time please explain what is happening in more detail, I thought the Riing Trio fans were not getting any colors at all.
Also fyi you should respond on github not thru emails because it is very hard to read. I did not see any attached files so maybe that does not work thru email either.

Anyways, glad you have it working.

closed 2020-07-03 by MoshiMoshi0