MoshiMoshi0/TTController #57: Configuration Help

Maurice1986, 2020-07-04:
**Describe your desired configuration**

I am trying to setup Razer Connect Effect on my pure plus 3 of the led is not working.

**Hardware info**
```
Info
-------------------------------
»
»        OS: Windows 10 Enterprise 2004 [19041.1.x86fre.vb_release.191206-1406]
»        Build: 0.10.0-alpha+Branch.master.Sha.a826afd7df70a522c78f4785f357baa124b6a937
»
-------------------------------
Controllers
-------------------------------
»
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8101
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 1173, Unknown: 255]
»                        Identifier: [9802, 8101, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 1168, Unknown: 255]
»                        Identifier: [9802, 8101, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8101, 3]
»                4:
»                        Data: [PortId: 4, Speed: 65, Rpm: 1167, Unknown: 255]
»                        Identifier: [9802, 8101, 4]
»                5:
»                        Data: [PortId: 5, Speed: 36, Rpm: 685, Unknown: 255]
»                        Identifier: [9802, 8101, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»  Name: Riing Plus
»        VendorId: 9802
»        ProductId: 8102
»        Ports:
»                1:
»                        Data: [PortId: 1, Speed: 65, Rpm: 1176, Unknown: 255]
»                        Identifier: [9802, 8102, 1]
»                2:
»                        Data: [PortId: 2, Speed: 65, Rpm: 1155, Unknown: 255]
»                        Identifier: [9802, 8102, 2]
»                3:
»                        Data: [PortId: 3, Speed: 65, Rpm: 1162, Unknown: 255]
»                        Identifier: [9802, 8102, 3]
»                4:
»                        Data: [PortId: 4, Speed: 36, Rpm: 690, Unknown: 255]
»                        Identifier: [9802, 8102, 4]
»                5:
»                        Data: [PortId: 5, Speed: 60, Rpm: 0, Unknown: 255]
»                        Identifier: [9802, 8102, 5]
»        Available effect types: Flow_Extreme, Flow_Fast, Flow_Normal, Flow_Slow, Spectrum_Extreme, Spectrum_Fast, Spectrum_Normal, Spectrum_Slow, Ripple_Extreme, Ripple_Fast, Ripple_Normal, Ripple_Slow, Blink_Extreme, Blink_Fast, Blink_Normal, Blink_Slow, Pulse_Extreme, Pulse_Fast, Pulse_Normal, Pulse_Slow, Wave_Extreme, Wave_Fast, Wave_Normal, Wave_Slow, PerLed, Full
»
-------------------------------
Sensors
-------------------------------
»
»  Fintek F71889AD:
»        Temperature:
»                Temperature #1 (/lpc/f71889ad/temperature/0): 35,0 °C
»                Temperature #3 (/lpc/f71889ad/temperature/2): 26,0 °C
»
»  Intel Core i7-3770K:
»        Temperature:
»                CPU Core #1 (/intelcpu/0/temperature/0): 32,0 °C
»                CPU Core #2 (/intelcpu/0/temperature/1): 30,0 °C
»                CPU Core #3 (/intelcpu/0/temperature/2): 30,0 °C
»                CPU Core #4 (/intelcpu/0/temperature/3): 32,0 °C
»                CPU Package (/intelcpu/0/temperature/4): 36,0 °C
»                CPU Core #1 Distance to TjMax (/intelcpu/0/temperature/5): 73,0 °C
»                CPU Core #2 Distance to TjMax (/intelcpu/0/temperature/6): 75,0 °C
»                CPU Core #3 Distance to TjMax (/intelcpu/0/temperature/7): 75,0 °C
»                CPU Core #4 Distance to TjMax (/intelcpu/0/temperature/8): 73,0 °C
»                Core Max (/intelcpu/0/temperature/9): 32,0 °C
»                Core Average (/intelcpu/0/temperature/10): 31,0 °C
»
»  Radeon RX 580 Series:
»        Temperature:
»                GPU Core (/gpu/0/temperature/0): 27,0 °C
»
»  ST1000LM014-1EJ164:
»        Temperature:
»                Temperature (/hdd/1/temperature/0): 31,0 °C
»
»  ADATA SX900:
»        Temperature:
»                Temperature (/ssd/0/temperature/0): 36,0 °C
```

**Current configuration**

```{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8101, 5],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3],
		[9802, 8102, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [                              
        {
          "Type": "RazerConnectEffect",
          "Config": {            
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3]
      ],
      "Speed": 65,
      "EffectType": "Full",
      "EffectColors": [
        [128, 0, 0]
      ]
    }
  ],
  "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
		[9802, 8101, 4],
		[9802, 8101, 5],
		[9802, 8102, 1],
		[9802, 8102, 2],
		[9802, 8102, 3],
		[9802, 8102, 4]
      ],
      "Config": {
        "LedCountHandling": "Lerp"
      }
    }
  ]
}
```

MoshiMoshi0, 2020-07-04:
Your config was slightly wrong, try this:
```json
{
  "Profiles": [
    {
      "Name": "Default",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
    		[9802, 8101, 4],
    		[9802, 8101, 5],
    		[9802, 8102, 1],
    		[9802, 8102, 2],
    		[9802, 8102, 3],
    		[9802, 8102, 4]
      ],
      "SpeedControllers": [
        {
          "Type": "PwmSpeedController",
          "Config": {
            "CurvePoints": [
              [30, 30],
              [45, 50],
              [55, 60],
              [65, 80],
              [80, 100]
            ],
            "Sensors": ["/intelcpu/0/temperature/4"],
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ],
      "Effects": [                              
        {
          "Type": "RazerConnectEffect",
          "Config": {            
            "Trigger": {
              "Type": "AlwaysTrigger"
            }
          }
        }
      ]
    }
  ],
  "ComputerStateProfiles": [
    {
      "StateType": "Shutdown",
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
    		[9802, 8101, 4],
    		[9802, 8102, 1],
    		[9802, 8102, 2],
    		[9802, 8102, 3]
      ],
      "Speed": 65,
      "Color": {
        "Full": [255, 0, 0]
      }
    }
  ],
  "PortConfigs": [
    {
      "Ports": [
        [9802, 8101, 1],
        [9802, 8101, 2],
        [9802, 8101, 3],
    		[9802, 8101, 4],
    		[9802, 8101, 5],
    		[9802, 8102, 1],
    		[9802, 8102, 2],
    		[9802, 8102, 3],
    		[9802, 8102, 4]
      ],
      "Config": {
        "LedCountHandling": "Lerp"
      }
    }
  ]
}
```

Maurice1986, 2020-07-04:
Thank you works like a charm

closed 2020-07-04 by Maurice1986