MoshiMoshi0/TTController #58: Configuration Help

duncanthrax, 2020-07-07:
Thank you for a great piece of software! Glad I got rid of the crap that TT is shipping ...

I have a Riing Trio. It has 30 LEDs in 3 ([12, 12, 6]) zones. Works fine, except I want to address all LEDs with a static color. When using ```StaticColorEffect``` with ```Full```, only the first zone is lit. When I use ```PerLed``` instead of full, every color past the first 12 gets ignored.

How can I address LEDs in the other zones? Maybe I'm missing something, but I didn't find it in the docs ...

MoshiMoshi0, 2020-07-07:
I think you did not configure a `PortConfig` with riing trio [device type](https://moshimoshi0.github.io/TTController/config/port#devicetype).

If you did, please post the whole config.

duncanthrax, 2020-07-07:
You're right! Case closed, thanks again, and sorry for the interruption :)

closed 2020-07-07 by duncanthrax